platformio/platformio-core #3990: Add support for subdirectories when adding git repos as a library

jimbobbennett, 2021-06-30:
Some libraries are available as folders inside a Git repo. One example being ArduCAM.

The Arduino code for the ArduCAM is in this repo: https://github.com/ArduCAM/Arduino. Rather than have one library in this repo, it contains multiple libraries each inside a folder in this repo. It would be a great enhancement to be able to specify a folder inside a git repo for a library dependency to remove the need to configure library directories, or download the repo and copy one folder into the `lib` folder.

For example, being able to write something like:

```ini
lib_deps =
    https://github.com/ArduCAM/Arduino/tree/master/ArduCAM
```

TheButlah, 2022-05-24:
This would also help pio work better with monorepos!

DreamPWJ, 2022-07-21:
This feature is important

christhomas, 2022-10-22:
Yeah this would be a very useful feature

dmezh, 2023-01-29:
Agreed, would be quite useful

teejaydub, 2023-02-02:
It would also make it possible to override individual platform libraries with forks, which unfortunately I need to do, and the workaround (separate management of a local copy of that repo along with a symlink:// lib_dep) is error-prone.

nicklasb, 2024-02-25:
I would agree to this, a whole bunch of ESP-IDF components, for example, are in the [ESP-BSP](https://github.com/espressif/esp-bsp/tree/master/components)-repository and I cannot add them as it is now.
There are many more examples of this, please add this functionality. 

Accepting a https://github.com/espressif/esp-bsp.git/components/esp_lvgl_port-syntax, by identifying the break at the ".git", or using some other scheme. Or add a "path", as in idf_component.yml. It is quite usable. 
Maybe the best variant, as then you could combine with other schemes, and it is the most clear.

nicklasb, 2024-02-25:
I also just noted that the maximum length of the URL, it seems like it is only 100 characters, which is super tight and will probably run out in many cases. The max length for most browsers are around 2000 characters, and several permanent link schemes has more than 100 characters. 
Basically I see no 2024 reason for restricting the URL length in such a severe manner.