platformio/platformio-core #3992: Serial Monitor swallows my print lines

ullix, 2021-07-02:
Any `Serial.print*()` command does not seem to make it to output when it is preceded by a `log_X()` command, with X= e, i, d, w, v. This simple script:

```
#include <Arduino.h>
#include <esp_log.h>

void setup() {
    Serial.begin(115200);
    Serial.println("setup started ###########################################################");

    log_i("%s",  "some text");
    log_i("%s",  "more text");
    log_i("%s",  "even more text");

    Serial.println("1");
    Serial.println("2");
    Serial.println("3");
    Serial.println("4");
    Serial.println("5");

    log_e("%s",  "but where are my numbers?");
}

void loop() {}
```
produces this output:
```
18:35:24.821 > setup started ###########################################################
18:35:24.821 > [I][main.cpp:8] setup(): some text
18:35:24.821 > [I][main.cpp:9] setup(): more text
18:35:24.821 > [I][main.cpp:10] setup(): even more text
18:35:24.821 > [E][main.cpp:18] setup(): but where are my numbers?
```

Yes, where are the numbers?

Platformio.ini:
```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
monitor_speed = 115200
upload_speed = 921600
upload_port = /dev/ttyUSB0
build_flags = -D CORE_DEBUG_LEVEL=5
monitor_filters =
	log2file
	time
	default

[platformio]
description = Serial Monitor swallows print lines
```

maxgerhardt, 2021-07-04:
I think `log_i` goes a `printf` path and `Serial` takes a different path and they might be overwriting each others buffers or something. Have you tried putting `Serial.flush();` before and after the `Serial.println()` block? Or a `delay()` to make sure `log_i` and its sub-functions are done using the UART?

maxgerhardt, 2021-07-04:
By the way, if you take this sketch and upload it via the Arduino IDE with the same settings (especially regarding debug level) and the same used core version (1.0.6 as of now), does the same output appear? If yes, it's immediately a Arduino-ESP32 issue, not a PlatformIO one.

ullix, 2021-07-04:
> Have you tried putting `Serial.flush();` before and after the `Serial.println()` block?

A flush() before the first println() suffices to get all numbers printed.

ullix, 2021-07-04:
Tested on Arduino 1.8.13 and 2.0.0Beta7:

Exactly the same behavior: no numbers unless a Serial.flush() precedes the println().

maxgerhardt, 2021-07-06:
Okay then it seems to be in issue in their core. You should try and file a issue there to get the `Serial` and `log_i()` etc. functions / streams playing more nicely with each other.

ullix, 2021-07-07:
And where might "there" be?

maxgerhardt, 2021-07-07:
The core repo is at https://github.com/espressif/arduino-esp32.

ullix, 2021-07-07:
reported here: https://github.com/espressif/arduino-esp32/issues/5362

SuGlider, 2021-07-16:
Fixed: https://github.com/espressif/arduino-esp32/issues/5362
Check PR https://github.com/espressif/arduino-esp32/pull/5390

closed 2021-09-13 by ivankravets