platformio/platformio-core #3995: create new raspberry pi4 project error

K0rkunc, 2021-07-04:
hi i wantto ccreate new rpi 4 project but gives errors

PIO Core Call Error: "The current working directory C:\\Users\\OnurNP\\Documents\\PlatformIO\\Projects\\as will be used for the project.\r\n\r\nThe next files/directories have been created in C:\\Users\\OnurNP\\Documents\\PlatformIO\\Projects\\as\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\n\n\nError: PlatformIO temporary does not support cross-compilation for WiringPi framework. Please use PIO Core directly on Raspberry Pi

septatrix, 2021-07-07:
Raspberry Pi 4 does not seem supported yet: https://github.com/platformio/platform-linux_arm/issues/14
Also the error message seems to suggest that you are developing from windows however PIO does not seem to support cross-compilation

ivankravets, 2021-09-13:
PlatformIO temporary does not support cross-compilation for WiringPi framework. Please use PIO Core directly on Raspberry Pi

closed 2021-09-13 by ivankravets