platformio/platformio-core #3997: Renode Framework

Emna-b, 2021-07-05:
What kind of issue is this?

- [*] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

I am working on a project with Renode Framework. In fact I am trying to add new boards to the list of supported boards by the Renode Framework. How can I do so ?
Any help is greatly appreciated.

maxgerhardt, 2021-07-13:
>In fact I am trying to add new boards to the list of supported boards by the Renode Framework. 

So you want to extend Renode to work with more boards? Ask at the renode repo https://github.com/renode/renode.

ivankravets, 2021-07-28:
Please forward to https://github.com/renode/renode

closed 2021-07-28 by ivankravets