platformio/platformio-core #4004: Authentication issue

SergiuToporjinschi, 2021-07-13:
What kind of issue is this?
- [X ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:
Windows 
**PlatformIO Version** (`platformio --version`):
PlatformIO Core, version 5.1.1

### Description of problem
Yesterday I tried to publish a new lib. But I forgot the account password. So I did a `pio account forgot` I received a link on my e-mail address and I changed the password in the webpage; 
I managed to publish the new lib but there it was a problem with  it because when I tried to  used in other project I saw that 
the owner is my e-mail address so in Registry Library in Installation tab the dependency example was looking like "sergiu.toporjinschi(AT)gmail.com/ESPConsole @ ^1.0.2" and because of this I could not install it.
All my other library uses the user name as prefix lib "sergiutoporjinschi" but this one was using the e-mail address;
I've done a "pio account show" which shows my e-mail address as user name I tried to changed it and response was that there is already other user is 'sergiutoporjinschi'

so I have the filling that when I've changed the password a new account was created

I removed (unpubished) the new library from sergiu.Toporjinschi(AT)gmail.com  and I tried again to authenticate using the username and my new password but I received an error 

pio account login 
Username or email: sergiutoporjinschi
Password: 
Error: Internal server error. Please report to https://github.com/platformio/platformio-core/issues

Now if there are to accounts one with sergiutoporjinschi and one with my e-mail address I would like to keep the one with the user name and remove the one with  my e-mail address but I would like to be able to authenticate with  my user name.
Thank you!
PS. I replaced @ with  (AT) because I don't know if this is a public post.

#### Steps to Reproduce

1. IS more like a specific thing... 
2.
3.

### Actual Results

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
Insert here...
```

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info

maxgerhardt, 2021-07-19:
@ivankravets account problems with `@` in username.

ivankravets, 2021-07-21:
Fixed. Please re-login.

closed 2021-07-21 by ivankravets

SergiuToporjinschi, 2021-07-23:
Works, Thank you