platformio/platformio-core #4008: Compile marlin is error on cmd but vscode is success

zhpt, 2021-07-19:
What kind of issue is this?

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:Windows10

**PlatformIO Version** (`platformio --version`):5.1.1

### Description of problem

I want to compile Marlin sources files (version 2.0.7.2).
I try to compile on cmd with "platformio run", but it was error about "Marlin\src\module/stepper/trinamic.h:29:10: fatal error: TMCStepper.h: No such file or directory".

But when I try on VS Code , it can compile success.

[image]
[image]
[image]

What's wrong with it?
Thank you.

#### Steps to Reproduce

1.open cmd
2.platformio run

### Actual Results
compile success

### Expected Results
Marlin\src\module/stepper/trinamic.h:29:10: fatal error: TMCStepper.h: No such file or directory

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
#
# Marlin Firmware
# PlatformIO Configuration File
#
# For detailed documentation with EXAMPLES:
#
# https://docs.platformio.org/en/latest/projectconf/index.html
#

# Automatic targets - enable auto-uploading
#targets = upload

#
# By default platformio build will abort after 5 errors.
# Remove '-fmax-errors=5' from build_flags below to see all.
#

[platformio]
src_dir      = Marlin
boards_dir   = buildroot/share/PlatformIO/boards
default_envs = mega2560
include_dir  = Marlin

#
# The 'common' values are used for most Marlin builds
#
[... 1333 lines cut ...]
platform        = native
framework       =
build_flags     = -D__PLAT_LINUX__ -std=gnu++17 -ggdb -g -lrt -lpthread -D__MARLIN_FIRMWARE__ -Wno-expansion-to-defined
src_build_flags = -Wall -IMarlin/src/HAL/LINUX/include
build_unflags   = -Wall
lib_ldf_mode    = off
lib_deps        =
src_filter      = ${common.default_src_filter} +<src/HAL/LINUX>

#
# Just print the dependency tree
#
[env:include_tree]
platform    = atmelavr
board       = megaatmega2560
build_flags = -c -H -std=gnu++11 -Wall -Os -D__MARLIN_FIRMWARE__
src_filter  = +<src/MarlinCore.cpp>
```

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info

maxgerhardt, 2021-07-19:
What does `pio --version` return in cmd.exe and [VSCode CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli)?

zhpt, 2021-07-20:
> What does `pio --version` return in cmd.exe and [VSCode CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli)?
cmd return "PlatformIO Core, version 5.1.1"
vscode terminal also return this "PlatformIO Core, version 5.1.1"
but vscode PlatformIO IDE version is v2.3.2
[image]
[image]
[image]

maxgerhardt, 2021-07-20:
> but vscode PlatformIO IDE version is v2.3.2

Yes, the extension version is different from the core version, that's expected. 

What about that `Microsoft_PowerShell_profile.ps1` error? You have custom profile scripts for the shell, and that might influence it? 

Also, can you, in the `cmd.exe`, do a `pio upgrade --dev` and try `pio run` again? I don't yet see why it would work in the IDE but on in the `cmd.exe`.

maxgerhardt, 2021-07-20:
Oh wait, I take that back. The path of the project folder in `H:.....` seems to be *extremely long*. Windows has limitations on maximum path length. 

Please move the folder to a shorter path, like `H:\Marlin`, and try `pio run` in there from `cmd.exe`. Does it work now?

zhpt, 2021-07-22:
@maxgerhardt Thank you for your help. Follow your step is compiled successful on cmd.exe. And this error mainly beacuse my program modified the platformio.ini file unexpectedly . Thanks again.

closed 2021-09-13 by ivankravets