platformio/platformio-core #4009: ESP8266 Debugging Support

mateja176, 2021-07-20:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [x] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Description of problem

I would like to debug the code I wrote for my NodeMCU 1.0, but when I attempt to start debugging I run into the following error:
```txt
Error: Currently, PlatformIO does not support debugging for `NodeMCU 1.0 (ESP-12E Module)`.
Please request support at https://github.com/platformio/platformio-core/issues 
or visit -> https://docs.platformio.org/page/plus/debugging.html
```

#### Steps to Reproduce

1. Open NodeMCU 1.0 based Arduino Project
2. Build and upload your project
3. Start debugging

### Additional info

I'm aware that there's [debugging support](https://docs.platformio.org/en/latest/plus/debugging.html#nodemcu) for the newer ESP32 MCU, nevertheless the ESP8266 is still a very popular and widespread chip.

valeros, 2021-07-21:
Duplicate https://github.com/platformio/platform-espressif8266/issues/43

closed 2021-07-21 by valeros

AgainPsychoX, 2024-07-06:
You might be interested: https://github.com/platformio/platform-espressif8266/issues/43#issuecomment-2211974061