platformio/platformio-core #4018: Library Dependency Finder should take symlinked directories into account

trullock, 2021-08-02:
What kind of issue is this?

- [x] **PlatformIO Core**.
------------------------------------------------------------------

### Configuration

**Operating system**: Windows 10

**PlatformIO Version**: 5.1.1

### Description of problem

This is the same as the closed iss #823, I'm reopening the issue here so it gets seen. Feel free to close and resume conversations on #823 if more appropriate.

I have a library (`mylib`) which contains mylib.h and mylib.cpp

`./lib/mylib` is a symbolic link (or junction, both fail) to a folder elsewhere

in my `main.cpp` i do: `#include "./lib/mylib/mylib.h"` and intellisense all works fine

When compiling, LDF doesn't pick up `mylib.cpp`, even if I include the folder explicity with `lib_extra_dirs`

If I make `./lib/mylib` a real dir with the files in it, compilation works. Swap it out for a symlink, compilation fails because the cpp is never compiled so linking fails

Symlinks should be transparent to LDF, it shouldn't care if a folder is symlinked or not it should just explore it

It handles symlinked files fine. if i make `./lib/mylib` a real folder and then symlink all the files inside it, this works.

ivankravets, 2021-08-03:
> in my main.cpp i do: #include "./lib/mylib/mylib.h"

It should be just `#include <mylib.h>`.

Does it work?

trullock, 2021-08-03:
`#include <mylib.h>` works!

So perhaps this is not the issue as I've described it but its certainly some sort of issue.

I'd expect `<>` to just be some path magic, to save having to write the whole relative path a la `"./lib/mylib/mylib.h"`

Id still expect `#include "./lib/mylib/mylib.h"` to work

ivankravets, 2021-08-04:
It is great that it works. This is a normal behavior of the [Library Dependency Finder](https://docs.platformio.org/en/latest/librarymanager/ldf.html). You can turn it off. See docs.

closed 2021-08-04 by ivankravets

trullock, 2021-08-05:
Hi

Can you point me at the bit of the docs which describes how to "turn it off"? I can't see what you're referring to

Thanks

ivankravets, 2021-08-05:
See https://docs.platformio.org/en/latest/projectconf/section_env_library.html#projectconf-lib-ldf-mode

trullock, 2021-08-06:
Ah right great thanks. Yeah I've read that. Sorry, I was under the impression you meant part of the docs referred to symlinks and how theyre handled

ivankravets, 2021-08-06:
We have another feature request https://github.com/platformio/platformio-core/issues/3348

In any case, we recommend using LDF.