platformio/platformio-core #4020: Call to configure_debug_options() gone

maxgerhardt, 2021-08-04:
What kind of issue is this?

- [X ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Win10 x64 

**PlatformIO Version** (`platformio --version`): `version 5.2.0b1`

### Description of problem

I am implementing a custom platform for GigaDevice microcontrollers (https://github.com/CommunityGD32Cores/platform-gd32) and am in the need of dynamically adding some stuff to the debug server configuration (specificially, `debug["tools"][link]["extra_cmds"`) for semihosting. The activation should be user-controllable in the `platformio.ini` through `board_debug.semihosting = yes`. To implement this dynamic option I wanted to use the `configure_debug_options()` function of the platform to set these options.

The old implementation did

https://github.com/platformio/platformio-core/blob/b8cc867ba46395bf144eb5e313ce6f1305706ab4/platformio/commands/debug/command.py#L116-L122

The [current implementation](https://github.com/platformio/platformio-core/blob/3be0f58c30ffb848dda24721872e8957ff91bf76/platformio/commands/debug.py#L36-L167) does not seem to call into `platform.configure_debug_options()` anymore at all. This entire repository now has 0 references to `configure_debug_options` when searching.

This is visible when when adding a simple `print` statement at the start of configure_debug_options():

```python
    def configure_debug_options(self, initial_debug_options, ide_data):
        print("CONFIGURE DEBUG OPTIONS CALLED!!")
```

This code can e.g. be also be put in `platform-ststm32` [here](https://github.com/platformio/platform-ststm32/blob/c44b39e48810069f8ad7846a96520099cab81b7e/platform.py#L188-L190), just to demonstrate that this doesn't work anymore.

Given the `platformio.ini`

```ini
[env:genericSTM32F103C8]
platform = ststm32
board = genericSTM32F103C8
framework = arduino
debug_speed = 500
debug_tool = cmsis-dap
```

and executing `pio debug --interface=gdb -x .pioinit` also shows in the, in my case, `.platformio\.cache\.piodebug-4adqg4qc\.pioinit` file that no `-c "adapter speed 500"` has been added into the invoke command

>target extended-remote | "C:/Users/Max/.platformio/packages/tool-openocd/bin/openocd.exe" -s "C:/Users/Max/.platformio/packages/tool-openocd" -c "gdb_port pipe; tcl_port disabled; telnet_port disabled" -s "C:/Users/Max/.platformio/packages/tool-openocd/scripts" -f "interface/cmsis-dap.cfg" -c "transport select swd" -f "target/stm32f1x.cfg"

So the entire `debug_speed` option setting should be broken as of now (which for ESP32 people proved very important in the past). Due that I think it's rather urgent for investigation and I'm tagging @valeros for that.

(Print statements in the `_add_default_debug_tools()` function called by `get_boards()` work as normal and are visible)

#### Steps to Reproduce

1. Add print call into `configure_debug_options` implementation of a platform (either custom or existing)
2. Start debuggging (`pio debug --interface=gdb -x .pioinit`)
3. Observe output regarding added print statement

### Actual Results

No print output.

### Expected Results

Print output because the function has been called.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:genericSTM32F103C8]
platform = ststm32
board = genericSTM32F103C8
framework = arduino
debug_speed = 500
debug_tool = cmsis-dap
```

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>
void setup(){}
void loop(){}
```

### Additional info

maxgerhardt, 2021-08-08:
I'm also pretty sure that this issue is the cause of the issue described in https://community.platformio.org/t/cannot-debug-pi-pico-with-cmsis-dap-cmsis-dap-command-cmd-dap-swj-clock-failed/21300?u=maxgerhardt.

maxgerhardt, 2021-09-04:
I think this is a really important regression and should be looked into -- it's also blocking some of our development efforts at the [GD32 platform](https://github.com/CommunityGD32Cores/platform-gd32) (can't dynamically add flags to the GDB commands without this). @ivankravets?

ivankravets, 2021-09-04:
Sorry for the late response. Please use `configure_debug_session` for the further development. We plan to release PlatformIO Core 5.2 on the 13 of September. A temporary workaround is to use both methods. See https://github.com/platformio/platform-espressif32/blob/develop/platform.py#L167

ivankravets, 2021-09-16:
PlatformIO Core 5.2 is out. Please use new API.

closed 2021-09-16 by ivankravets