platformio/platformio-core #4023: Blackmagic Probe Autodetection Picks UART macOS

glingy, 2021-08-06:
### Configuration

**Operating system**: macOS Catalina 10.15.7

**PlatformIO Version** (`platformio --version`): PlatformIO, version 5.0.3

### Description of problem
MacOS auto-detects incorrect serial port for black magic probe. On macOS, the serial ports are labeled by the descriptor index: `/dev/cu.usbmodemXXXXXXX#` where XXXXXX is the serial port identifier provided by the usb device and # is the interface descriptor id. I do not know if there is a better way of distinguishing these, my computer shows `None` for `serial.tools.list_parts.ListPartInfo.interface`, but my computer does recognize the interface name:
[image]. Adding a condition for macOS to filter by the interface descriptor id seems to be more reliable than searching for "GDB" in the usb description which will always be the same for both ports. For context, printing `item` on line 140 of `builder/tools/pioupload.py` shows:
```
{'port': '/dev/cu.usbmodem5596E0E13', 'description': 'Black Magic Probe (STLINK/V2) v1.7.1-194-g8089e05-dirty', 'hwid': 'USB VID:PID=1D50:6018 LOCATION=20-2'}
{'port': '/dev/cu.usbmodem5596E0E11', 'description': 'Black Magic Probe (STLINK/V2) v1.7.1-194-g8089e05-dirty', 'hwid': 'USB VID:PID=1D50:6018 LOCATION=20-2'}
```
So identifying by usb description is not working. The usb descriptor id is a constant in the blackmagic probe firmware, and should be a good option to filter on for auto-detection unless the descriptors change in a new release of that firmware.

#### Steps to Reproduce

1. Plug in a BMP (I'm using a blue pill clone, but should be the same with a real one)
2. Create any new project with the blackmagic debugger and attempt to download using it without specifying `upload_port`
3. It hangs on upload because it's trying to use the uart port for gdb.
4. Specify the upload_port, and it works

### Additional info
I'm working on making a PR and testing linux because I'm curious if there's a difference there. According to the pyserial docs: https://pyserial.readthedocs.io/en/latest/tools.html#serial.tools.list_ports.ListPortInfo it should always give the usb descriptor, not the individual interface descriptors for that "description" field in which case GDB filtering may have been broken on all non-windows platforms for a while.

glingy, 2021-08-06:
Potential fix for macOS if this is reproducible and not just my computer:
line 140 of builder/tools/pioupload.py
```
if IS_MACOS and port.startswith("/dev/cu.usbmodem") and port.endswith("1"):
    return port
```

glingy, 2021-08-06:
After testing on linux, ubuntu 20.04.2 returns "Black Magic Probe (STLINK/V2) v1.7.1-194-g8089e05-dirty - Black Magic UART Port" and "Black Magic Probe (STLINK/V2) v1.7.1-194-g8089e05-dirty - Black Magic GDB Port" for the description field, so filtering by "GDB" seems to work fine there.

glingy, 2021-08-09:
Note: I was able to test and received the same issue on Windows where it consistently picked COM4, but GDB was COM3. I'm not super familiar with Windows COM port ordering, but it seems to also be related to the usb interface descriptor number in some way. I'm not sure this is a great solution, but filtering by one digit in the hwid line (0 or 2) or guessing the first COM port instead of the second might be better options. I'm going to keep testing and make a fork with what I find

ivankravets, 2021-09-13:
Does it work if you specify `debug_port` manually using https://docs.platformio.org/en/latest/projectconf/section_env_debug.html#debug-port ?

You can also manually control `upload_port` https://docs.platformio.org/en/latest/projectconf/section_env_upload.html#upload-port

jcw, 2021-09-13:
FWIW, I've always had to specify the exact ports, and then things work fine.
And yes, the GDB is always the one ending in "1" and the console port always ends in "3" (by design).
Both `upload` and `monitor` could benefit (when there is only 1 BMP attached) if they knew about this convention.

ivankravets, 2022-06-17:
Thanks for the report! Could you re-test with `pio upgrade --dev`? Does it find a port properly?

closed 2022-06-17 by ivankravets

jcw, 2022-06-17:
On MacOS, I can now omit `monitor_port` (terrific!), but I still have to specify `upload_port`:

```
$ pio run -t upload -e jnzero-echo
Processing jnzero-echo (board: nucleo_l053r8; platform: ststm32; framework: cmsis)
[...]
AVAILABLE: blackmagic, cmsis-dap, jlink, mbed, stlink
CURRENT: upload_protocol = blackmagic
Looking for BlackMagic port...
Auto-detected: /dev/cu.usbmodemDFE3C8B13
Uploading .pio/build/jnzero-echo/firmware.elf
[...]
```

As you can see, it picked the `xxx3` port for uploading, which then hangs.

ivankravets, 2022-06-18:
Fixed. Could you re-run `pio upgrade --dev`. Does it work well for uploading, debugging, and device monitoring?

jcw, 2022-06-18:
Yep, latest now works for upload **and** monitor - i.e. port specs not needed when a single BMP is present \o/
(I'm not set up for debugging, can't check that right now)

commit b72c1636 (2022-06-17): Improved a serial port finder for Black Magic Probe // Resolve #4023