platformio/platformio-core #4028: Add support for renesas MCU board **R7FA2L1AB2DFM**

babanazar, 2021-08-09:
What kind of issue is this?

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

------------------------------------------------------------------

Add support for renesas MCU board **R7FA2L1AB2DFM**

Charragh, 2022-02-17:
No Renesas boards are supported in PlatformIO yet. I hope Renesas boards will be supported in the future. 

I had a discussion about this in the community: https://community.platformio.org/t/platfromio-doesnt-support-renesas-platforms-yet/26096

ivankravets, 2022-02-17:
Please vote (👍) for the issue using a "smile" icon in the comment above.

Charragh, 2022-02-21:
done

ivankravets, 2022-02-21:
Sorry, I meant thumb up 👍 (icon) 😂

Charragh, 2022-02-21:
done

Kubson320, 2022-09-18:
Is it planned to add support for DA14683 and other chips?

briandilley, 2023-01-19:
I'm wondering how feasible this is given smart snippets.

jayemar, 2023-02-09:
DA14699 would also be helpful in order to support the DA14699-00HRDB-P development board.

runger1101001, 2023-06-29:
And the UNO R4 is now out :-) The UNO R4 WiFi looks like a pretty cool board actually. The R4 Minima is also a big upgrade on the R3 in terms of the MCU.

The C33 uses a R7FA6M5BH2CBG (Cortex M33)
The UNO R4s use the R7FA4M1AB3CFM (Cortex M4)

All are already supported in Arduino IDE, so have working arm-gcc toolchains.

What needs to be done to get them supported? Is it on the roadmap?

lmolter54, 2023-06-30:
FWIW, even the Arduino 2.1.0 IDE can't upload to the board yet. Go figure. You have to use the antiquated 1.8.13 IDE. Not me. I'll wait either for PIO or 2.1.x of their IDE.

LYB030, 2023-07-04:
> FWIW, even the Arduino 2.1.0 IDE can't upload to the board yet. Go figure. You have to use the antiquated 1.8.13 IDE. Not me. I'll wait either for PIO or 2.1.x of their IDE.

the ide 2.1.1 has solve the problem

maxgerhardt, 2023-07-04:
I got my Arduino Uno R4 Minima and Uno R4 WiFi today so I used 5 hours of my time to create the PlatformIO integration for that: https://github.com/maxgerhardt/platform-renesas

You can install it per the "Usage" instructions in the readme and then create new projects for the R4 Minima, R4 WiFi and Portenta C33 in the regular "New Project" wizard. Example projects for each board [are provided too](https://github.com/maxgerhardt/platform-renesas/tree/main/examples).

[image]
[image]

Debugging is **not yet possible**. Uploading works.

I'm happy to hand this over to @ivankravets after it has been developed properly.

runger1101001, 2023-07-04:
Wow, thanks so much @maxgerhardt that's great news!

Your efforts are really greatly appreciated! I'm going to try to switch my UNO R4 project over to PlatformIO tomorrow.

runger1101001, 2023-07-05:
I couldn't wait until tomorrow. It's working for me on OS X. :-)

vshymanskyy, 2023-07-05:
@maxgerhardt it works! 
I was able to use it for all 3 boards for `Blynk.NCP`: https://github.com/blynkkk/BlynkNcpExample

[image]

Thanks for your time and effort

ivankravets, 2023-07-05:
@maxgerhardt, thanks for the MVP! I've just created a new repository https://github.com/platformio/platform-renesas-ra and we will be back soon with updates.

LYB030, 2023-07-05:
> I got my Arduino Uno R4 Minima and Uno R4 WiFi today so I used 5 hours of my time to create the PlatformIO integration for that: https://github.com/maxgerhardt/platform-renesas
> 
> You can install it per the "Usage" instructions in the readme and then create new projects for the R4 Minima, R4 WiFi and Portenta C33 in the regular "New Project" wizard. Device drivers [are here](https://github.com/arduino/ArduinoCore-renesas/tree/1.0.1/drivers). Example projects for each board [are provided too](https://github.com/maxgerhardt/platform-renesas/tree/main/examples).
> 
> [image] [image]
> 
> Debugging is **not yet possible**. Uploading works.
> 
> I'm happy to hand this over to @ivankravets after it has been developed properly.

[image]
got error show by vscode,but upload and build all works fine

maxgerhardt, 2023-07-05:
Let's discuss that in https://github.com/maxgerhardt/platform-renesas/issues/1 and let's avoid long quote replies.

maxgerhardt, 2023-07-05:
Update: I figured out that the Uno R4 WiFi's ESP32-S3 bridge / coprocessor is not only a USB-Serial bridge for the Renesas bridge, but also has CMSIS-DAP functionality compiled in. Thus, we can do debugging with on-board tools!

[image]

vshymanskyy, 2023-07-05:
So. Looks like we'll have to provide a custom build of `Blynk.NCP` for this... At some point :rofl:

valeros, 2023-07-07:
Hi guys, we've added an initial support for the Renesas hardware here https://github.com/platformio/platform-renesas-ra. Currently, there are only several boards available though the Arduino framework. Please keep in mind we couldn't test it thoroughly as we don't have any of this hardware at hand, so your feedback is very welcome. 

Many thanks @maxgerhardt for the initial research and hints!

vshymanskyy, 2023-07-09:
I was able to port the CDC Bridge and DAP functionality to Blynk.NCP.
Looks working fine so far, including the debugger!
This will require a separate NCP build/variant for UNO R4 WiFi, but I’m ok with that (it was needed anyway)

maxgerhardt, 2023-07-10:
After some [work](https://github.com/platformio/platform-renesas-ra/issues?q=is%3Aissue+), the official platform-renesas version 0.0.2 version now has all functionality and bugfixes that my original platform has, so now there's no point in using my version anymore.  I've **renamed and archived** my original repo. Please use `platform = renesas-ra@0.0.2` as the stable official platform instead :)

Thanks to the PlatformIO team, @vshymanskyy  and @LYB030 for developing and testing the platform, now we have something stable to work from!

vshymanskyy, 2023-07-13:
To those interested in using both ESP32 and Renesas chips:
https://github.com/vshymanskyy/UNO-R4-WiFi-freedom

elral, 2023-07-17:
We have a project which also uses servos.
Porting from Mega/Uno to the new Uno R4 was mostly successful, but it seems that the servo library is missing.

Additionally strlcpy(...) is missing compared to Mega/Uno/Nano, but this is no big issue as I can add it.

vshymanskyy, 2023-07-17:
Please report such kind of bugs directly to Arduino

maxgerhardt, 2023-07-17:
Servo library **does** support Renesas and is right here: https://github.com/arduino-libraries/Servo/releases and PlatformIO also has it as regular `lib_deps = arduino-libraries/Servo@1.2.1`.
Missing `strlcpy` is either a toolchain bug or a configuration issue (e.g., may need to define `_GNU_SOURCE`, `_BSD_SOURCE` or whatveer.

elral, 2023-07-17:
> Servo library **does** support Renesas and is right here: https://github.com/arduino-libraries/Servo/releases and PlatformIO also has it as regular `lib_deps = arduino-libraries/Servo@1.2.1`.

Ohmg, my simple fault :(

> Missing strlcpy is either a toolchain bug or a configuration issue (e.g., may need to define _GNU_SOURCE, _BSD_SOURCE, ..)

Hmhm, seems I have to look around.

Thanks for help!

maxgerhardt, 2023-07-17:
Yeah, a short exploration into `string.h` shows

[image]

So adding to the `platformio.ini`
```ini
build_flags = -D_BSD_SOURCE
```
enables the declaration of that and it links fine.

MuhdZamwani, 2023-09-06:
> > FWIW, even the Arduino 2.1.0 IDE can't upload to the board yet. Go figure. You have to use the antiquated 1.8.13 IDE. Not me. I'll wait either for PIO or 2.1.x of their IDE.
> 
> the ide 2.1.1 has solve the problem

I have same problem, can't upload my code to board even com was detected and I use platformio in vscode. Do i have to use arduino ide instead of platformio?

maxgerhardt, 2023-09-06:
Did you install [the drivers](https://github.com/arduino/ArduinoCore-renesas/tree/main/drivers)?

MuhdZamwani, 2023-09-07:
> Did you install [the drivers](https://github.com/arduino/ArduinoCore-renesas/tree/main/drivers)?

Not yet. How I can install it into my platformio?

maxgerhardt, 2023-09-07:
You don't install them into *PlatformIO*, you install them in your operating system. Those are Windows drivers, after all. You could e.g. install them by downloading the entire repo as a ZIP file (https://github.com/arduino/ArduinoCore-renesas/archive/refs/heads/main.zip), unpacking it and executing the `drivers/dpinst-amd64.exe`.

It's also handy that you can boot your board into bootloader mode by quickly pressing the reset button twice on it. After the driver installation, there should be no "unknown devices" in the Windows device manager, neither in bootloader mode nor in regular mode.

MuhdZamwani, 2023-09-07:
> 

> You don't install them into _PlatformIO_, you install them in your operating system. Those are Windows drivers, after all. You could e.g. install them by downloading the entire repo as a ZIP file [...]

I done install the driver and press reset on the board, but I still get no device found on COM3. I done uninstall and install arduino 2.2.1

maxgerhardt, 2023-09-07:
Boot the device into bootloader mode. Are there multiple COM ports in the device manager? Maybe it just picks the wrong one of two.

MuhdZamwani, 2023-09-07:
Its already work after I boot into bootloader mode. But I still dont get it, why could happen like that after I upload my code that have SSL certificate? Can you explain to me please? For future reference.