platformio/platformio-core #4033: Add support for multiple ST-Link adapters

jcw, 2021-08-18:
- [x] **Feature Request**.
      I've set up a Raspberry Pi as build farm for STM32 µC boards. The problem is that the ST-Link interfaces on these boards all look the same, apart from the serial number, exposed in the USB descriptors. For OpenOCD, which is the default upload tool, an extra parameter needs to be specified (`hla_serial <serialnum>`).

Attached is a `platformio.ini` file which properly uploads and connect to the matching serial port. It works, but it's messy, and I need to specify the board type again, even though PIO already has that info in its board definition file. Also included is a `main.cpp`, to test the builds / uploads (which will obviously fail at some point, without these exact same boards present).

[blink.zip](https://github.com/platformio/platformio-core/files/7007132/blink.zip)

It would be useful to streamline this all a bit more, perhaps with a new `upload_serial` config parameter. For the serial connection, a second simplification would be to use this same serial number to match the proper port (in `minicom`?) when `monitor_port` is not set. That way, all you'd need to pick a _specific_ ST-Link would be to add `upload serial = ...`.

(see also [my forum post](https://community.platformio.org/t/choosing-stlink-v2-programmer/10716/4) on this topic)

ivankravets, 2021-09-08:
Yep, known issue. We still have found the best solution for it. `hla_*` has multiple options -> https://openocd.org/doc/html/Debug-Adapter-Configuration.html
The same story with other on-chip debuggers, J-Link, etc. 

Nevertheles, PlatformIO allows you to override variables in runtime. So, let use [dynamic variables](https://docs.platformio.org/en/latest/projectconf/dynamic_variables.html) and [post script](https://docs.platformio.org/en/latest/projectconf/advanced_scripting.html):

`platformio.ini`
```ini
[env]
platform = ststm32
framework = cmsis
monitor_speed = 115200
lib_deps = jeeh
extra_scripts = post:extra_script.py

[env:f103]
board = nucleo_f103rb
build_flags = -Wno-format -DSTM32F1
custom_hla_serial = 066DFF525257775087154313
monitor_port = /dev/serial/by-id/usb-STMicroelectronics_STM32_STLink_${env:f103.custom_hla_serial}-if02

```

`extra_script.py`
```python
Import("env")

f_index = env.get("UPLOADERFLAGS", []).index("-f")
hla_serial = env.GetProjectOption("custom_hla_serial", None)
if f_index > 0 and hla_serial:
    env["UPLOADERFLAGS"].insert(f_index, ["-c", f"hla_serial {hla_serial}"])

```

Should we add an example to https://docs.platformio.org/en/latest/projectconf/advanced_scripting.html?

maxgerhardt, 2021-09-29:
>Should we add an example to https://docs.platformio.org/en/latest/projectconf/advanced_scripting.html?

Would it make sense to integrate this into the core? E.g., if `upload_protocol = stlink` and `upload_port` is set explicitly and matches the serial number format, automatically inject the `-c hla_serial <serial>` command. Or instead of `upload_port`, support a `hla_serial = ..` option by standard.