platformio/platformio-core #4038: LPC1769 missing from the PlatformIO board list

AZWoody, 2021-08-23:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:

**PlatformIO Version** (`platformio --version`):

### Description of problem

#### Steps to Reproduce

1.
2.
3.

### Actual Results

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
Insert here...
```

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info

AZWoody, 2021-08-23:
I had to reload studio and when I tried to set the system board in Platform IO the LPC1769  board is missing from the list. Only the LPC17698 is listed. I don't know who to contact to get it put back in so I can setup Marlin 2.0 on my Big Tree Tech SKR1.4 Turbo board. Please help me!

valeros, 2021-08-23:
Hi @AZWoody ! This repository is not the right please to ask such questions. Anyway, it seems that the `nxplpc` dev-platform got corrupted, so you can try to remove the folder where all PlatformIO dev-platforms are installed (`c:\Users\YOUR_USERNAME\.platformio\platforms` (Windows) or `/home/YOUR_USERNAME/.platformio/platforms` (Unix)) and try to compile your project again (might take some time).

Next time please submit your issue directly to the Marlin repository https://github.com/MarlinFirmware/Marlin or PlatformIO Community https://community.platformio.org/

closed 2021-08-23 by valeros