platformio/platformio-core #4041: Package `tool-clangtidy` missing dependencies on latest linuxes

positron96, 2021-08-30:
- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Ubuntu 20.04 x64

**PlatformIO Version** (`platformio --version`): `PlatformIO Core, version 5.1.1`

### Description of problem

Most recent `clang-tidy` from package `tool-clangtidy` 1.100000.0 depends on `libtinfo.so.5`. However, my Ubuntu 20.04 as well as [Debian Buster, Manjaro](https://stackoverflow.com/questions/48674104/clang-error-while-loading-shared-libraries-libtinfo-so-5-cannot-open-shared-o) and [Arch](https://community.platformio.org/t/clang-tidy-error-on-arch-linux-libtinfo-so-5/18641) have `libtinfo.so.6` installed.

#### Steps to Reproduce

1. Have Ubuntu 20.04 (or more recent). 
2. run a `pio check` with clang-tidy analyzer.

### Actual Results

Analysis fails:
```
Checking stm32 > clangtidy (platform: ststm32; board: bluepill_f103c8; framework: arduino)
-----------------------------------------------------------------------------------------------------------------

Error: Failed to execute check command! Exited with code 127.
Error: clangtidy failed to perform check! Please examine tool output in verbose mode.
========================================== [FAILED] Took 0.03 seconds ==========================================

Environment    Tool       Status    Duration
-------------  ---------  --------  ------------
stm32          clangtidy  FAILED    00:00:00.029
===================================== 1 failed, 0 succeeded in 00:00:00.029 =====================================
```

Running with `--verbose` as  suggested does not give any more information.

### Expected Results

clang analyzes the code

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:** any config with `check_tool=clangtidy`
```ini
[platformio]
default_envs = stm32

[env]
check_tool = clangtidy

[env:stm32]
platform = ststm32
board = bluepill_f103c8
framework = arduino
```

**Source file to reproduce issue:** any will do. Here is an empty one
```cpp
void setup() {}
void loop() {}
```

### Additional info

Interestingly, it fails silently in 5.0.1 (result says "No defects found").

https://community.platformio.org/t/clang-tidy-error-on-arch-linux-libtinfo-so-5/18641

https://github.com/microsoft/vscode-cpptools/issues/4664

https://github.com/microsoft/vscode-cpptools/issues/3174

valeros, 2021-09-10:
Hi @positron96 ! We've updated the `tool-clangtidy` package to the latest v12.0.1, please retest your project with the development version of PlatformIO. Just update your PlatformIO installation via the  `pio upgrade --dev` command in your IDE terminal. Thanks!

closed 2021-09-10 by valeros

positron96, 2021-09-10:
Hi, thanks! Will test in next few days!

positron96, 2021-09-21:
Yep, seems to be working. Though I did not have to install dev version, pio 5.2.0 now installs `tool-clangtidy @ 1.120001.0`

commit 63a2465b (2021-09-10): Update check tools to the latest available // Resolve #4041