platformio/platformio-core #4044: arduino-derived custom frameworks have error "Multiple ways to build the same target were specified"

jonbitzen, 2021-09-04:
- [ ] **Feature Request**.
      Change the frameworks configuration to allow a generic flag which allows arduino-derived custom frameworks to compile successfully without generating the error "Multiple ways to build the same target were specified".

I am using a modified arduino framework which allows an Arduino Leonardo based board to present itself as an Xbox 360 USB controller. 

In the platforms manifest file for atmelavr (~/.platformio/platforms/atmelavr/platform.json) the new platform was added as "xinput" and correctly associated with the new arduino-derived framework package "framework-xinput-avr".

After doing this, and attempting to create a project using a custom board (also set up separately and recognized by the IDE) and framework, I got a framework build error "Multiple ways to build the same target were specified" as below:

```
PIO Core Call Error: "The current working directory /home/dev_user/Documents/PlatformIO/Projects/xinput-test1 will be used for the project.

The next files/directories have been created in /home/dev_user/Documents/PlatformIO/Projects/xinput-test1
include - Put project header files here
lib - Put here project specific (private) libraries
src - Put project source files here
platformio.ini - Project Configuration File

Error: Processing leonardo-xinput (platform: atmelavr; board: leonardo-xinput; framework: xinput)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/leonardo-xinput.html
PLATFORM: Atmel AVR (3.3.0) > Arduino Leonardo with Xinput
HARDWARE: ATMEGA32U4 16MHz, 2.50KB RAM, 28KB Flash
DEBUG: Current (simavr) On-board (simavr)
PACKAGES: 
 - framework-xinput-avr 1.0.2 
 - toolchain-atmelavr 1.70300.191015 (7.3.0)

*** Multiple ways to build the same target were specified for: 
/home/dev_user/Documents/PlatformIO/Projects/xinput-test1/.pio/build/leonardo-xinput/FrameworkArduino/wiring_pulse.o 

(from 
['/home/dev_user/.platformio/packages/framework-xinput-avr/cores/arduino/wiring_pulse.S'] 
and from 
['/home/dev_user/.platformio/packages/framework-xinput-avr/cores/arduino/wiring_pulse.c']
)

File \"/home/dev_user/.platformio/penv/lib/python3.8/site-packages/platformio/builder/tools/platformio.py\", line 350, in BuildLibrary
========================== [FAILED] Took 0.25 seconds =========================="
```

However, this issue is handled correctly for the unmodified arduino framework that comes with platformio.

I found that in the platformio build system, the file "~/.platformio/penv/lib/python3.8/site-packages/platformio/builder/tools/platformio.py" contains a hard-coded check on the framework name to see if it is arduino, and then engages special handling 
```
def BuildFrameworks(env, frameworks):
    if not frameworks:
        return

    if "BOARD" not in env:
        sys.stderr.write(
            "Please specify `board` in `platformio.ini` to use "
            "with '%s' framework\n" % ", ".join(frameworks)
        )
        env.Exit(1)

    board_frameworks = env.BoardConfig().get("frameworks", [])
    if frameworks == ["platformio"]:
        if board_frameworks:
            frameworks.insert(0, board_frameworks[0])
        else:
            sys.stderr.write("Error: Please specify `board` in `platformio.ini`\n")
            env.Exit(1)

    for f in frameworks:
        if f == "arduino":
            # Arduino IDE appends .o the end of filename
            Builder.match_splitext = scons_patched_match_splitext
            if "nobuild" not in COMMAND_LINE_TARGETS:
                env.ConvertInoToCpp()

        if f in board_frameworks:
            SConscript(env.GetFrameworkScript(f), exports="env")
        else:
            sys.stderr.write("Error: This board doesn't support %s framework!\n" % f)
            env.Exit(1)
```
Changing if if f == "arduino" line to if f == "arduino" or f == "xinput": lets PlatformIO in VSCode create the project normally.

It'd be good if there was a generalized convention in either the framework-name (e.g., require arduino-derived frameworks to include "arduino" in the framework name, and then check for the substring) or some other framework configuration to detect this specialized build generation use-case, so that the platformio build files don't have to edited to compile arduino-derived frameworks.

ivankravets, 2021-09-04:
Hi,

Thanks for using PlatformIO. Quick answers:

1) Please use the same framework name = arduino, if you use the same API and syntax is the same.
2) If you would like to use own framework name, avoid naming source files with the same name but different extensions:
- wiring_pulse_asm.S
- wiring_pulse.c

closed 2021-09-04 by ivankravets

maxgerhardt, 2024-10-28:
Running into this too with `rv003usb.S` and `rv003usb.c` being in the same folder (<https://github.com/cnlohr/rv003usb/>). 

```
*** Multiple ways to build the same target were specified for: 
C:\repos\ch32v003fun\.pio\build\usb_composite\src\examples\usb_composite\rv003usb\rv003usb.o 
 (from ['C:\\repos\\ch32v003fun\\examples\\usb_composite\\rv003usb\\rv003usb.S'] 
and from ['C:\\repos\\ch32v003fun\\examples\\usb_composite\\rv003usb\\rv003usb.c'])
```
Would it not be better to change the core to allow such file names, without having to rename the source files?