platformio/platformio-core #4046: Add support for BouffaloLab BL602/BL604

normanr, 2021-09-06:
What kind of issue is this?

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

------------------------------------------------------------------

Please add support for BL602/BL604 made by BouffaloLab

> The BL602 is a general purpose microcontroller based on the “SiFive E24 Core” RISC-V processor.

Notice that it's only _based on_ the SiFive E24 Core, so I don't think the existing SiFive platform will work for this chip.

There's a lot of documentation and SDK's avaiable:
- https://www.bouffalolab.com/bl602
- https://github.com/bouffalolab (SDK's, NuttX fork, flashing tools)
- en docs: https://pine64.github.io/bl602-docs/
- [BL602 vs Other RISC-V Microcontrollers](https://lupyuen.github.io/articles/pinecone#bl602-vs-other-risc-v-microcontrollers) (includes a list of 3 example boards)

There is a "[Nutcracker challenge](https://www.pine64.org/2020/10/28/nutcracker-challenge-blob-free-wifi-ble/)" to re-implement the Wifi and BLE blobs as open source, but I hope that the official SDKs (with the closed blobs) will be sufficient to get it working with PlatformIO in the meantime.

normanr, 2021-09-06:
[pine64/ArduinoCore-bouffalo](https://github.com/pine64/ArduinoCore-bouffalo) looks like the start of an ArduinoCore Core for BL602 (found via [mkroman/awesome-bouffalo](https://github.com/mkroman/awesome-bouffalo))

maxgerhardt, 2021-09-13:
Funny, I thought about a possible PlatformIO integration for these types of boards too after they were [featured in a YouTube video](https://youtu.be/-bYa3F1sl6o?t=1310) yesterday. I think these boards will gain more attention through that soon.

But I think this issue should be moved to https://github.com/platformio/platform-sifive, the existing PlatformIO integration for SiFive boards. In fact it shouldn't be too hard to add a simple board definition + builder script for Arduino to that platform. I'll give it a try soon..

maxgerhardt, 2021-09-13:
I've completed writing the PlatformIO support for the pine64 Boufallo core, which supports its BL602-based Pinecone board. The core isn't really specific to that board though, any BL602 board should work.

@normanr do you have a board and can test https://github.com/maxgerhardt/pio-bl602-boufallo-arduino-test?

normanr, 2021-09-14:
I have a MagicHome LED controller with the BL602 chip, but I don't have anything set up to flash it at the moment. (The FTDI cable I used to flash the ESP8285 based MagicHome I bought a year ago has since been used to interface with an air quality sensor). I'd like to find some time in the next couple of weeks to try it out.

normanr, 2021-10-31:
I got around to testing it today and initially it couldn't find `eflash_loader_40m.bin`. I updated the SiFive platform to pick up the `get_contrib_path` fix from ngkz, and it flashed onto my MagicHome LED controller without any issues.

One thing I noticed was that the ROM and RAM numbers listed in Platform IO don't match the docs at https://wiki.pine64.org/wiki/PineCone. I'm not sure how much that matters.

Now to try and see if ESPHome or Tasmota will build! (or how badly they don't)

maxgerhardt, 2021-10-31:
>I got around to testing it today and initially it couldn't find eflash_loader_40m.bin. I updated the SiFive platform to pick up the get_contrib_path fix from ngkz, and it flashed onto my MagicHome LED controller without any issues.
I got a PR for that some time ago and merged it already, in https://github.com/pine64/ArduinoCore-bouffalo/pull/6#issuecomment-930448270 I already tested it to be working.

You might want to delete `<home folder>/.platformio/packages/tool-bl60x-flash` to get the latest version upon the next upload and teh problem should be solved.

normanr, 2021-11-01:
Yep I found that, and that's what inspired me to manually trigger the update of the sifive package.

One issue I found during the process of trying to get Tasmota to build: `LibraryPropertiesManifestParser._parse_platforms` in https://github.com/platformio/platformio-core/blob/develop/platformio/package/manifest/parser.py hard-codes the mapping for architecture to platform, and if there's no entry in the map then the architecture is _discarded_ (unlike `LibraryJsonManifestParser._parse_platforms` which falls back to adding the architecture as the platform).

The SPI and Wire libraries in ArduinoCore-bouffalo set architectures to riscv, so a mapping of `"riscv": "sifive"m` is required to allow them to be used. A bit strange but it does work.

Also I noticed that in https://github.com/maxgerhardt/platform-sifive/blob/develop/boards/pinecone.json, `mcu` is set to `sivife_e24`, not `sifive_e24`. I'm guessing that's a typo?

maxgerhardt, 2021-11-01:
>trying to get Tasmota to build

The Arduino core is in an extremely early stage, with not even the base Arduino-API implemented, lest any WiFi library etc. I'm sure it's impossible to build at this stage without developing the Arduino core further first.

>The SPI and Wire libraries in ArduinoCore-bouffalo set architectures to riscv, so a mapping of "riscv": "sifive", is required to allow them to be used. A bit strange but it does work.

Thanks, that's very interesting. The Arduino core might be just using the wrong architecture here, or I need to adapt the platform..

jsaiko, 2022-01-20:
I have interest in contributing to this project.  Just converted from arduino to platform IO.  Also, I typically code on linux and am having trouble navigating windows.  Could someone point me in the right direction on how to add "https://github.com/maxgerhardt/pio-bl602-boufallo-arduino-test.git" to my platform IO installation?

maxgerhardt, 2022-01-20:
Just `git clone` the project and then open the project in the VSCode [PIO Home page](https://docs.platformio.org/en/latest/home/index.html#welcome-project-manager) via "Open Project". Using the [project task](https://docs.platformio.org/en/latest/integration/ide/vscode.html#project-tasks) "Build" it should then start compiling.

jsaiko, 2022-01-20:
I was making it way harder than it needed to be haha.  Thanks, its working now.

jsaiko, 2022-01-21:
I am working on getting wifi up.  It looks like the bl_iot_sdk libraries are only partially compiled and linked.  I built the libraries from https://github.com/bouffalolab/bl_iot_sdk.git and am trying to link against those.  If I can get this to work, would the next steps be writing "wrappers" for the arduino framework classes?

normanr, 2022-01-21:
note that if you're using blflash instead of the Dev Cube uploader, then you may need to patch the bin file before uploading, see https://github.com/apache/incubator-nuttx/issues/4336 for more info.

jsaiko, 2022-01-21:
I'm stuck on a prebuilt static library, components/network/wifi/lib/libwifi.a  It was compiled using a different version and I am getting the following error when I attempt to link.  This might be why the additional static libraries were not included.  This is my first time messing with risc architecture though.  I'll look at it more tomorrow.

lib\libwifi.a(arch_main.o): conflicting priv spec version (major/minor/revision).

maxgerhardt, 2022-01-21:
The Arduino core is in an extremely primitive, early stage, in which the WiFi libraries are not yet linked in. The BL-IoT SDK package comes from [here](https://nutcracker.pine64.org/arduino/bl_iot_sdk_arduino/bl_iot_sdk_arduino_0.0.1.zip) as listed in their [board index](https://nutcracker.pine64.org/arduino/package_bouffalo_index.json), which might already contain mistakes leading to linker errors. I'll try to link in the library and / or add support for the native SDK. And have a look at that [other flasher tool](https://github.com/apache/incubator-nuttx/issues/4336#issuecomment-932998791).

maxgerhardt, 2022-01-21:
> I built the libraries from https://github.com/bouffalolab/bl_iot_sdk.git 

Oh well, actually that could very well be the source of the error then. If you're taking the latest master branch of that, it may not work, because the Arduino core is based on this older SDK version, as linked above. Try compiling them from https://github.com/maxgerhardt/pio-bl-iot-sdk-arduino.

jsaiko, 2022-01-21:
> Well.. it seems the better way is to cleanly update the SDK pacakge for the Arduino core to the newest git version _and_ include in the WiFi libraries.

It looks like the newest version of the SDK from the bouffalolab git has an updated toolchain which is causing their precompiled wifi libraries to be ABI incompatible with binaries built using the sifive toolchain.  Updating platformio to use the toolchain thats included in https://github.com/bouffalolab/bl_iot_sdk.git should in theory work.  I'm a platformio noob though so I have lots of reading to do.

maxgerhardt, 2022-01-21:
Oh yes, if you want PlatformIO to use the same toolchain, you need to make the compiler available to PlatformIO as a "package". You can read up on some of it [here](https://github.com/bouffalolab/bl_iot_sdk/tree/master/toolchain/riscv). In this case, you would have to upload the toolchain for a specific OS (e.g., [this Windows toolchain here](https://github.com/bouffalolab/bl_iot_sdk/tree/master/toolchain/riscv/MSYS) in a repo, and add `package.json` that properly declares the name and version of the package. 

```json
{
  "name": "toolchain-riscv",
  "version": "1.100200.211109",
  "description": "GNU toolchain for RISC-V, including GCC",
  "keywords": [
    "build tools",
    "compiler",
    "assembler",
    "linker",
    "preprocessor",
    "risc-v"
  ],
  "license": "GPL-2.0-or-later",
  "system": [
    "windws_amd64"
  ],
  "repository": {
    "type": "git",
    "url": "https://github.com/riscv/riscv-gnu-toolchain"
  }
}
```

Then use [`platform_packages`](https://docs.platformio.org/en/latest/projectconf/section_env_platform.html#platform-packages) to override the source for the compiler in the `platformio.ini`.

```
platform_packages = 
   toolchain-riscv@<repo link>
```

jsaiko, 2022-01-21:
I created packages for the 10.2.0 RISC-V toolchain and I am no longer getting those compiling errors.  It's unfortunately not going to be as easy as just replacing the precompiled libs in "framework-bl-iot-sdk-arduino".  They were modified from the original bouffalo code and some of the functions were moved to "framework-arduinobouffalo" package code.  

libwifi.a is closed source.  We need the original libwifi.a that was distributed with that version of the sdk.  Without that, we are going to have to use the newer version from the bouffalo sdk.  Using the newer version is going to require a rework of much of the things folks have done here.  What probably should be done is to take the bits from the bouffalo sdk along with libwifi, the work that the pine64 folks did, and create a single new Arduino framework package.  This is a lot of work and may be out of my technical abilities.  Also, the pine64 folks are reverse engineering the closed source blobs which could make using libwifi irrelevant.

Anyhow, here are the repos in case anyone wants to use the updated toolchain.

`
platform_packages =
   toolchain-riscv@https://github.com/jsaiko/toolchain-riscv-10.2.0-win64.git
   toolchain-riscv@https://github.com/jsaiko/toolchain-riscv-10.2.0-linux64.git
`

maxgerhardt, 2022-01-22:
>This is a lot of work and may be out of my technical abilities. Also, the pine64 folks are reverse engineering the closed source blobs which could make using libwifi irrelevant.

This RE work is likely not going to be completed anytime soon because of its complexity. If we want to create an Arduino core for BL60x that people are wanting to use, it needs to have WiFi, and the easiest way is to link in the proprietary libray. When a free, open-source implementation of it becomes available, it can be easily switched out, especially if it's API compatible.

If you have a new, precompiled version of framework-bl-iot-sdk-arduino and the needed adaptions the for that SDK that the Arudino core uses, I'd be very happy to see that as a PR in https://github.com/maxgerhardt/ArduinoCore-bouffalo and https://github.com/maxgerhardt/pio-bl-iot-sdk-arduino. (Or at the Arduino upstream repo). Updating that and further implementing features in the Arduino core would already be a huge step forward.

jsaiko, 2022-01-22:
I started a [repo](https://github.com/jsaiko/framework-bouffalo-sdk.git) to build a framework package out of the bouffalo SDK.  I feel like a working implementation there would be a good place to start.  The sdk is surprising similar to the espidf at first glance.

maxgerhardt, 2022-01-23:
Thanks for the invite, and good work. It may make sense to create a Github organization like I have for the [GigaDevice Arduino core project](https://github.com/CommunityGD32Cores/) and create a discord server (or, something) for collaboration. We could definitely push the Arduino core implementation forward by a ton.

jsaiko, 2022-01-23:
That sounds like a great idea.  And I am pretty much glued to discord these days:  https://discord.gg/NT7fx6R2  Thank you

Lankaster, 2023-04-24:
Are there any progress reports on BouffaloLabs Chips Platformio support?

maxgerhardt, 2023-04-24:
At least I have not further worked on this due to lack of time and other projects. The repos are completely open for forking + development though.

DoomHammer, 2023-04-25:
How can one test it? Do I need to build my own PlatformIO extension from source to use this?

maxgerhardt, 2023-04-25:
No the VSCode extension stays constant. It's all handled dynamically by pointing PlatformIO to use a new platform url. Easiest is to just download https://github.com/Community-BL-IOT/pio-bl602-boufallo-arduino-test and open it in VSCode.

Opisek, 2024-04-02:
This discussion seems to be mostly about BL60x chips. Does anyone know how compatible these tests are with BL616?

kali-777, 2024-09-17:
I'm looking for that also I just bought 2x Ai Thinker AiPi-VOICE Ai-M61-32S Boards with speaker & mic too cheap to resist! 😁 
It was listed under ESP Modules, silly me I thought it's just some kind of ESP32 "variant" even GPIO PINs are SAME, but sadly its NOT?! 😣 
These are based on Bouffalo BL616/BL618 like a Sipeed M0sDock(BL616) and M0pDock(BL618) , however i really don't know ALL are using SAME T-HEAD CPU?! 🤔
BL602 series, Wi-Fi/BLE
BL702 series, BLE/Zigbee
BL606 series, Wi-Fi/BT/BLE/Zigbee/Ethernet/USB2.0 six-in-one
BL616/BL618 series, Wi-FI6/BT5/Thread 
BL808 series, 3 cores + 1 npu
T-HEAD C906 RV64GCV 480MHz (D0)
T-HEAD E907 RV32GCP 320MHz (M0)
T-HEAD E902 RV32EMC 160MHz (LP)

Here there is NO BL616/BL618 mentioned 🙄
[](https://github.com/sifive/freedom-u-sdk)

Now i'm "stuck" i don't want to download their SDK CLI or even 'worst' use Ai Thinker HTTP website?! 😨 
I would like to use VSCode IDE with Platformio just like any another ESP32 boards! SO please could You add support for these Chips or please clarify its ALL the "SAME" CPU?!  🤩  
Thanks