platformio/platformio-core #4051: Unable to upload remotely to a board

Renaud11232, 2021-09-12:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:

* Ubuntu 20.04 for the agent
* W10 21H1 for the other machine

**PlatformIO Version** (`platformio --version`):

5.1.1 on both machines. The Windows one is using PlatformIO IDE for VSCode

### Description of problem

Once I've successfully compiled a project I try to upload it remotely to my Arduino board but I get the following error in the IDE : `Error: Unknown environment names 'mega2560'. Valid names are 'include_tree'` and the project is not uploaded to the board.

#### Steps to Reproduce

On the agent machine : 

```bash
pio remote agent start
```

On my Windows machine : 

I usually just click "Build" then "Remote upload" but I get the same result when using the command line :

```
pio run -e mega2560
pio remote run -t upload -e mega2560
```

### Actual Results

The compilation succeeds then the upload fails : 
```
> Executing task: C:\Users\gaspa\.platformio\penv\Scripts\platformio.exe remote run --target upload --environment mega2560 <

Building project locally
Processing mega2560 (platform: atmelavr; board: megaatmega2560; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/megaatmega2560.html
PLATFORM: Atmel AVR (3.4.0) > Arduino Mega or Mega 2560 ATmega2560 (Mega 2560)
HARDWARE: ATMEGA2560 16MHz, 8KB RAM, 248KB Flash
DEBUG: Current (avr-stub) On-board (avr-stub, simavr)
PACKAGES:
 - framework-arduino-avr 5.1.0
 - toolchain-atmelavr 1.70300.191015 (7.3.0)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 5 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Wire> 1.0
|-- <SPI> 1.0
|-- <EEPROM> 2.0
|-- <SoftwareSerial> 1.0
Building in release mode
Checking size .pio\build\mega2560\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [===       ]  29.8% (used 2441 bytes from 8192 bytes)
Flash: [==        ]  21.0% (used 53370 bytes from 253952 bytes)
========================================================================================= [SUCCESS] Took 8.61 seconds =========================================================================================

Environment    Status    Duration
-------------  --------  ------------
mega2560       SUCCESS   00:00:08.611
========================================================================================= 1 succeeded in 00:00:08.611 ========================================================================================= 
Uploading firmware remotely
Error: Unknown environment names 'mega2560'. Valid names are 'include_tree'
Arrêt du processus de terminal "C:\Users\gaspa\.platformio\penv\Scripts\platformio.exe 'remote', 'run', '--target', 'upload', '--environment', 'mega2560'". Code de sortie : 1.
```

No visible error are shown on the agent side:
[image]

### Expected Results

Either the process should normally succeed or, if it's a user error, the message should be a bit more clear

### If problems with PlatformIO Build System:

I directly downloaded an unmodified version of Marlin from their releases. At the time of writing this, I tried to compile/upload  [Marlin@2.0.9.1](https://github.com/MarlinFirmware/Marlin/releases/tag/2.0.9.1) with no customisation at all.

### Additional info

I also tried running the agent on Windows, the same error happened. When I upload locally, the process does not fail.

Thanks

jefflessard, 2022-02-25:
I just ran into the same problem while trying to remote with Marlin.

The problem is that the `ini` folder (specified in extra_configs) of Marlin is not synced to the remote agent.

Here is the relevant section of platformio.ini as an example :
```
[platformio]
src_dir      = Marlin
boards_dir   = buildroot/share/PlatformIO/boards
default_envs = chitu_f103
include_dir  = Marlin
extra_configs =
    ini/avr.ini
    ini/due.ini
   ...
```

I tried to use [shared_dir](https://docs.platformio.org/en/latest/projectconf/section_platformio.html#id22) to sync the ini folder using :
```
[platformio]
shared_dir   = ini
...
```

But there is a bug (unless there is some magic that i don't know of) in [platformio/commands/remote/client/run_or_test.py](https://github.com/platformio/platformio-core/blob/4839fe37a3d1433737c80147ce9a205100dc4925/platformio/commands/remote/client/run_or_test.py).
The remote location is hardcoded to `shared` instead of the specified one (`ini`).
```
            psync.add_item(cfg.path, "platformio.ini")
            psync.add_item(cfg.get("platformio", "shared_dir"), "shared")
            psync.add_item(cfg.get("platformio", "boards_dir"), "boards")
```

As a temporary fix, I manually copied the ini file to the remote project location (`~/.platformio/remote/projects/Marlin-[GUID]`) and it worked.

I think the best resolution would be to sync extra_configs files to the remote agent. Otherwise, fixing the hardcoded shared remote location could be an option.

ivankravets, 2022-06-28:
> The remote location is hardcoded to shared instead of the specified one (in).

Yes, you need to put any files/configs to the https://docs.platformio.org/en/latest/projectconf/section_platformio.html#shared-dir

closed 2022-06-28 by ivankravets