platformio/platformio-core #4054: Error: No such command "access"

GiulioCocconi, 2021-09-14:
What kind of issue is this?

- [ x ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:
Manjaro ARM

**PlatformIO Version** (`platformio --version`):

platformio-5.2.0-1 (from AUR)

### Description of problem
When I use PIO the system prompts this message:

`Usage: pio [OPTIONS] COMMAND [ARGS]...
Try 'pio -h' for help.

Error: No such command "access"`

I can't do anything...

#### Steps to Reproduce

1. Update platformio using AUR
2. Use `pio -h` command

### Actual Results
`Usage: pio [OPTIONS] COMMAND [ARGS]...
Try 'pio -h' for help.

Error: No such command "access"`

### Expected Results
Platform io displays help

### Additional info

ivankravets, 2021-09-16:
It seems PlatformIO Core was not installed correctly. Please use the official installation guide https://docs.platformio.org/en/latest/core/installation.html#super-quick-mac-linux

closed 2021-09-16 by ivankravets

mantielero, 2021-09-18:
Thanks @ivankravets. I managed to get it working following the guide.

I am in ArchLinux. I observed that not all the dependencies from the installer were installed by the archlinux package. Doing one package at a time, I saw that installing `python-zeroconf` fixed the problem.

This might help others.