platformio/platformio-core #4055: FileNotFoundError for cppcheck in platformio 5.2.0

jenkinz, 2021-09-14:
What kind of issue is this?

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:

Ubuntu 20.04.3 LTS

**PlatformIO Version** (`platformio --version`):

PlatformIO Core, version 5.2.0

### Description of problem

`pio check` (specifically cppcheck) fails due to a `FileNotFoundError` for the `cppcheck` tool (unable to find the tool executable).

Inspecting the path reported by platformio for the expected tool location, `~/.platformio/packages/tool-cppcheck/cppcheck`, indeed revels `cppcheck` is not found here. The binary is instead actually located at `~/.platformio/packages/tool-cppcheck/bin/cppcheck` (in the `bin` directory).

#### Steps to Reproduce

1. Configure `cppcheck` as check tool in `platformio.ini`:
```
[env:myenv]
platform = native
check_tool = cppcheck
```
2. Run `pio check`

### Actual Results

```
$ pio check --fail-on-defect low --fail-on-defect medium --fail-on-defect high
Checking debug > cppcheck (platform: native)
--------------------------------------------------------------------------------
Tool Manager: Installing platformio/tool-cppcheck @ ~1.250.0
Downloading...
Unpacking...
Tool Manager: tool-cppcheck @ 1.250.0 has been installed!
Error: Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/platformio/__main__.py", line 115, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/usr/local/lib/python3.9/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.9/site-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/usr/local/lib/python3.9/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.9/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.9/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/platformio/commands/check/command.py", line 135, in cli
    rc = ct.check(
  File "/usr/local/lib/python3.9/site-packages/platformio/commands/check/tools/cppcheck.py", line 247, in check
    self.execute_check_cmd(cmd)
  File "/usr/local/lib/python3.9/site-packages/platformio/commands/check/tools/base.py", line 176, in execute_check_cmd
    result = proc.exec_command(
  File "/usr/local/lib/python3.9/site-packages/platformio/proc.py", line 112, in exec_command
    with subprocess.Popen(*args, **kwargs) as p:
  File "/usr/local/lib/python3.9/subprocess.py", line 951, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/local/lib/python3.9/subprocess.py", line 1821, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: '~/.platformio/packages/tool-cppcheck/cppcheck'
============================================================
An unexpected error occurred. Further steps:
* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command
* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html
* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues
============================================================
```

### Expected Results

Check completes without error due to `cppcheck` being found in the expected location.

valeros, 2021-09-14:
Hi @jenkinz, thanks for reporting! Please pull the latest package version via the `pio update` command in your terminal.

closed 2021-09-14 by valeros

jenkinz, 2021-09-14:
That worked, thanks!