platformio/platformio-core #4056: Worked last week, now Fails to compile: ESP32 KeyError: "Invalid board option 'build.cpu'":

drphil3d, 2021-09-15:
- [ ] **PlatformIO Core**.

### Configuration

**Operating system**:
Windows 10
**PlatformIO Version** (`platformio --version`):
PlatformIO Core, version 5.2.0
### Description of problem
Compile fails on project that's been working flawlessly for the past 8 months. Last week was able to compile just fine, noting has changed in the code base yet this week I get an obscure error that I'm unable to find any relevant documentation or solutions to. I'm getting this error on my other Windows 10 desktop as well. However, I'm able to compile a similar project with the same PIO file without issue.

I've done the regedit fix for python 3.9, didn't help. Uninstalled 3.9, same problem. 
I currently have Python 3.7.7

```
> Executing task: C:\Users\Phil\.platformio\penv\Scripts\platformio.exe run <

Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.3.2) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10006.210326 (1.0.6)
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - tool-openocd-esp32 2.1000.20210721 (10.0)
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
KeyError: "Invalid board option 'build.cpu'":
  File "C:\Users\Phil\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 178:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Phil\.platformio\packages\tool-scons\scons-local-4.2.0\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Phil\.platformio\packages\tool-scons\scons-local-4.2.0\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Phil\.platformio\platforms\espressif32\builder\main.py", line 218:
    target_elf = env.BuildProgram()
  File "C:\Users\Phil\.platformio\packages\tool-scons\scons-local-4.2.0\SCons\Util.py", line 748:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Phil\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 62:    
    env.ProcessProjectDeps()
  File "C:\Users\Phil\.platformio\packages\tool-scons\scons-local-4.2.0\SCons\Util.py", line 748:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Phil\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 141:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\Phil\.platformio\packages\tool-scons\scons-local-4.2.0\SCons\Util.py", line 748:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Phil\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1089:
    project.install_dependencies()
  File "C:\Users\Phil\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 907:
    if _is_builtin(spec):
  File "C:\Users\Phil\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 899:
    for lb in self.env.GetLibBuilders():
  File "C:\Users\Phil\.platformio\packages\tool-scons\scons-local-4.2.0\SCons\Util.py", line 748:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Phil\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1020:
    lb = LibBuilderFactory.new(env, lib_dir)
  File "C:\Users\Phil\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 61:
    obj = getattr(sys.modules[__name__], clsname)(env, path, verbose=verbose)
  File "C:\Users\Phil\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 139:
    self.process_extra_options()
  File "C:\Users\Phil\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 279:
    self.env.ProcessFlags(self.build_flags)
  File "C:\Users\Phil\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 570:
    LIBPATH=os.path.join(self.src_dir, board_config.get("build.cpu"))
  File "c:\users\phil\.platformio\penv\lib\site-packages\platformio\platform\board.py", line 46:
    raise KeyError("Invalid board option '%s'" % path)
============================================= [FAILED] Took 0.94 seconds =============================================
The terminal process "C:\Users\Phil\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.
```

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[platformio]
src_dir = src
lib_dir = lib
default_envs = esp32dev
;[common_env_data]

[common]
lib_ldf_mode = deep+
lib_deps_builtin = 
	SPIFFS
	FS
	SPI
lib_deps = 
	bodmer/TFT_eSPI@^2.3.70
	lbernstone/Tone32@^1.0.0
	contrem/arduino-timer@^2.3.0
	mikalhart/TinyGPSPlus@^1.0.2
	sparkfun/SparkFun Qwiic Scale NAU7802 Arduino Library@^1.0.4
	ivanseidel/LinkedList@0.0.0-alpha+sha.dac3874d28
	paulstoffregen/Time@^1.6
	sparkfun/SparkFun BQ27441 LiPo Fuel Gauge Arduino Library@^1.1.0
	lvgl/lvgl@7.11.0
	adafruit/RTClib@^1.13.0
	h2zero/NimBLE-Arduino@^1.2.0
	contrem/arduino-timer @ ^2.3.0
 	;adafruit/Adafruit BME280 Library@^2.1.4.
	;cesanta/mDash @ ^1.2.14

build_flags = 
	-DCORE_DEBUG_LEVEL=5
	-DPS_LOG_LEVEL=5
	;-DCONFIG_BT_NIMBLE_DEBUG
	-Wno-unused-variable
	-Wno-unused-function
	-DDEBUG_REPORT_HEAP_SIZE
	-DDEBUG_REPORT_STACK_FREE
	-D LV_CONF_INCLUDE_SIMPLE
	-I src
	-DUSER_SETUP_LOADED=1
	-D ST7796_DRIVER
	-DTFT_BL=32
	-DTFT_MISO=19
	-DTFT_MOSI=23
	-DTFT_SCLK=18
	-DTFT_CS=17
	-DTFT_DC=2
	-DTFT_RST=4
	-DTOUCH_CS=0
	-DLOAD_GLCD
	-DLOAD_FONT2
	-DLOAD_FONT4
	-DLOAD_FONT6
	-DLOAD_FONT7
	-DLOAD_FONT8
	-DLOAD_GFXFF
	-DLED_RED=25
	-DLED_GREEN=26
	-DLED_BLUE=27
	-DBUZZER_PIN=5
	-DPIN_TARE=33
	-DPIN_UP=36
	-DPIN_DOWN=34
	-DPIN_ENTER=39
	-DBTLE_DB=5
	;-L.pio/libdeps/esp32dev/mDash/src/esp32/ -llibmDash

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
upload_protocol = esptool
upload_speed = 921600
board_build.partitions = min_spiffs.csv
;upload_protocol = espota
;upload_port = 192.168.1.212
upload_port = COM11
;monitor_port = COM5
build_type = debug
;debug_tool = esp-prog
;upload_protocol = esp-prog
monitor_speed = 115200
monitor_flags = 
	--eol=CRLF
	--echo
	--filter=esp32_exception_decoder
board_build.f_cpu = 240000000L
; set frequency to 80MHz
board_build.f_flash = 80000000L
build_flags = ${common.build_flags}
src_filter = 
	+<*.h> +<*.s> +<*.S> +<*.cpp> +<*.c> +<*.ino> +<src/>
	-<.git/> -<data/> -<test/> -<tests/>
lib_deps = 
	bodmer/TFT_eSPI@^2.3.70
	lbernstone/Tone32@^1.0.0
	contrem/arduino-timer@^2.3.0
	mikalhart/TinyGPSPlus@^1.0.2
	sparkfun/SparkFun Qwiic Scale NAU7802 Arduino Library@^1.0.4
	ivanseidel/LinkedList@0.0.0-alpha+sha.dac3874d28
	paulstoffregen/Time@^1.6
	sparkfun/SparkFun BQ27441 LiPo Fuel Gauge Arduino Library@^1.1.0
	lvgl/lvgl@7.11.0
	adafruit/RTClib@^1.13.0
	h2zero/NimBLE-Arduino@^1.2.0
	contrem/arduino-timer @ ^2.3.0
 	;adafruit/Adafruit BME280 Library@^2.1.4
	;cesanta/mDash @ ^1.2.14
```

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info

maxgerhardt, 2021-09-16:
Also noted in https://community.platformio.org/t/precompiled-libraries-error-in-5-2-0/23486/

The `platformio\builder\tools\piolib.py` tries to access the `board.cpu` field which many board definitions do not provide, e.g., all the ESP32 ones (https://github.com/platformio/platform-espressif32/blob/v1.10.0/boards/firebeetle32.json#L1-L11) 

https://github.com/platformio/platformio-core/blob/e6fd766fff101af48fdbf00adc5975821f104a67/platformio/builder/tools/piolib.py#L566-L571

This was introduced in a recent commit that was supposed to correctly apply the "precompiled" flags of the `library.properties", https://github.com/platformio/platformio-core/commit/dce5a39b10d50e402813b17c578c05dbfe79e55b, from 3 days ago.

CC @ivankravets  critical.

maxgerhardt, 2021-09-16:
A workaround for now is to downgrade PlatformIO to the previous release. In a [CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli), type `pio system info` to find the "Python Executable", then execute a command with that path plus `-m pip uninstall platformio`, then again with `-m pip install -Iv  "platformio==5.1.1"`. But I think this will be fixed rather quickly.

ivankravets, 2021-09-16:
Please give us a few minutes, we will release 5.2.1.

@maxgerhardt, thanks for the report! 🙏

maxgerhardt, 2021-09-16:
One thing that may be important when writing the fix is not only defaulting to `board.mcu` for when `board.cpu` isn't there (and throwing a nicer error message if that also isn't there), but how the code handles it when the project already uses `build_flags` to link in the library like in the original issue. Will it work when they are multiple same `-l<library>` flags in the linker command? Should the code detect that and compensate?  Just a few thoughts. Also PlatformIO CI should add a few test cases for test-compiling a project with a precompiled lib for multiple platforms to prevent regressions.

maxgerhardt, 2021-09-16:
Hmm the fix which now leaves CPU empty if it's not existing is not correct.

https://github.com/platformio/platformio-core/blob/55469327c602cd8909dd2fe6cfea556c6cc88dc7/platformio/builder/tools/piolib.py#L569-L571

E.g., the BSEC library does have an `esp32` folder: https://github.com/BoschSensortec/BSEC-Arduino-library/tree/master/src

That fix would just add the source folder of the library to the search path. I'd propse something like

```python
            cpu_arch = board_config.get("build.cpu", board_config.get("build.mcu", ""))
            if cpu_arch == "": 
                pass # error handling
            self.env.PrependUnique(
                LIBPATH=os.path.join(self.src_dir, cpu_arch)
            )

drphil3d, 2021-09-16:
I added this line to my environment. Didn't fix anything.
board_build.mcu = esp32

maxgerhardt, 2021-09-16:
That is already there in the board definition, so it won't have any effect. If anything, on the buggy 2.0.0 version, try ` board_build.cpu = esp32` in the `platformio.ini`. But dynamic changes may not yet register in that code. Best to wait for an official & released bugfix. (`pio upgrade --dev` in the [docs](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli))

drphil3d, 2021-09-16:
sounds good, thank you. I was loosing my mind I'm glad I reported the issue. 🥇

ivankravets, 2021-09-16:
We tried to follow Arduino's specification https://arduino.github.io/arduino-cli/0.19/library-specification/#precompiled-binaries . However, those guys are strange. MCU & CPU are the same for them. As result, we see a mix of CPU & MCU among the library https://github.com/BoschSensortec/BSEC-Arduino-library/tree/master/src

I've just added a workaround at https://github.com/platformio/platformio-core/commit/369e994b0d6e81ebcdd8a6f9b76ebe14425f3b2f . It should work in all cases. Please re-test with `pio upgrade --dev`.

closed 2021-09-16 by ivankravets

maxgerhardt, 2021-09-17:
Can confirm that PlatformIO now links against precompiled libraries in the example of my original issue with a ARM Cortex-M4 based `mkrwifi1010`. Also tested an ESP32 with the BSEC library (unmodified) and it works. The nice thing is now that even when I delete the BSEC library's `library.json` file, it now links correctly, just using the now-respected properties of the `library.properties`.

freemanr7, 2021-09-26:
I'm having similar problems as was presented by drphil3d.  I'm having some difficulty understanding how to incorporate your fix into my installation.  Based on info below, I think I may have broken something else.  Can you recommend how I might get this fixed please?  Thanks,  Robert
***************************************************************************************************************************************
Obsolete PIO Core v5.2.0 is used (previous was 5.2.1b2)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/page/faq.html#multiple-platformio-cores-in-a-system
***************************************************************************************************************************************
Processing esp32thing_plus (platform: espressif32; board: esp32dev; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.3.2) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10006.210326 (1.0.6) 
 - tool-esptoolpy 1.30100.210531 (3.1.0) 
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
KeyError: "Invalid board option 'build.cpu'":
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/platformio/builder/main.py", line 178:
    env.SConscript("$BUILD_SCRIPT")
  File "/Users/robertfreeman/.platformio/packages/tool-scons/scons-local-4.2.0/SCons/Script/SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/Users/robertfreeman/.platformio/packages/tool-scons/scons-local-4.2.0/SCons/Script/SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/Users/robertfreeman/.platformio/platforms/espressif32/builder/main.py", line 218:
    target_elf = env.BuildProgram()
  File "/Users/robertfreeman/.platformio/packages/tool-scons/scons-local-4.2.0/SCons/Util.py", line 748:
    return self.method(*nargs, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/platformio/builder/tools/platformio.py", line 62:
    env.ProcessProjectDeps()
  File "/Users/robertfreeman/.platformio/packages/tool-scons/scons-local-4.2.0/SCons/Util.py", line 748:
    return self.method(*nargs, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/platformio/builder/tools/platformio.py", line 141:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "/Users/robertfreeman/.platformio/packages/tool-scons/scons-local-4.2.0/SCons/Util.py", line 748:
    return self.method(*nargs, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/platformio/builder/tools/piolib.py", line 1091:
    lib_builders = env.GetLibBuilders()
  File "/Users/robertfreeman/.platformio/packages/tool-scons/scons-local-4.2.0/SCons/Util.py", line 748:
    return self.method(*nargs, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/platformio/builder/tools/piolib.py", line 1020:
    lb = LibBuilderFactory.new(env, lib_dir)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/platformio/builder/tools/piolib.py", line 61:
    obj = getattr(sys.modules[__name__], clsname)(env, path, verbose=verbose)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/platformio/builder/tools/piolib.py", line 139:
    self.process_extra_options()
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/platformio/builder/tools/piolib.py", line 279:
    self.env.ProcessFlags(self.build_flags)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/platformio/builder/tools/piolib.py", line 570:
    LIBPATH=os.path.join(self.src_dir, board_config.get("build.cpu"))
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/platformio/platform/board.py", line 46:
    raise KeyError("Invalid board option '%s'" % path)
===================================================== [FAILED] Took 0.62 seconds =====================================================
The terminal process "platformio 'run'" terminated with exit code: 1.

maxgerhardt, 2021-09-26:
>Obsolete PIO Core v5.2.0 is used (previous was 5.2.1b2)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/page/faq.html#multiple-platformio-cores-in-a-system

You have installed PlatformIO incorrectly, there are multiple cores in your system. Follow the refernecd link to get more info. In short, try to uninstall all PlatformIO installations (`pip uninstall platformio`, `pip3 uninstall paltformio`, `sudo pip3 uninstall platformio`, remove `<home folder>/.platformio` completely) and [reinstall it](https://docs.platformio.org/en/latest/core/installation.html#installer-script). (Once all cores are removed though, the PlatformIO VSCode installatino will recognize that and try to install PlatformIO again).

freemanr7, 2021-09-27:
Thanks very much. I was able to get things restored and get the dev version
installed. That took care of the errors tied to this thread. I am still not
able to get that BSEC library to link. Not sure if I need to change a
setting or flag somewhere. Getting closer :-)

Kind regards,

Robert

ivankravets, 2021-09-27:
@freemanr7 what is your `platformio.ini`?

freemanr7, 2021-09-27:
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32thing_plus]
platform = espressif32
board = esp32thing_plus
framework = arduino
upload_port = /dev/cu.SLAB_USBtoUART

The compilation errors:

> Executing task: platformio run <

Processing esp32thing_plus (platform: espressif32; board: esp32thing_plus; framework: arduino)
-------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32thing_plus.html
PLATFORM: Espressif 32 (3.3.2) > SparkFun ESP32 Thing Plus
HARDWARE: ESP32 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10006.210326 (1.0.6) 
 - tool-esptoolpy 1.30100.210531 (3.1.0) 
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 32 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <EEPROM> 1.0.3
|-- <SPI> 1.0
|-- <Wire> 1.0.1
Building in release mode
Linking .pio/build/esp32thing_plus/firmware.elf
.pio/build/esp32thing_plus/src/main.cpp.o:(.literal._Z4loopv+0x4): undefined reference to `Bsec::run()'
.pio/build/esp32thing_plus/src/main.cpp.o:(.literal._Z9loadStateR4Bsec+0xc): undefined reference to `Bsec::setState(unsigned char*)'
.pio/build/esp32thing_plus/src/main.cpp.o:(.literal._Z5setupv+0x2c): undefined reference to `Bsec::begin(unsigned char, TwoWire&)'
.pio/build/esp32thing_plus/src/main.cpp.o:(.literal._Z5setupv+0x30): undefined reference to `Bsec::setConfig(unsigned char const*)'
.pio/build/esp32thing_plus/src/main.cpp.o:(.literal._Z5setupv+0x34): undefined reference to `Bsec::updateSubscription(bsec_virtual_sensor_t*, unsigned char, float)'
.pio/build/esp32thing_plus/src/main.cpp.o:(.literal._Z5setupv+0x38): undefined reference to `bsec_get_version'
.pio/build/esp32thing_plus/src/main.cpp.o:(.literal._Z9saveStateR4Bsec+0x4): undefined reference to `Bsec::getState(unsigned char*)'
.pio/build/esp32thing_plus/src/main.cpp.o:(.literal.startup._GLOBAL__sub_I_bsecInst+0x8): undefined reference to `Bsec::Bsec(void (*)(bme68x_data const&, BsecOutput const&))'
.pio/build/esp32thing_plus/src/main.cpp.o: In function `loop()':
/Users/robertfreeman/Documents/PlatformIO/Projects/BME688-BSEC2/src/main.cpp:55: undefined reference to `Bsec::run()'
.pio/build/esp32thing_plus/src/main.cpp.o: In function `loadState(Bsec&)':
/Users/robertfreeman/Documents/PlatformIO/Projects/BME688-BSEC2/src/main.cpp:174: undefined reference to `Bsec::setState(unsigned char*)'
.pio/build/esp32thing_plus/src/main.cpp.o: In function `setup()':
/Users/robertfreeman/Documents/PlatformIO/Projects/BME688-BSEC2/src/main.cpp:42: undefined reference to `Bsec::begin(unsigned char, TwoWire&)'
/Users/robertfreeman/Documents/PlatformIO/Projects/BME688-BSEC2/src/main.cpp:43: undefined reference to `Bsec::setConfig(unsigned char const*)'
/Users/robertfreeman/Documents/PlatformIO/Projects/BME688-BSEC2/src/main.cpp:45: undefined reference to `Bsec::updateSubscription(bsec_virtual_sensor_t*, unsigned char, float)'
.pio/build/esp32thing_plus/src/main.cpp.o: In function `Bsec::getVersion()':
/Users/robertfreeman/Documents/PlatformIO/Projects/BME688-BSEC2/include/bsec.h:186: undefined reference to `bsec_get_version'
/Users/robertfreeman/Documents/PlatformIO/Projects/BME688-BSEC2/include/bsec.h:186: undefined reference to `bsec_get_version'
/Users/robertfreeman/Documents/PlatformIO/Projects/BME688-BSEC2/include/bsec.h:186: undefined reference to `bsec_get_version'
/Users/robertfreeman/Documents/PlatformIO/Projects/BME688-BSEC2/include/bsec.h:186: undefined reference to `bsec_get_version'
.pio/build/esp32thing_plus/src/main.cpp.o: In function `saveState(Bsec&)':
/Users/robertfreeman/Documents/PlatformIO/Projects/BME688-BSEC2/src/main.cpp:194: undefined reference to `Bsec::getState(unsigned char*)'
.pio/build/esp32thing_plus/src/main.cpp.o: In function `_GLOBAL__sub_I_bsecInst':
/Users/robertfreeman/Documents/PlatformIO/Projects/BME688-BSEC2/src/main.cpp:19: undefined reference to `Bsec::Bsec(void (*)(bme68x_data const&, BsecOutput const&))'
collect2: error: ld returned 1 exit status
*** [.pio/build/esp32thing_plus/firmware.elf] Error 1
================================== [FAILED] Took 4.92 seconds ==================================
The terminal process "platformio 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

maxgerhardt, 2021-09-27:
Please edit your previous post to format the output as code using three backticks in [Markdown](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code) and not just copy-paste the output. I can't see the dependency graph at all in this formatting. 

Where does the BSEC library come from? You don't include it via `lib_deps` as shown [here](https://platformio.org/lib/show/6979/BSEC%20Software%20Library/installation). 

When you open a [CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli) what does `pio --version` return?

freemanr7, 2021-09-28:
@maxgerhardt 
Thank you very much for your counsel.  I'm not sure I understand your formatting request, but I will work to learn more and do a better job in the future.  

The BSEC library is from Bosch.  You helped ukarthiksg on [https://community.platformio.org/t/bsec-2-0-and-bme688/21813](url).  

pio --version now returns:  PlatformIO Core, version 5.2.1b2.

I added
>  build_flags = -L/Users/robertfreeman/Documents/PlatformIO/Projects/BME688-BSEC2/lib/esp32/ -lalgobsec.a
to the ini file and found some files I had the wrong version and/or in the wrong folders.  Seems to be working now.  Compiling with no issues.  

Warm Regards,

Robert

maxgerhardt, 2021-09-28:
Okay so you were not talking about [BSEC Arduino Library](https://github.com/BoschSensortec/BSEC-Arduino-library) but **BSEC2**. 

But it's still the same story -- no extra build flags are needed for linking to work **if you correctly copy the library**. When I download the referenced `bsec_2-0-6-1_generic_release_04302021.zip` file, unpack it, create a new blank project for the ESP32Dev, copy the  BSEC_2.0.6.1_Generic_Release_04302021\examples\bsec2` folder into `lib/`, copy the `BSEC_2.0.6.1_Generic_Release_04302021\examples\bsec2\examples\basic_config_state\basic_config_state.ino` into `src/` and add `build_flags = -DLED_BUILTIN=2` (to define the **LED** -- the sketch needs that) to the `platformio.ini`, it compiles without problems. The `<BSEC Software Library> 2.0.100` is picked up by PlatformIO and the precompiled library from the `esp32/` folder is linked in automatically.

```
Linking .pio\build\esp32dev\firmware.elf
Retrieving maximum program size .pio\build\esp32dev\firmware.elf
Checking size .pio\build\esp32dev\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   5.5% (used 18016 bytes from 327680 bytes)
Flash: [==        ]  21.6% (used 283430 bytes from 1310720 bytes)
Building .pio\build\esp32dev\firmware.bin
=============== [SUCCESS] Took 9.31 seconds ===============
```

freemanr7, 2021-10-01:
> The 2.0.100is picked up by PlatformIO and the precompiled library from theesp32/`

That sounds very elegant. I will give it a try.

Thanks very much,

Robert