platformio/platformio-core #4067: Framework source files not found when indenting a section name in platformio.ini file

progirep, 2021-09-25:
Here is a weird behavior that I believe is not intended.

Let's assume that I have the following platformio.ini file in a directory:

```
[env:nucleo_f446re]
platform = ststm32
board = nucleo_f446re
framework = stm32cube

[platformio]
src_dir = Src

    [env]
    # nothing here
```
So the "[env]" is accidentally indented. When running "pio run" in the directory of this platformio.ini file, platformio "forget" to tell the compiler which source file to compile when compiling the source files for the framework:

```
me@my-compiler:~/path/to/testProject$ pio run -v
Processing nucleo_f446re (platform: ststm32; board: nucleo_f446re; framework: stm32cube)
--------------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/nucleo_f446re.html
PLATFORM: ST STM32 (14.1.0) > ST Nucleo F446RE
HARDWARE: STM32F446RET6 180MHz, 128KB RAM, 512KB Flash
DEBUG: Current (stlink) On-board (stlink) External (blackmagic, cmsis-dap, jlink)
PACKAGES: 
 - framework-stm32cubef4 1.25.2 
 - tool-ldscripts-ststm32 0.1.0 
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Searching for startup file at:  /home/me/.platformio/packages/framework-stm32cubef4/Drivers/CMSIS/Device/ST/STM32F4xx/Source/Templates/gcc/startup_stm32f446xx.S
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 48 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
arm-none-eabi-gcc -o .pio/build/nucleo_f446re/FrameworkHALDriver/Src/stm32f4xx_hal.o -c -Os -ffunction-sections -fdata-sections -Wall -mthumb -mcpu=cortex-m4 -nostdlib -DPLATFORMIO=50200 -DSTM32F446xx -DUSE_HAL_DRIVER -DF_CPU=180000000L -ISrc
arm-none-eabi-gcc: fatal error: no input files
compilation terminated.
arm-none-eabi-gcc -o .pio/build/nucleo_f446re/FrameworkHALDriver/Src/stm32f4xx_hal_adc.o -c -Os -ffunction-sections -fdata-sections -Wall -mthumb -mcpu=cortex-m4 -nostdlib -DPLATFORMIO=50200 -DSTM32F446xx -DUSE_HAL_DRIVER -DF_CPU=180000000L -ISrc
arm-none-eabi-gcc: fatal error: no input files
compilation terminated.
arm-none-eabi-gcc -o .pio/build/nucleo_f446re/FrameworkHALDriver/Src/stm32f4xx_hal_adc_ex.o -c -Os -ffunction-sections -fdata-sections -Wall -mthumb -mcpu=cortex-m4 -nostdlib -DPLATFORMIO=50200 -DSTM32F446xx -DUSE_HAL_DRIVER -DF_CPU=180000000L -ISrc
arm-none-eabi-gcc: fatal error: no input files
compilation terminated.
*** [.pio/build/nucleo_f446re/FrameworkHALDriver/Src/stm32f4xx_hal.o] Error 1
arm-none-eabi-gcc -o .pio/build/nucleo_f446re/FrameworkHALDriver/Src/stm32f4xx_hal_can.o -c -Os -ffunction-sections -fdata-sections -Wall -mthumb -mcpu=cortex-m4 -nostdlib -DPLATFORMIO=50200 -DSTM32F446xx -DUSE_HAL_DRIVER -DF_CPU=18000*** [.pio/build/nucleo_f446re/FrameworkHALDriver/Src/stm32f4xx_hal_adc.o] Error 1
0000L -ISrc
*** [.pio/build/nucleo_f446re/FrameworkHALDriver/Src/stm32f4xx_hal_adc_ex.o] Error 1
arm-none-eabi-gcc: fatal error: no input files
compilation terminated.
*** [.pio/build/nucleo_f446re/FrameworkHALDriver/Src/stm32f4xx_hal_can.o] Error 1
=============================================== [FAILED] Took 1.15 seconds ===============================================
```

While it's certainly correct that the platformio.ini file is wrong, adding "[env]" to the source directories shouldn't let platformio forget to tell GCC which files to compile, I guess.

ivankravets, 2021-09-26:
It means that `src_dirs = Src

    [env]
    # nothing here`

closed 2021-09-26 by ivankravets

progirep, 2021-09-26:
I am aware of the meaning of what I wrote in the platformio.ini file. The actual problem is that the user receives an error message that is unrelated to what is wrong in the platformio.ini file.

The issue is regarding the fact that this erroneous line causes platformio to "forget" to pass the framework source file name when compiling the framework object files. Since these are not in "src_dir", passing a wrong "src_dir" should not affect their compilation.

On further inspection, it's not the "[env]" that causes the behavior, but rather the newline.

So
```
[platformio]
src_dir = Src OtherSrc
```
lets the framework object files compile correctly while

```
[platformio]
src_dir = Src 
    OtherSrc
```

does not. 

While indeed the project would not compile either way, the actual problem is that the user receives an error message that is unrelated to what is wrong in the platformio.ini file. With

`src_dir = Src OtherSrc`

the user gets as final linker error `(.text.Reset_Handler+0x32): undefined reference to `main'`, which is helpful. But with 

```
src_dir = Src 
    OtherSrc
```
the error message indicates that platformio calls GCC with a missing parameter, which is unrelated to not finding source files due to a wrong directory name.