platformio/platformio-core #4071: Bug in CLion for CMakeListsPrivate.txt generation when using lib_extra_dirs in Windows

john-contact, 2021-09-30:
Hey everyone, hopefully you can help me resolve this or at least I can bring this to your attention.

I'm on Windows 10 using the PlatformIO plugin (version 212.5080.54).

It seems that if I use an absolute path in `lib_extra_dirs`, it attempts to be smart in its generation of CMakeListsPrivate.txt by replacing the drive and user directory with `$ENV{HOMEDRIVE}$ENV{HOMEPATH}`, which then evaluates to include backslashes and breaks cmake. This occurs even if you explicitly use forward slashes in the absolute path.

In platform.ini:

```
lib_extra_dirs=C:/Users/john_contactci/AppData/Local/Maestro/libs
```

In the resulting generated CMakeListsPrivate.txt:
```
    FILE(GLOB_RECURSE EXTRA_LIB_SOURCES
        ${CMAKE_CURRENT_LIST_DIR}/.pio/libdeps/teensy40/*.*
        $ENV{HOMEDRIVE}$ENV{HOMEPATH}/AppData/Local/Maestro/libs/*.*
    )
```

Later on `EXTRA_LIB_SOURCES` is appended to `SRC_LIST`:
```
list(APPEND SRC_LIST ${EXTRA_LIB_SOURCES})
```

And this results in the following error from CMakeLists.txt:
```
CMake Error at CMakeLists.txt:33 (add_executable):
  Syntax error in cmake code when parsing string

    C:\Users\john_contactci/AppData/Local/Maestro/libs/maestro-embedded-config/include/MaestroEmbeddedConfiguration.h

  Invalid character escape '\U'.
```
Due to the backslash in the path.

For reference, line 33 of CMakeLists.txt:
```
add_executable(Z_DUMMY_TARGET ${SRC_LIST})
```
Which is why I mention the bit about `EXTRA_LIB_SOURCES` being appended to `SRC_LIST` above.

Things I have tried to no avail:
- Defining a `PLATFORMIO_LIB_EXTRA_DIRS` env var, which results in the same problem.
- Throwing quotes around the path, which just results in it being treated as a relative path rather than absolute.
- Using backslashes but escaping them (i.e. `\\`) in the absolute path.

Any assistance or work around would be greatly appreciated!

Thanks!

valeros, 2021-12-02:
Hi @john-contact , thanks for reporting! It should be fixed in the development branch. It would be great if you could retest the latest PlatformIO core with your project. Just run `pio upgrade --dev` and reinit your project.

closed 2021-12-02 by valeros

john-contact, 2021-12-02:
Hey @valeros , thanks for picking this one up!

I think this about 99% of the way there.

I now get:

```
Cannot resolve compiler path '/.platformio/packages/toolchain-gccarmnoneeabi/bin/arm-none-eabi-g++.exe'
```

If I take a look at the generated `CMakeListsPrivate.txt`, I see the definition for the c++ compiler path:

```
SET(CMAKE_CXX_COMPILER "${ENV_HOME_PATH}/.platformio/packages/toolchain-gccarmnoneeabi/bin/arm-none-eabi-g++.exe")
```

Which given the output I see implies that `ENV_HOME_PATH` is undefined (or empty). A few lines down (after the C and C++ compiler paths and corresponding flags are defined), I see your patch using `TO_CMAKE_PATH`:

```
file(TO_CMAKE_PATH $ENV{HOMEDRIVE}$ENV{HOMEPATH} ENV_HOME_PATH)
```

If I move that `TO_CMAKE_PATH` line to be before the compiler path+flags definitions, the issue is gone and everything appears fixed at that point.

Basically, from looking at the associated commit `f13734d`, I think that if you move your patch in file `platformio/project/tpls/clion/CMakeListsPrivate.txt.tpl` to instead be before line 60 it will fix it. Like this:

```
...

# Convert "Home Directory" that may contain unescaped backslashes on Windows
% if "windows" in systype:
file(TO_CMAKE_PATH $ENV{HOMEDRIVE}$ENV{HOMEPATH} ENV_HOME_PATH)
% end

SET(CMAKE_C_COMPILER "{{ _normalize_path(cc_path) }}")
SET(CMAKE_CXX_COMPILER "{{ _normalize_path(cxx_path) }}")
SET(CMAKE_CXX_FLAGS "{{ _normalize_path(to_unix_path(cxx_flags)) }}")
SET(CMAKE_C_FLAGS "{{ _normalize_path(to_unix_path(cc_flags)) }}")

...
```

Thanks again for the help with this, and let me know if you need anything from my end!

valeros, 2021-12-02:
Thanks for the quick response, my fault. Please pull the latest changes via `pio upgrade --dev` and try again.

john-contact, 2021-12-02:
@valeros Awesome, that fixed it. Thanks again!

commit f13734dd (2021-12-02): Convert Home Directory path into a cmake-style path on Windows

Resolve #4071