platformio/platformio-core #4073: Including Amaioranos "hsm" lib goes wrong on Linux

Bascy, 2021-10-02:
What kind of issue is this?
- [ ] **PlatformIO Core**.

### Configuration

**Operating system**: Linux

**PlatformIO Version** (`platformio --version`): 5.2.1b4

### Description of problem
When using dependency on https://github.com/amaiorano/hsm platformio does not install the correct folder with source codes on Linux. 

Windows / VSCode works fine, and includes only folder `include` which is enough

With LTS of platformio compiling on Linux downloads the folder `hsm/samples/hsm_book_samples/source/ch4/`
When compiling after `pio upgrade --dev` only the folder `hsm/samples/hsm_book_samples/` is included

Related to issue #3887

ivankravets, 2021-10-08:
Thanks for the report! Please re-test with `pio upgrade --dev`.

Bascy, 2021-10-09:
Whole lib is downloaded now. thanks