platformio/platformio-core #4074: Big Sur - The terminal process "platformio 'run'" terminated with exit code: 1.

TrickyToo, 2021-10-02:
VSCode Bug Sur - Platformio run fails i.e. The terminal process "platformio 'run'" terminated with exit code: 1.

Executing task in folder esp32dev - WordClock mk2: platformio run <

Error: Traceback (most recent call last):
  File "/Users/Richard/.platformio/penv/lib/python3.8/site-packages/platformio/__main__.py", line 115, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/Users/Richard/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/Users/Richard/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/Users/Richard/.platformio/penv/lib/python3.8/site-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/Users/Richard/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 1254, in invoke
    cmd_name, cmd, args = self.resolve_command(ctx, args)
  File "/Users/Richard/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 1297, in resolve_command
    cmd = self.get_command(ctx, cmd_name)
  File "/Users/Richard/.platformio/penv/lib/python3.8/site-packages/platformio/commands/__init__.py", line 64, in get_command
    mod = __import__(mod_path, None, None, ["cli"])
  File "/Users/Richard/.platformio/penv/lib/python3.8/site-packages/platformio/commands/run/command.py", line 25, in <module>
    from platformio.commands.device.command import device_monitor as cmd_device_monitor
  File "/Users/Richard/.platformio/penv/lib/python3.8/site-packages/platformio/commands/device/__init__.py", line 15, in <module>
    from platformio.commands.device.filters.base import DeviceMonitorFilter
  File "/Users/Richard/.platformio/penv/lib/python3.8/site-packages/platformio/commands/device/filters/base.py", line 15, in <module>
    from serial.tools import miniterm
  File "/Users/Richard/.platformio/penv/lib/python3.8/site-packages/serial/tools/miniterm.py", line 16, in <module>
    from serial.tools.list_ports import comports
  File "/Users/Richard/.platformio/penv/lib/python3.8/site-packages/serial/tools/list_ports.py", line 29, in <module>
    from serial.tools.list_ports_posix import comports
  File "/Users/Richard/.platformio/penv/lib/python3.8/site-packages/serial/tools/list_ports_posix.py", line 31, in <module>
    from serial.tools.list_ports_osx import comports
  File "/Users/Richard/.platformio/penv/lib/python3.8/site-packages/serial/tools/list_ports_osx.py", line 32, in <module>
    kIOMasterPortDefault = ctypes.c_void_p.in_dll(iokit, "kIOMasterPortDefault")
ValueError: dlsym(RTLD_DEFAULT, kIOMasterPortDefault): symbol not found

--------------------------------------------------------------------------
PlatformIO Core, version 5.2.1b4
OS: Big Sur verion 11.6

The source code previously complied, something has changed so it fails even to start to compile

slwise, 2021-10-05:
I'm receiving the same error messages.

OS: Big Sur 11.6

TrickyToo, 2021-10-06:
Further Information: Instead of using Platfomio controls to run or upload, I successfully ran the equivalent command on the terminal command line: i.e. "platformio run" or "platformio run --target upload", without any of the previously reported errors, I find this very strange.

ivankravets, 2021-10-09:
@TrickyToo does this solution https://github.com/platformio/platformio-core/issues/4074#issuecomment-938758235 work for you?

TrickyToo, 2021-10-10:
Yes it works for me
Many Thanks