platformio/platformio-core #4094: Shorten PIO update output

jcw, 2021-11-01:
- [x] **Feature Request**.

When I run `pio update`, the output lines are quite long, and wrap on my ≈ 80–char wide terminal window. With lots of frameworks installed, that generates somewhat hard-to read output. Here's a (particularly long) example:

    Updating platformio/toolchain-gccarmnoneeabi  1.70201.0 @ >=1.60301.0,<1.80000.0 [Up-to-date]

Would it be an idea to drop the leading `platformio/` text? As far as I can tell, it's redundant anyway.

(MacOS, PIO 5.2.3b3)

ivankravets, 2021-11-01:
> Would it be an idea to drop the leading platformio/ text?

This is a package owner's name. We can't remove it. How about listing only outdated packages and do not show `[Up-to-date]`? Or, show `[Up-to-date]` with `--verbose` flag?

jcw, 2021-11-01:
> listing only outdated packages

Nice, that would address the issue as well, i.e. omit less-important output.
Using `--verbose` to get the full output back also sounds like a good idea.

ivankravets, 2022-02-12:
Hi @jcw,

We solved this issue as part of https://github.com/platformio/platformio-core/issues/3373

The command `pio update` is deprecated and will be removed later in favor of `pio pkg outdated`.

What do you think about https://docs.platformio.org/en/latest/core/userguide/pkg/cmd_pack.html ?

P.S: Thanks for your donation and for supporting PlatformIO! 🙏

closed 2022-02-12 by ivankravets