platformio/platformio-core #4096: Windows NetBeans build fails if there's a space in the path to platformio.exe

RussellAult, 2021-11-03:
- [x] **PlatformIO Core**

------------------------------------------------------------------

### Configuration

**Operating system**: Windows 10 21H1 Pro

**PlatformIO Version** (`platformio --version`): 5.2.2

### Description of problem
The `<buildCommand>` and `<cleanCommand>` in `.\nbproject\configurations.xml` don't surround the full path to the `platformio.exe` executable with `"`s, which causes the commands to fail if there's a space in the path (since everything after the space is interpreted as an argument)

#### Steps to Reproduce

1. Install PlatformIO Core on Windows using `get-platformio.py` for a user whose username has a space in it
2. Create a new folder and run `pio project init --ide netbeans --board uno`
3. Open the project in NetBeans
4. Add some code
5. Run the "Build" command

### Expected Results
The project should build

### Actual Results
`Cannot run program "C:/Users/Firstname" (in directory "C:\Users\Firstname Lastname\Documents\project"): CreateProcess error=2, The system cannot find the file specified`

### Work-around
Manually editing `.\nbproject\configurations.xml` to add `"` around the full path for `<buildCommand>` and `<cleanCommand>` (i.e. `<buildCommand>C:/Users/Firstname Lastname/.platformio/penv/Scripts/platformio.exe -f -c netbeans run</buildCommand>` becomes `<buildCommand>"C:/Users/Firstname Lastname/.platformio/penv/Scripts/platformio.exe" -f -c netbeans run</buildCommand>`) allows the project to build as expected

valeros, 2021-11-15:
Thanks for reporting, fixed in the dev branch.  Please  upgrade your PlatformIO installation via `pio upgrade --dev` and try again.

closed 2021-11-15 by valeros

RussellAult, 2021-11-20:
Thanks! Will do!

commit 5cb5c971 (2021-11-15): Wrap the path to PlatformIO core in the NetBeans project template

This fixes a possible issue when the path to PlatformIO contains a whitespace

Resolve #4096