platformio/platformio-core #4098: Debugger does not honor selected environment

manuelbl, 2021-11-03:
The new environment switch in the status bar is a welcome enhancement. Unfortunately, it is not honored by the debugger. Instead, the debugger uses the first environment or the one selected by `default_envs`.

As the environment switch applies to IntelliSense and the *Build* and *Upload* actions in the status bar, users expect that the debugger behaves the same. I did so too.

In my case it even lead to a very confusing situation:
- Clicking the green arrow the debugger pane started a rebuild of the project (debug configuration) – as expected.
- After that, nothing happened anymore.
- It took me quite some time until I remembered to switch to the *DEBUG CONSOLE*. I found an error basically saying *unexpected idcode: 0x0bb11477, expected 1 of 1: 0x2ba01477*.
- It took me even more time to figure out that this error message was not due to a Chinese clone of the STM32 chip, but rather due to a mismatch between the environment use by the debugger and the board connected to my computer.

I see two options to fix it:
- Honor the currently select environment when starting the debugger (preferred)
- Label the debug configuration (`launch.json`) such that it is obvious which one is used

closed 2021-11-05 by ivankravets

dzid26, 2023-04-06:
> I've jus tried it and it didn't work. The correct environment was used to build a debug version. But the debugger is then started with a different (incorrect) environment.
> 
> `launch.json` looks ok. But the first line in the _DEBUG CONSOLE_ already points at the incorrect environment (the selected environment is _nucleo-f303re_, the debugger however used _nucleo_f042k6_, which is the top most in `platformio.ini`):
> 
> 
> Strangely enough, it looks as if the target was build twice: the output of the correct one appears in the _TERMINAL_ view, the output of the incorrect one appears in _DEBUG CONSOLE_.

I still have this problem. 
It's building twice (related?) and debug console uses an environment from `default_envs` rather than one containing `build_type = debug`. 

I am poking around because I get the annoying full double rebuild when debugging. 
I am on Windows

ivankravets, 2023-04-11:
@dzid26, could you share your `platformio.ini` content?

dzid26, 2023-04-11:
I am using [this ](https://github.com/dzid26/StepperServoCAN/blob/StepperServoCAN/firmware/platformio.ini) normally. 
But I was also testing separate debug environment in relation to this issue with the following:
```
;PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[platformio]
src_dir      = src
include_dir  = src/BSP
default_envs = ServoCAN

[env]
platform = ststm32

board = genericSTM32F103C8
upload_protocol = stlink

build_flags =
  -D STM32F10X_MD
  -D SYSCLK_FREQ_64MHz
  -D USE_STDPERIPH_DRIVER
  -O3 -flto
  -W -Wall -pedantic -std=c99
  -fsingle-precision-constant
  -Wdouble-promotion
  -Wfloat-conversion
  -I src/OP
  -I src/CMSIS
  -I src/LIB/inc
  -I src/APP
  -Wl,-Map,${BUILD_DIR}/firmware.map
  -D HSE_VALUE=16000000

board_build.ldscript = ./src/APP/STM32F103C8_DEFAULT.ld
board_upload.maximum_size = 63488
extra_scripts = linker_options_pio.py  ;skips definitions for system functions

check_tool = cppcheck, clangtidy
check_skip_packages = yes ;don't inlcude compiler package inludes 
check_patterns = 
  src/BSP/*.c
  src/BSP/*.h
check_flags =
  cppcheck:--addon=misra.json --std=c99 --language=c  --suppress=*:*/src/CMSIS/* --suppress=*:*/src/LIB/*
  clangtidy: --checks=-*,clang-analyzer-*,cert-*,bugprone-*,performance-*,hicpp-*,cppcoreguidelines-*,readability-*

[env:ServoCAN]

build_flags = ${env.build_flags}
  -g  #include variable symbols in elf file

  
[env:ServoCAN_debug]
build_type = debug
debug_tool = stlink
debug_build_flags = -Og -g3 -ggdb3 #disable most optimizations
  -D IGNORE_CAN_CHECKSUM
  -D DEBUG
debug_extra_cmds =
  monitor arm semihosting enable
  monitor arm semihosting_fileio enable
```

I think the problem is related to how `launch.json` is interpreted by the terminal and debug console, because when I change the default env on the toolbar or modify `"projectEnvName":` to `"ServoCAN_debug"` manually then both terminal and debug_console build the same thing. ...Albeit still the whole thing is rebuild twice which is my main trouble.

ivankravets, 2023-04-12:
See https://github.com/platformio/platformio-core/blob/develop/platformio/debug/helpers.py#L65

Does it makes sense to set "Environment Switcher" to "Auto"?

ivankravets, 2023-04-12:
@dzid26, thanks, I reproduced this issue. This is a bug and is different from this issue. Could I ask you to open a new issue at https://github.com/platformio/platformio-core/issues to track this problem?

commit 973f7701 (2021-11-05): Fixed an issue when VSCode's debugger does not honor default environment // Resolve #4098