platformio/platformio-core #4099: Failure to build/grab dependencies when building from network share

bizzehdee, 2021-11-04:
I am trying to have my projects hosted on network shared SSD using VSCode 1.61.2 and PlatformIO (Core 5.2.2 and Home 3.3.4) on Windows 10.

I have already tried the DisableUNCCheck reg patch and it has made no difference.

### Error Log

> Executing task: C:\Users\darren\.platformio\penv\Scripts\platformio.exe run --environment LPC1768 <

Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.3.zip; board: nxp_lpc1768; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Error: Traceback (most recent call last):
  File "C:\Program Files\Python39\lib\ntpath.py", line 647, in realpath
    path = _getfinalpathname(path)
FileNotFoundError: [WinError 161] The specified path is invalid: '\\\\VAULT\\Work_Darren\\Marlin\\https:\\github.com\\p3p\\pio-nxplpc-arduino-lpc176x\\archive\\0.1.3.zip'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\darren\.platformio\penv\lib\site-packages\platformio\__main__.py", line 129, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "C:\Users\darren\.platformio\penv\lib\site-packages\click\core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\darren\.platformio\penv\lib\site-packages\click\core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "C:\Users\darren\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "C:\Users\darren\.platformio\penv\lib\site-packages\click\core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\darren\.platformio\penv\lib\site-packages\click\core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\darren\.platformio\penv\lib\site-packages\click\core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "C:\Users\darren\.platformio\penv\lib\site-packages\click\decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "C:\Users\darren\.platformio\penv\lib\site-packages\platformio\commands\run\command.py", line 134, in cli
    process_env(
  File "C:\Users\darren\.platformio\penv\lib\site-packages\platformio\commands\run\command.py", line 176, in process_env
    result = {"env": name, "duration": time(), "succeeded": ep.process()}
  File "C:\Users\darren\.platformio\penv\lib\site-packages\platformio\commands\run\processor.py", line 67, in process
    result = init_platform(self.options["platform"]).run(
  File "C:\Users\darren\.platformio\penv\lib\site-packages\platformio\commands\platform.py", line 322, in init_platform
    return PlatformFactory.new(name)
  File "C:\Users\darren\.platformio\penv\lib\site-packages\platformio\platform\factory.py", line 52, in new
    pkg = PlatformPackageManager().get_package(pkg_or_spec)
  File "C:\Users\darren\.platformio\penv\lib\site-packages\platformio\package\manager\base.py", line 237, in get_package
    if not self.test_pkg_spec(pkg, spec):
  File "C:\Users\darren\.platformio\penv\lib\site-packages\platformio\package\manager\base.py", line 255, in test_pkg_spec
    if os.path.realpath(spec.url) == os.path.realpath(pkg.path) or (
  File "C:\Program Files\Python39\lib\ntpath.py", line 651, in realpath
    path = _getfinalpathname_nonstrict(path)
  File "C:\Program Files\Python39\lib\ntpath.py", line 601, in _getfinalpathname_nonstrict
    path = _getfinalpathname(path)
FileNotFoundError: [WinError 161] The specified path is invalid: '\\\\VAULT\\Work_Darren\\Marlin\\https:\\github.com\\p3p\\pio-nxplpc-arduino-lpc176x\\archive\\0.1.3.zip'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

The terminal process "C:\Users\darren\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'LPC1768'" terminated with exit code: 1.

adrianomazotti, 2021-11-05:
Same problem here

rLhRw4fjX, 2021-11-09:
Same problem

adrianomazotti, 2021-11-09:
I fixed the problem on my PIO installation. It was easier than I thought.
Exit the VS code, delete the folder C:/Users/Your_folder/.platformio and open the VS code again. The VS code identified the missing folder and started the installation again. After that everything is working perfectly. I hope it works for you.

rLhRw4fjX, 2021-11-10:
adrianomazotti, It works, thanks

jbuszkie, 2021-11-10:
I have a similar issue.... it doesn't like my github lib_dep.
I have this in my ini ->   https://github.com/strandaster/telnetspy.git#master
When I try to compile I get this error:
FileNotFoundError: [WinError 161] The specified path is invalid: '\\\\192.168.2.30\\code\\ESP32\\ESP32_Base_code\\git+https:\\github.com\\strandaster\\telnetspy.git#master':

This is on a VM that I built today and with the code on a network drive.  
If I copy the whole folder over to a local drive it works.  I have another VM with the same software (for the most part) and it compiles fine with the networks drive??  Both 5.2.3 and 3.4.0 with python at 3.9.8

Why does it work on one VM and not the other?

And should I create my own issue for this?

ivankravets, 2021-11-10:
@jbuszkie could you share your `platformio.ini`?

jbuszkie, 2021-11-10:
There's not much too it...
`[env]
lib_deps =
  # Using a library name
  ;Wire
  ;OneWireHub
  ESP8266 Influxdb
  ;Adafruit HTU21DF Library
  ;Adafruit Unified Sensor
  ;RingBuf
  ;TelnetSpy@master
  ;https://github.com/yasheena/telnetspy#master
  https://github.com/strandaster/telnetspy.git#master
  ;https://github.com/jbuszkie/telnetspy.git#v1.1
  ;https://github.com/wizard97/SimplyAtomic.git

[env:nodemcu-32s]
platform = espressif32
board = nodemcu-32s
framework = arduino

upload_protocol = espota
;upload_port = 192.168.2.102
upload_port = ESP32_jb.local
upload_flags = 
 --auth="xxx"
 --port=3232

monitor_speed = 115200

[env:d1]
platform = espressif8266
board = d1
framework = arduino

; set frequency to 80MHz
board_build.f_cpu = 80000000L

; set frequency to 160MHz
;board_build.f_cpu = 160000000L

upload_speed = 921600

upload_protocol = esptool
;upload_port = COM12
;upload_protocol = espota
;upload_port = 192.168.2.60
;upload_port = ESP8266_jb.local
;upload_flags = 
; --auth="xxx"
; --port=8266

build_flags = 
 ; -Wl,-Teagle.flash.4m3m.ld
 ; -DDEBUG_ESP_PORT=Serial
 ; -DDEBUG_ESP_WIFI

monitor_speed = 115200
`

ivankravets, 2021-11-10:
No, triple backticks for code formatting.

bizzehdee, 2021-11-10:
> I fixed the problem on my PIO installation. It was easier than I thought. Exit the VS code, delete the folder C:/Users/Your_folder/.platformio and open the VS code again. The VS code identified the [...]

Tried this and it made no difference, still having the same issue

bizzehdee, 2021-11-10:
Figured out the issue and created pull request #4106

jbuszkie, 2021-11-10:
I downgraded python to 3.7.9 and it compiles fine!  so something with the UNC path.

jbuszkie, 2021-11-10:
> Figured out the issue and created pull request #4106

@bizzehdee 
Dumb question...  How would I try your fork to see if it fixes my issue as well?
How Do I load a non released version?  I normally use the extension manager.

bizzehdee, 2021-11-10:
> > Figured out the issue and created pull request #4106
> 
> @bizzehdee Dumb question... How would I try your fork to see if it fixes my issue as well? How Do I load a non released version? I normally use the extension manager.

I manually patched my official install by opening up C:\Users\myusername\.platformio\penv\Lib\site-packages\platformio\package\manager\base.py and making the changes to that

jbuszkie, 2021-11-10:
I can confirm that @bizzehdee's patch fixes my issue as well.  Now why it works on my other VM with 3.9.8 python installed, I'll never know!

ivankravets, 2021-11-12:
Resolved in https://github.com/platformio/platformio-core/commit/4687665ff3706f02fae8cd1ea220ae502a4f6ec9

Please re-test with `pio upgrade --dev`.

Does it work now?

closed 2021-11-12 by ivankravets

jbuszkie, 2021-11-12:
how do I downgrade?  I want to check that it's broken with 5.2.3 and then fixed again with 5.2.4a3

ivankravets, 2021-11-12:
Please remove `%HOME%/.platformio/penv` folder and restart VSCode.

jbuszkie, 2021-11-12:
Nevermind..  I just uninstalled pio and deleted everything on my .platformoio directory in my home directory.  Then installed pio and 5.2.3 installed.

jbuszkie, 2021-11-12:
I can confirm 5.2.3 fails.
I can confirm 5.2.4a3 works.

Thanks for the quick fix in the repo!

jbuszkie, 2021-11-12:
> Please remove `%HOME%/.platformio/penv` folder and restart VSCode.

lol..  I figured it out just as your message came across! :-)

mirceastoian, 2021-11-18:
I have the same issue, trying to compile Marlin for one of my printers and the files are located on a network share.
I'm now on 5.2.4a3, deleted penv folder and restarted VSCode.
Deleting the entire .platformio folder didn't helped either.
I have Python 3.9.6.
Anything else I can try?

jbuszkie, 2021-11-18:
I'm certainly no expert...  But is the error the same as mine?

phil31, 2021-11-26:
> Tried this and it made no difference, still having the same issue

after an hard disk failure and a new fresh install, i had this UNC path error.
applying a copy of my old .platformio directory solved all my compilation errors !

thanls, regards