platformio/platformio-core #4102: Bug on the generation of CMakeListsPrivate.txt definitions - not escaping spaces

pedrosousabarreto, 2021-11-05:
### Configuration

**Operating system**: 
Ubuntu  21.04
cmake version 3.20.2

Using CLion 2021.2.3

**PlatformIO Version** (`platformio --version`):
5.2.2

### Description of problem
After the upgrade to 5.2.2, the generation of CMakeListsPrivate.txt is not escaping spaces in add_definitions values of the CMAKE_BUILD_TYPE blocks.
Apparently this version no longer wraps the values in single quotes.

#### Steps to Reproduce

1. update PIO to 5.2.5
2. execute "pio project init" OR run the "re-init" command from CLion's PlatformIO tools menu

CMake execution log now returns missing files errors:
```
 
/snap/clion/169/bin/cmake/linux/bin/cmake -DCMAKE_BUILD_TYPE=esp32 -DCMAKE_DEPENDS_USE_COMPILER=FALSE -G "CodeBlocks - Unix Makefiles" /home/pedro/Code/Arduino/HomeAlarmConsole
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pedro/Code/Arduino/HomeAlarmConsole

Problems were encountered while collecting compiler information:
	xtensa-esp32-elf-g++: error: ESP32: No such file or directory
	xtensa-esp32-elf-g++: error: DEVKIT: No such file or directory
	xtensa-esp32-elf-g++: error: V1 -std=gnu++11: No such file or directory
 
```
### Actual Results

Example of file generated by v5.2.2
Notice the missing single quotes after -D and not escaped spaces on the last line:

```
if (CMAKE_BUILD_TYPE MATCHES "esp32")
    add_definitions(-DPLATFORMIO=50202)
    add_definitions(-DARDUINO_ESP32_DEV)
    add_definitions(-DVERSION=0.1.2)
    add_definitions(-DDEBUG)
    add_definitions(-DESP32)
    add_definitions(-DESP_PLATFORM)
    add_definitions(-DF_CPU=240000000L)
    add_definitions(-DHAVE_CONFIG_H)
    add_definitions(-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\")
    add_definitions(-DARDUINO=10805)
    add_definitions(-DARDUINO_ARCH_ESP32)
    add_definitions(-DARDUINO_VARIANT=\"doitESP32devkitV1\")
    add_definitions(-DARDUINO_BOARD=\"DOIT ESP32 DEVKIT V1\")
...
```

### Expected Results

Same file generated by v5.1.1.
Notice the single quotes wrapping the flag values:

```
if (CMAKE_BUILD_TYPE MATCHES "esp32")
    add_definitions(-D'PLATFORMIO=50101')
    add_definitions(-D'ARDUINO_ESP32_DEV')
    add_definitions(-D'VERSION=0.1.2')
    add_definitions(-D'DEBUG')
    add_definitions(-D'ESP32')
    add_definitions(-D'ESP_PLATFORM')
    add_definitions(-D'F_CPU=240000000L')
    add_definitions(-D'HAVE_CONFIG_H')
    add_definitions(-D'MBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"')
    add_definitions(-D'ARDUINO=10805')
    add_definitions(-D'ARDUINO_ARCH_ESP32')
    add_definitions(-D'ARDUINO_VARIANT=\"doitESP32devkitV1\"')
    add_definitions(-D'ARDUINO_BOARD=\"DOIT ESP32 DEVKIT V1\"')
...
```

CMake works well if I escape the spaces or wrap the value in single quotes.

Both variants work:
```
add_definitions(-D'ARDUINO_BOARD=\"DOIT ESP32 DEVKIT V1\"')
add_definitions(-DARDUINO_BOARD=\"DOIT\ ESP32\ DEVKIT\ V1\")
```

The generated one fails:
```
add_definitions(-DARDUINO_BOARD=\"DOIT ESP32 DEVKIT V1\")
```

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini

[common]
framework = arduino
monitor_filters = time, default, send_on_enter, esp32_exception_decoder
upload_port = /dev/ttyUSB2
monitor_port = /dev/ttyUSB2
upload_speed = 460800
monitor_speed = 115200
lib_deps = 
	256dpi/MQTT@^2.5.0
	bblanchon/ArduinoJson@^6.18.3
	sstaub/Ethernet3 @ ^1.5.5
	ArduinoOTA=https://github.com/jandrassy/ArduinoOTA

build_flags = 
	-D VERSION=0.1.2
	-D DEBUG

[env:esp32]
extends = common
platform = espressif32
board = esp32doit-devkit-v1
monitor_filters = time, default, send_on_enter, esp32_exception_decoder

```

ivankravets, 2021-11-05:
It seems this issue is caused by https://github.com/platformio/platformio-core/pull/4085

@elmot, do you have any ideas?

@pedrosousabarreto, could you upgrade CLion to 2021.3?

pedrosousabarreto, 2021-11-05:
@ivankravets Updated to Clion 2021.3 from snap edge channel and still have the same issue.

This is the generated line (same as before):
`add_definitions(-DARDUINO_BOARD=\"DOIT ESP32 DEVKIT V1\")`

Clion 30.21.3 EAP
Build #CL-213.5605.4, built on November 2, 2021

pedrosousabarreto, 2021-11-05:
Added a comment on #4085 with a possible fix. Sorry I can't test it right now.

https://github.com/platformio/platformio-core/pull/4085/files#r743905480

PR #4105 (2021-11-09): Escape spaces in CLion CMakeListsPrivate template - FIXES #4102
This fix adds spaces to the regex substitutions on CMakeListsPrivate.txt add_definitions.

Fixes #4102
commit 2786bfbe (2021-11-09): Escape spaces in CLion CMakeListsPrivate template - FIXES #4085 (#4105)

This fix adds spaces to the regex substitutions on CMakeListsPrivate.txt add_definitions.

Fixes #4102
commit 7d78e4a6 (2021-11-09): Fixed an issue with the CLion project generator when a macro contains a space // Resolve #4102
commit 4687665f (2021-11-12): Improved support for projects located on a network share // Resolve #3417 , Resolve #3926 , Resolve #4102
closed 2021-11-09 by ivankravets