platformio/platformio-core #4104: RPi Zero | AttributeError: module 'serial.tools.miniterm' has no attribute 'Transform'

unitythemaker, 2021-11-08:
What kind of issue is this?

- [x] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

------------------------------------------------------------------

### Configuration

**Operating system**: ``Raspbian (Linux raspberrypi 5.10.63+ #1459 Wed Oct 6 16:40:27 BST 2021 armv6l GNU/Linux)``

**PlatformIO Version** (`platformio --version`):
```
pi@raspberrypi:~/ESProject $ pio --version
PlatformIO Core, version 5.2.3
pi@raspberrypi:~/ESProject $ python3 --version
Python 3.7.3
pi@raspberrypi:~/ESProject $ pip3 --version
pip 21.3.1 from /usr/local/lib/python3.7/dist-packages/pip (python 3.7)
```
### Description of problem
Not compatible with RPi Zero WH. I've tried re-installing the package.

#### Steps to Reproduce

1. Get a RPi Zero W
2. Upgrade the system (``sudo apt update && sudo apt upgrade -y``)
3. Install the pre-requisties (``sudo apt-get install python3 python3-pip``)
4. Install the platformio package (``sudo python3 -m pip install -U platformio``)
5. (Optional) Create a new project (``mkdir ESProject && cd ESProject && platformio init -b esp01``)
6. (Optional) Try putting some code and running ``platformio run`` or ``sudo pio run``
7. It will not only fail with run but with any platformio command.

### Actual Results
```
# no change when I try without sudo
pi@raspberrypi:~/ESProject $ sudo pio --help
Error: Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/platformio/__main__.py", line 129, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 1052, in main
    with self.make_context(prog_name, args, **extra) as ctx:
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 914, in make_context
    self.parse_args(ctx, args)
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 1615, in parse_args
    rest = super().parse_args(ctx, args)
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 1370, in parse_args
    value, args = param.handle_parse_result(ctx, opts, args)
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 2347, in handle_parse_result
    value = self.process_value(ctx, value)
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 2309, in process_value
    value = self.callback(ctx, self, value)
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 1270, in show_help
    echo(ctx.get_help(), color=ctx.color)
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 693, in get_help
    return self.command.get_help(self)
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 1295, in get_help
    self.format_help(ctx, formatter)
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 1326, in format_help
    self.format_options(ctx, formatter)
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 1524, in format_options
    self.format_commands(ctx, formatter)
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 1588, in format_commands
    cmd = self.get_command(ctx, subcommand)
  File "/usr/local/lib/python3.7/dist-packages/platformio/commands/__init__.py", line 64, in get_command
    mod = __import__(mod_path, None, None, ["cli"])
  File "/usr/local/lib/python3.7/dist-packages/platformio/commands/ci.py", line 25, in <module>
    from platformio.commands.run.command import cli as cmd_run
  File "/usr/local/lib/python3.7/dist-packages/platformio/commands/run/command.py", line 25, in <module>
    from platformio.commands.device.command import device_monitor as cmd_device_monitor
  File "/usr/local/lib/python3.7/dist-packages/platformio/commands/device/__init__.py", line 15, in <module>
    from platformio.commands.device.filters.base import DeviceMonitorFilter
  File "/usr/local/lib/python3.7/dist-packages/platformio/commands/device/filters/base.py", line 20, in <module>
    class DeviceMonitorFilter(miniterm.Transform):
AttributeError: module 'serial.tools.miniterm' has no attribute 'Transform'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

pi@raspberrypi:~/ESProject $
```

### Expected Results
Expected platformio command to not give ``AttributeError: module 'serial.tools.miniterm' has no attribute 'Transform'`` error.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp01]
platform = espressif8266
board = esp01
framework = arduino
```

**Source file to reproduce issue:**
```cpp
/*
  Blink

  Turns an LED on for one second, then off for one second, repeatedly.

  Most Arduinos have an on-board LED you can control. On the UNO, MEGA and ZERO
  it is attached to digital pin 13, on MKR1000 on pin 6. LED_BUILTIN is set to
  the correct LED pin independent of which board is used.
  If you want to know what pin the on-board LED is connected to on your Arduino
  model, check the Technical Specs of your board at:
  https://www.arduino.cc/en/Main/Products

  modified 8 May 2014
  by Scott Fitzgerald
  modified 2 Sep 2016
  by Arturo Guadalupi
  modified 8 Sep 2016
  by Colby Newman

  This example code is in the public domain.

  https://www.arduino.cc/en/Tutorial/BuiltInExamples/Blink
*/

// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
}
```

### Additional info
- I think the source of that issue is RPi Zero has an armv6l processor instead of an armv7l+ one.
- Keep up the good work!

ivankravets, 2021-11-08:
Hi,

1. Please DON'T USE `sudo` for any operations related to PlatformIO. We are not responsible for 3rd party toolchains.
2. Please DO NOT install PlatformIO globally into the system. Use installer script instead https://docs.platformio.org/en/latest/core/installation.html#installation-methods

So, please remove ALL previous PlatformIO Core ( see https://docs.platformio.org/en/latest/faq.html#multiple-platformio-cores-in-a-system ) and install PlatformIO Core using the installer script mentioned above. It seems you will also need to use `sudo` to remove `~/.platformio` folder (but just to remove it).

unitythemaker, 2021-11-08:
> Hi,
>
>     1. Please DON'T USE `sudo` for any operations related to PlatformIO. We are not responsible for 3rd party toolchains.
>
>     2. Please DO NOT install PlatformIO globally into the system. Use [...]

Thanks, it worked and that error is gone but I got a new error now.
I re-installed package and did what you said.

Tell me if I need to create a new issue about this.

Then created a new folder, entered into it, ran ``pio init -b esp01``. Added this code: https://github.com/platformio/platform-espressif8266/blob/master/examples/arduino-blink/src/main.cpp
Ran ``pio run -t upload --upload-port /dev/serial0`` after putting ESP01 in programming mode.
This problem is not Pi Zero specific, I got the same error on my RPi3B. However, I was able to program ESP via Arduino.

Log
```
pi@raspberrypi:~/ESP $ pio run -t upload --upload-port /dev/serial0
Processing esp01 (platform: espressif8266; board: esp01; framework: arduino)
---------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01.html
PLATFORM: Espressif 8266 (3.2.0) > Espressif Generic ESP8266 ESP-01 512k
HARDWARE: ESP8266 80MHz, 80KB RAM, 512KB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.30002.0 (3.0.2)
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - tool-mklittlefs 1.203.210628 (2.3)
 - tool-mkspiffs 1.200.0 (2.0)
 - toolchain-xtensa 2.100300.210717 (10.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 35 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Retrieving maximum program size .pio/build/esp01/firmware.elf
Checking size .pio/build/esp01/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [===       ]  34.2% (used 27980 bytes from 81920 bytes)
Flash: [======    ]  60.1% (used 260893 bytes from 434160 bytes)
Configuring upload protocol...
AVAILABLE: espota, esptool
CURRENT: upload_protocol = esptool
Looking for upload port...
Use manually specified: /dev/serial0
Uploading .pio/build/esp01/firmware.bin
esptool.py v3.0
Serial port /dev/serial0
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP8266: Timed out waiting for packet header
*** [upload] Error 2
========================== [FAILED] Took 38.60 seconds ==========================
```

Platform IO config:
```ini
[env:esp01]
platform = espressif8266
board = esp01
framework = arduino
```

ivankravets, 2021-11-08:
It works, great! 

> A fatal error occurred: Failed to connect to ESP8266: Timed out waiting for packet header

Please forward to https://community.platformio.org/

closed 2021-11-08 by ivankravets