platformio/platformio-core #4107: Building fails if the board is not currently connected.

HubKing, 2021-11-12:
### Configuration

**Operating system**: Windows 10.

**PlatformIO Version** (`platformio --version`): 2.4.0 (2021-11-05)

### Description of problem
Building, not uploading, fails if the board is not currently connected. And it shows a message like "Please specify `upload_port` for environment", even though `upload_port` is present in the platformio.ini. 

I think that it is unlikely that it really needs the board connected to the computer just to build the project. It might be that it needs some sort of board information other than the specified information in the platformio.ini to build the project, but even if that were the case, only the very first build should fail, because, in my opinion, it should cache the necessary information on the computer, instead of retrieving it from the board every time, because, as far as I know, that information would not change.

#### Steps to Reproduce

1. Configure a project for a board.
2. Disconnect the board.
3. Press the Build button.

### Actual Results
Error: Please specify `upload_port` for environment or use global `--upload-port` option.

### Expected Results
Build succeeds.

ivankravets, 2021-11-12:
No, we don't require any board to build a project. Please ask help here https://community.platformio.org/

closed 2021-11-12 by ivankravets

HubKing, 2021-11-12:
I did some more testing, and strangely it does not happen in a new project I created for this testing. But this does occur with multiple existing projects as the screenshot below. There must be some condition that causes this... I will add it later, if I find the condition.

[image]

ivankravets, 2021-11-12:
Which task do you use? You need to use BUILD task, not UPLOAD.

HubKing, 2021-11-12:
> Which task do you use? You need to use BUILD task, not UPLOAD.

I clicked the check mark button on the status bar, and I made sure that it was not the right arrow button.

ivankravets, 2021-11-12:
Could you click on the terminal icon (bottom status bar) and type `pio run`. Does it still ask about the upload port?

HubKing, 2021-11-12:
> 
> 
> Could you click on the terminal icon (bottom status bar) and type `pio run`. Does it still ask about the upload port?

Yeah, it looks like so.

https://user-images.githubusercontent.com/10900489/141530748-7d48773b-6650-400a-9379-1f85599fbe28.mp4

ivankravets, 2021-11-12:
Could you share the ZIP archive with this project to reproduce the issue?

HubKing, 2021-11-12:
> 
> 
> Could you share the ZIP archive with this project to reproduce the issue?

I think I found the reason. I am sorry for having wasted your time. I had followed [this post about resetting ESP within PlatformIO](https://community.platformio.org/t/is-there-a-way-to-reset-restart-reboot-the-board-within-the-ide/17684), and added 

    extra_scripts = reset.py

to the platformio.ini file. Removing that line and the script solved the problem. I did not know that the script is supposed to automatically run; I thought that I have to click the script manually under the "Project Tasks" tree to run it.

ivankravets, 2021-11-12:
Great that you resolved this issue!

Happy coding with PlatformIO!