platformio/platformio-core #4108: Large Project Initialisation is slow/non-completing for CLion

SJ-Innovation, 2021-11-13:
Creating an issue to keep track of my debugging process. 

I develop for the Marlin 3D printer firmware. We use a large platformio.ini file to define which files each of our different environments use. I personally develop using CLion, and have used the CLion initialiser and plugin in the past with great success. 
When initialisating the CLion environment using `pio init --ide clion` for Marlin dev it starts, and never finishes. 

I have narrowed the cause of this to the template generator for `CMakeListsPrivate.txt` which is generated using a Bottle template from the template variables. Of which in the Marlin environment, due to our large platformio.ini size, is quite large. This means the call to `bottle.template` (`platformio/project/generator.py L:161`) in which the template variables are expanded into the template itself, takes _forever_. 

Im not sure how much _forever_ as ive never seen it complete. It doesnt error, it just sits there screaming on one core... RAM usage does not increase, suggesting its not leaking anything, but also doesnt really help metricise how _done_ it is... 

I reckon this is because of how the template variables are expanded as kwargs, and because theyre large, bottle.template is choking a little bit. 

I am investigating further, if anyone has any 2-cents theyd like to drop id be all ears...
 
- Have a good one :)

SJ-Innovation, 2021-11-13:
Okay, instrumenting has revealed the issue is in the `CMakeListsPrivate.txt.tpl` file, `load_project_ide_data` where `leftover_envs` is quite large

SJ-Innovation, 2021-11-13:
Current workaround:
`CMakeListsPrivate.txt.tpl L:96` prepend `pass ##` to comment out and properly indent so no further ide data is loaded.
 Means I lose the data for unused environments, switching between them in CLion requires a `pio init` after changing default environment to the one i want to use, but this means i can develop in the environment i was planning on using anyway. 

Will continue searching for suitable fix.

ivankravets, 2021-11-14:
>  Marlin 3D printer firmware

Just remove a ton of envs from `platformio.ini` and use only one which you need.

closed 2021-11-14 by ivankravets

SJ-Innovation, 2021-11-14:
Yeah i'd gathered that. I'm working on a proper solution.