platformio/platformio-core #4110: Platformio is not building when referencing workspace_dir and core_dir on new Version 5.2.3

tfir1981, 2021-11-16:
What kind of issue is this?

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Ubuntu 20.04 LTS

**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 5.2.3

### Description of problem
After my PlatformIO updates from 5.2.2 to current 5.2.3 my Project cannot build anymore.

```bash
warning: Calling missing SConscript without error is deprecated.
Transition by adding must_exist=False to SConscript calls.
Missing SConscript '.piocore/platforms/ststm32/builder/.piocore/packages/framework-zephyr/scripts/platformio/platformio-build-pre.py'
File "/<PathToMyProject>/.piocore/platforms/ststm32/builder/main.py", line 110, in <module>
```

It seems to me, that there are two paths connected `.piocore/platforms/ststm32/builder/` and `.piocore/packages/framework-zephyr/scripts/platformio/platformio-build-pre.py`. Actually i could not find the location where the paths where connected. In `platformio-build-pre.py` i added some prints, but there where the paths right.

#### Steps to Reproduce

What i found out is, that it has something to do with the parameter `workspace_dir` and `core_dir` of my ini-File. 

**This config runs fine**
```ini
[platformio]
workspace_dir = .pio
core_dir = .piocore
```

**This not:**
```ini
[platformio]
workspace_dir = ${test.workspace_dir}
core_dir = ${test.core_dir}

[test]
workspace_dir = .pio
core_dir = .piocore
```

ivankravets, 2021-12-02:
Thanks for the report! Please re-test with `pio upgrade --dev`.

Does it work now?

closed 2021-12-02 by ivankravets

tfir1981, 2021-12-02:
I did an update with `pio upgrade --dev` and it works.
Thank you!

commit 014090c4 (2021-12-02): Fixed an issue when referencing "*_dir" option from a custom project configuration environment // Resolve #4110