platformio/platformio-core #4114: ZSH completions don't work

7ocb, 2021-11-20:
I did a "pio system completion install --shell zsh", and when I enter "pio" and hit tab, nothing is completed.

It's installed:

Versions:

❯ pio --version
PlatformIO Core, version 5.2.4a4
❯ zsh --version
zsh 5.8 (x86_64-pc-linux-gnu)
❯ uname -a
Linux elkbook 5.15.2-arch1-1 #1 SMP PREEMPT Fri, 12 Nov 2021 19:22:10 +0000 x86_64 GNU/Linux

❯ env COMMANDLINE="pio" _PIO_COMPLETE=complete-zsh  pio
❯ python --version
Python 3.9.7

ivankravets, 2021-12-02:
Thanks for the report! We totally rewrote tab completion support. Please remove any code related to `pio` from your `.zshrc` before installing new code.

1. Remove `pio` related code from `~/.zshrc`
2. Upgrade to the latest PIO Core via `pio upgrade --dev`
3. `pio system completion install zsh`
4. Restart shell.

Does it work now?

closed 2021-12-02 by ivankravets

7ocb, 2021-12-03:
@ivankravets yes, it looks like it working, thanks!

7ocb, 2021-12-03:
@ivankravets 

[image]

Looks that for `pio run` it completes options that are not `run`'s?

ivankravets, 2021-12-03:
These options are global per any `pio` command. Should it be a bug of the Click - Python library? We use it for CLI and for tab completion.

jcw, 2021-12-04:
On MacOS Monterey, using bash, I get:

> RuntimeError: Shell completion is not supported for Bash versions older than 4.4.

ivankravets, 2021-12-04:
@jcw Latest macOS versions do not have `bash` shell and use `zsh` instead. Could you remove any `pio` tracks from `~/.zshrc` and try `pio system completion install zsh` using the latest development version PlatformIO Core?

jcw, 2021-12-04:
I'm used to bash, so I've changed my login:

```
$ sh --version
GNU bash, version 3.2.57(1)-release (arm64-apple-darwin21)
Copyright (C) 2007 Free Software Foundation, Inc.
```

It's an old one, clearly (due to licensing issues). I have no `~/.zshrc`.
Maybe I'll need to switch, but for now, I'll just pass on PIO completion.
Let me just note that shell completions for other commands work just fine.

ivankravets, 2021-12-04:
@jcw This is not our limitation, we use Python's Click library. It seems they don't have support for older than 4.4 Bash version.

P.S: Highly recommend to check [Fish shell](https://fishshell.com/), I've been using it for the last 5 years. Crazy cool.

jcw, 2021-12-04:
> Crazy cool.

Ok, will do, Ivan :) - I've tried Fish long ago, but will revisit - the point is that I regularly log onto servers and RasPi's. Having bash on all of 'em simplifies my life a little.

ivankravets, 2021-12-04:
>  I regularly log onto servers and RasPi's

You can change the default shell at these devices as well (see `chsh` command ) :)

jcw, 2021-12-04:
Heh, I _knew_ you'd say that. Sure, I could (and might), but a little bit of pushback is in order here, IMO: a tool such as PIO should be invariant w.r.t. shell choice (just as it is to editor choice, platform choice, etc). I understand that in this case it's beyond your control, and as I said: no big deal for me (I use short aliases for some key PIO cmds).

commit 55f8471a (2021-12-02): Improved tab completion support for Bash, ZSH, and Fish shells // Resolve #4114