platformio/platformio-core #4116: PlatformIO 5.2.3+clion+MinGW无法初始化esp32项目

sddtly, 2021-11-21:
PlatformIO 5.2.3+clion+MinGW无法初始化esp32项目,
提示:
xtensa-esp32-elf-g++.exe: error: IoT: No such file or directory
xtensa-esp32-elf-g++.exe: error: Uno -std=gnu++11: No such file or directory

`"C:\Program Files\JetBrains\CLion 2021.2\bin\cmake\win\bin\cmake.exe" -DCMAKE_BUILD_TYPE=esp32vn-iot-uno -DCMAKE_DEPENDS_USE_COMPILER=FALSE -G "CodeBlocks - MinGW Makefiles" F:\project\CLionProjects\esp32_arduino\untitled1
-- The C compiler identification is GNU 8.1.0
-- The CXX compiler identification is GNU 8.1.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - failed
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: F:/project/CLionProjects/esp32_arduino/untitled1/cmake-build-esp32vn-iot-uno

收集编译器信息时遇到问题:
	xtensa-esp32-elf-g++.exe: error: IoT: No such file or directory
	xtensa-esp32-elf-g++.exe: error: Uno -std=gnu++11: No such file or directory

[已完成]
`

NINO-3D, 2021-11-21:
> PlatformIO 5.2.3+clion+MinGW无法初始化esp32项目, 提示: xtensa-esp32-elf-g++.exe: error: IoT: No such file or directory xtensa-esp32-elf-g++.exe: error: Uno -std=gnu++11: No such file or directory [...]
完成] `

ivankravets, 2021-11-22:
Please complete all steps described in https://docs.platformio.org/en/latest/integration/ide/clion.html

Especially, "CLion does not load project build environments from “platformio.ini”".

closed 2021-11-22 by ivankravets