platformio/platformio-core #4117: Network problem on Windows

koxo, 2021-11-21:
Hi,
I want to report a problem on projects using a network drive on Windows.
When i create my project, i tell to use a folder on a network drive  :  example : "X:\MyProject".
In "c_cpp_project" i have : 

```
//
// !!! WARNING !!! AUTO-GENERATED FILE!
// PLEASE DO NOT MODIFY IT AND USE "platformio.ini":
// https://docs.platformio.org/page/projectconf/section_env_build.html#build-flags
//
{
    "configurations": [
        {
            "name": "PlatformIO",
            "includePath": [
                "**/srv-nas/home/MyProject**/MyProject/include",
                "**/srv-nas/home/MyProject**/MyProject/src",
                "C:/VSCode/platformio/packages/framework-arduino-avr/cores/arduino",
                ....................................................................................
                ""
            ],
            "browse": {
                "limitSymbolsToIncludedHeaders": true,
                "path": [
                    "**/srv-nas/home/MyProject/MyProject**/include",
                    "**/srv-nas/home/MyProject/MyProject**/src",
                    "C:/VSCode/platformio/packages/framework-arduino-avr/cores/arduino",
                    ....................................................... 
                    ""
                ]
            },
.............................................................

```
So i get squiggles in the editor.

If add a library, i get :  "/srv-nas/home/MyProject/MyProject/.pio/libdeps/uno/LiquidCrystal/src" instead of "//srv-nas/home/MyProject/MyProject/.pio/libdeps/uno/LiquidCrystal/src" or "X:/MyProject/MyProject/.pio/libdeps/uno/LiquidCrystal/src".

If i modify the lines : "**/srv-nas/home/MyProject**/MyProject/include" into "**//srv-nas/home/MyProject**/MyProject/include" or "**X:/MyProject**/MyProject/include" it's good.

If i tell a drive path, the project should keep a drive path, if i enter a unc path it should use a unc path.

Thank you for your help

Best Regards

ivankravets, 2021-11-21:
Duplicate of https://github.com/platformio/platformio-core/issues/3926

Please re-test with `pio upgrade --dev`. Does it work now?

koxo, 2021-11-21:
Hi Ivan,
It's good now ! 
Thank you for all
Best regards
KoXo

koxo, 2022-01-29:
Hi Ivan
I have updated my version to 5.25a6 and the problem of network path is back !
Do you have forget something ?
Best regards
KoXo

ivankravets, 2022-02-03:
I've just re-tested and everything works. Please check that you don't have multiple PIO Cores in the system  https://docs.platformio.org/en/latest/faq.html#multiple-platformio-cores-in-a-system

koxo, 2022-02-03:
Hi Ivan,
I have make some tests this morning, in fact it's good with a network drive.
The remaining problem is for a network location
if i am a user in a classical domain with the folder "Documents" redirected to an UNC path like "\\server\usershare\documents", 
in c_cpp_propserties.json i get "/server/usershare/documents/platformio/MyTest" instead of "**/**/server/usershare/documents/platformio/MyTest". So i get squiggles and if i use libraries in my path, the autocompletion does not works correctly.
[https://imgur.com/GSJgiH0](url)
Thank you for your help.
KOXO

celsocos, 2023-04-20:
Hello. I also found problems with a network drive, and Ivan suggested to place information here so we can improve this point.

SCENARIO:
A virtual PC (local machine) with VS Code and Platform IO, Arduino project.
I have a network share "\\\ccsrv\Projetos\" and inside the network share a folder "Projects"

So the drive letter P: is mapped to shared folder "\\\ccsrv\Projetos" and in the c_cpp_properties.json we have the correct path using the P: drive letter, like “P:/Projects/Server Modbus/include” and similar.

No more problems with incomplete UNC path inside c_cpp_properties.json , that is good.

However, the CMD.EXE is started with the UNC path

‘\\\CCSRV\Projetos\Projects\Server Modbus’
CMD.EXE was started with the above path as the current directory.
UNC paths are not supported. Defaulting to Windows directory.

In my system, _the provided registry hack did not work_, and in fact the registry edit only supress the warnings. I expected the “current directory” to be ‘P:\Projects\Server Modbus’ with the cmd.exe, no UNC path.

But for some reason, the system still uses the UNC path in the complilation.
Suggestions:
1) Maybe use the drive letter and not UNC path with CMD.EXE may fix?
2) Even better: Maybe use the power shell instead of cmd.exe? Power shell does supports UNC path.

NOTE: the UNC path is very tricky, because it can be undestood as escape sequences. To display a proper text here I had to use three \ because of this, and had other problems with double \ in code too, The first \ seems to disapear like magic. If you add a quote or a dot the behavior can change. Similar with the text in this post that I had to fix using three \ to diplay just two \  and had problems with quotes too :)

Please let me know if I can help with something else.

koxo, 2023-04-20:
Hi,
Thank you for your response.
In a network environment with connected drive, i use this drive to open the project and it seems good.
I have look at the source of pio and i have found a problem in "fs.py" : 

**def to_unix_path(path):
    if not IS_WINDOWS or not path:
        return path
    return re.sub(r"[\\]+", "/", path)**

This function replaces the '\\' by '/', i think that the "+" is the problem in the reg expression.

**def to_unix_path(path):
    if not IS_WINDOWS or not path:
        return path
    return re.sub(r"[\\]", "/", path)**

i have try this : 
_import re

IS_WINDOWS = 1

def to_unix_path(path):
    if not IS_WINDOWS or not path:
        return path
    return re.sub(r"[\\]+", "/", path)

def to_unix_path2(path):
    if not IS_WINDOWS or not path:
        return path
    return re.sub(r"[\\]", "/", path)

print("*** no modification ****")

path = "P:\\Documents\\Platformio\\Projects\\Test"
path = to_unix_path(path)
print("local (OK) :", path)

path = "\\\\dc1\\ameunier$\\Documents\\Platformio\\Projects\\Test"
path = to_unix_path(path)
print("network (bad) :", path)

print("*** with modification ****")

path = "P:\\Documents\\Platformio\\Projects\\Test"
path = to_unix_path(path)
print("local (OK) :", path)

path = "\\\\dc1\\ameunier$\\Documents\\Platformio\\Projects\\Test"
path = to_unix_path2(path)
print("network (OK) :", path)_

-->

_*** no modification ****
local (OK) : P:/Documents/Platformio/Projects/Test
network (bad) : /dc1/ameunier$/Documents/Platformio/Projects/Test
*** with modification ****
local (OK) : P:/Documents/Platformio/Projects/Test
network (OK) : //dc1/ameunier$/Documents/Platformio/Projects/Test_

Best regards.
Koxo

ivankravets, 2023-04-20:
We have significantly improved support for network drives on Windows in the latest development version. See https://github.com/platformio/platformio-core/issues/3417

Please re-test with `pio upgrade --dev`.

Does it work now without any issues?

koxo, 2023-04-20:
Hi Ivan,

I have see the beta update. 

In fact you have modify function "to_unix_path" : 

**def to_unix_path(path):
    if not IS_WINDOWS or not path:
        return path
    return path.replace("\\", "/")**

So it's a little similar as my post above. (note that the two backslash of the reg expression were remove by github).

So it's good for me.

Thank you.

Koxo

commit df896ad4 (2023-04-20): Fix normalizing path to unix // Resolve #4117