platformio/platformio-core #4118: When pio "Update all", tool-scons Incompatible 4.40300.1 error occurs

TurnOffNOD, 2021-11-22:
What kind of issue is this?

- [x] **PlatformIO Core**.

### Configuration

**Operating system**:
Windows 10 64 bit.

**PlatformIO Version** (`platformio --version`):
PlatformIO Core, version 5.2.3

### Description of problem
I am using platformio on vscode. When I press the button "Updates->Update all", it will execute "pio update", the terminal will output its information as follows:

```cmd
C:\Users\(myusername)pio update
Updating platformio/contrib-piohome           3.4.0 @ ~3.4.0                     [Up-to-date]
Updating platformio/tool-scons                4.40200.0 @ ~4.40200.0             [Incompatible 4.40300.1]

Platform Manager
================
Platform atmelavr
--------
Updating platformio/atmelavr                  3.4.0                              [Up-to-date]
Updating platformio/toolchain-atmelavr        1.70300.191015 @ ~1.70300.0        [Up-to-date]
Updating platformio/framework-arduino-avr     5.1.0 @ ~5.1.0                     [Up-to-date]
Updating platformio/framework-arduino-avr-minicore 2.1.3 @ ~2.1.0                     [Up-to-date]
Updating platformio/tool-avrdude              1.60300.200527 @ ~1.60300.0        [Up-to-date]

Library Manager
===============
Library Storage: D:\Programs\PlatformIO\lib
```

### Actual Results
`Updating platformio/tool-scons                4.40200.0 @ ~4.40200.0             [Incompatible 4.40300.1]`
There shows error.

### Expected Results
No error.

### Additional info
This error is showing up when executing "pio update", so I think it is a pio core problem and commit issue here.

ivankravets, 2021-11-22:
This is normal behavior. There is a new version of `tool-scons`, but support for it was added in PlatformIO Core 5.2.4 that has not been released yet.

You can switch to the development version via `pio upgrade --dev`.

closed 2021-11-22 by ivankravets

credomane, 2021-11-22:
So....platform.io "stable" is more-or-less broken is what you are tell us? I can't build anything marlin as `platformio/tools-scon` won't install because it is incompatible with the "stable" version of platform.io. 

`C:\users\credomane\.platformio\python3\python.exe: can't open file 'C:\users\credomane\.platformio\packages\tool-scons\scons.py': [Errno 2] No such file or directory`

Platform.io should really install the newest compatible version of packages rather than just flat out refusing to install at all. I don't mine updating to the dev version to resolve this but it was a frustrating weekend of finding years old issues that don't apply anymore. Then today I find this closed github issue.

TurnOffNOD, 2021-11-23:
> This is normal behavior. There is a new version of `tool-scons`, but support for it was added in PlatformIO Core 5.2.4 that has not been released yet.
> 
> You can switch to the development version via `pio upgrade --dev`.

Oh.
And when will next stable version (PlatformIO Core 5.2.4) be released?

TurnOffNOD, 2021-12-01:
> This is normal behavior. There is a new version of `tool-scons`, but support for it was added in PlatformIO Core 5.2.4 that has not been released yet.
> 
> You can switch to the development version via `pio upgrade --dev`.

Hi @ivankravets
When will next stable version (PlatformIO Core 5.2.4) be released? Is there any schedule?

ivankravets, 2021-12-01:
Hi @TurnOffNOD,

We plan 5.2.4 for this month. Do you have any issues with 5.2.3?

TurnOffNOD, 2021-12-01:
> Hi @TurnOffNOD,
> 
> We plan 5.2.4 for this month. Do you have any issues with 5.2.3?

@ivankravets 
OK, I'll wait.
The problem is why this issue is committed.

ivankravets, 2021-12-01:
This is not an issue. PlatformIO Core 5.2.3 depends on the previous version of SCons where the 5.2.4 release introduces support for SCons 4.3.

credomane, 2021-12-01:
> This is not an issue. PlatformIO Core 5.2.3 depends on the previous version of SCons where the 5.2.4 release introduces support for SCons 4.3.

But it *is* an issue. PlatformIO Core 5.2.3 wants to install the incompatible Scons 4.3 and not the previous SCons release that is compatible. Therefore anyone installing fresh or (in my case) automatic update of platform.io in VSCode are gonna get Platform.IO 5.2.3 with a broken SCons.

The automatic updates is how I ended up in my predicament and eventually found this ticket. Sure the force update to the dev version fixed the problem but it isn't good that platform.io can "break" itself in this fashion.

ivankravets, 2021-12-01:
@credomane there is no issue. PlatformIO Core 5.2.3 will install compatible SCons. Do you have any problems with PlatformIO Core 5.2.3? If you have broken `tool-scons` package, please remove it from `%HOME_DIR/.platformio/packages` and restart build.