platformio/platformio-core #4125: PLATFORMIO_*_DIR does not override config options

MitchBradley, 2021-12-01:
What kind of issue is this?

- [ X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

### Configuration

**Operating system**: Linux, Windows 10, probably any other since the bug is in the core Python code

**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 5.1.1

### Description of problem

The documentation says:
```
   PLATFORMIO_SRC_DIR¶
   Allows one to override “platformio.ini” (Project Configuration File) option src_dir.
```
"Override" implies that, if the environment variable is present, it should take precedence over the value in platformio.ini.

Instead, if **src_dir** is present in platformio.ini, its value is used and the environment variable is ignored.  Contrary to the documentation, the platformio.ini option overrides the environment variable.

This bug applies not only to PLATFORMIO_SRC_DIR, but also to every PLATFORMIO_*_DIR variable.

#### Steps to Reproduce

Using the minimal project shown below, run the following commands
1. pio run
2. PLATFORMIO_SRC_DIR=foobar pio run

Case 2 is shown in the Linux shell command form.  On Windows you have to use a different way to set the environment variable - but it fails there too.

### Actual Results

Both 1. and 2. succeed

### Expected Results

Case 1 should succeed.
Case 2 should fail because overriding the source directory with the nonexistent path "foobar" should result in "Nothing to buld".

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[platformio]
src_dir = src

[env:native]
platform = native
```

**Source file to reproduce issue:**
This file goes in **src/main.c**
```c
int main(int argc, char**argv) { }

```

### Additional info

The buggy code is [here](https://github.com/platformio/platformio-core/blob/414a194c9dd1cd19808f34ccb09f01a9b0801563/platformio/project/config.py#L274) .

It says
```
            elif envvar_value and value == MISSING:
```
but it should say
```
            elif envvar_value:
```

The reason is because line 252 of that file sets `value` from the option in the platformio.ini file, and the `value == MISSING` clause prevents it from being overridden.

ivankravets, 2021-12-02:
Thanks for the report! Please re-test with `pio upgrade --dev`.

Does it work now?

closed 2021-12-02 by ivankravets

MitchBradley, 2021-12-02:
Yes, works perfectly now.  Thanks for the quick resolution.

commit e40b251c (2021-12-02): Fixed a bug when the system environment variable does not override a project configuration option // Resolve #4125