platformio/platformio-core #4127: Aurix TC27x family support

JoseOlin, 2021-12-03:
- [x] **Development Platform or Board**.

First thing's first. Thanks for this wonderful project.

I would like you to add support for [**AURIX TC27xT family** boards](https://www.infineon.com/cms/en/product/microcontroller/32-bit-tricore-microcontroller/32-bit-tricore-aurix-tc2xx/aurix-family-tc27xt/)
      
It seems to me an excellent low-cost option for applications based on the Arduino language, **automotive grade**, in addition to its **pins being compatible with the Arduino Mega**, so in principle it would facilitate the update of systems based on this board.

I saw that there is a channel dedicated to the [Infineon XMC family](https://github.com/Infineon/platformio-infineonxmc), but I didn't know whether to post the issue there, since the XMC are based on ARM microcontroller, while the TC2x family is based on the **TriCore AURIX**.

If it is necessary to do some test to support the card support, I have just purchased one [AURIX TC275 Lite KIT](https://www.infineon.com/cms/en/product/evaluation-boards/kit_aurix_tc275_lite/#!documents), that I make available for any necessary test. 

Thank you very much in advance.

maxgerhardt, 2022-01-08:
For integration into PlatformIO the toolchain is needed, but according to https://community.infineon.com/t5/AURIX/Compiler-for-AURIX-TriCore/td-p/315656?daysprune=-1 the toolchain they give out is for evaluation purposes only, for commercial purposes it needs to be "licensed from Tricore partners" -- this could be an issue?

JoseOlin, 2022-01-09:
Interesting...

Thanks for your answer.

In principle, it would not be a problem for us, as we are considering the board to update a proof of concept, which at the moment would not be commercialized.
We are mainly interested in being able to develop on Linux, ideally in Arduino language using PlatformIO.

Maybe it's a bit off topic, but do you know of any evaluation board with equivalent performance to the AURIX TC27x (automotive grade up to ASIL-D, temperature range up to 125 °C, CAN port, at least two microprocessor) that are already supported in PlatformIO?

We reviewed several options, the closest from Infineon was the XMC4700 which, if I remember correctly, only supports applications up to ASIL-C, but we are open to other option.

Thanks in advance for your support.

maxgerhardt, 2022-01-09:
>In principle, it would not be a problem for us, as we are considering the board to update a proof of concept, which at the moment would not be commercialized.

The problem is about redistribution. PlatformIO can only use the compiler if it's packaged correctly (that is, uploaded in some repo or in the registry with a fitting [`package.json`](https://docs.platformio.org/en/latest/core/userguide/package/cmd_pack.html). The EULA one aggrees to when downloading [the toolchain](https://free-entry-toolchain.hightec-rt.com/) explicitly [states](https://hightec-rt.com/en/eula.html) 

>4.5 No Assignment or Sharing. In no event shall Customer disclose, transfer, assign, publish, distribute, rent, lease or in any other way make Software available to third parties without the prior written consent by HighTec.

Which straight-out forbids "making software available without prior written constenst by HighTec", who make this compiler.

Without a correctly-packaged toolchain for PlatformIO, no integration can be made.

JoseOlin, 2022-01-11:
It's a shame as the board was an excellent entry point to automotive grade projects and it was the best hardware / price ratio we found, however being suitable for Open Source projects is a priority for us.

Hopefully Aurix will release a more open source friendly toolchain in the future, such as the XMC family.

nitko12, 2023-08-07:
Any news on this?

It would be awesome to be able to run Aurix code in platformio.

Has anyone reached out to HighTec about this?

maxgerhardt, 2023-08-07:
I haven't reached out about this to anyone at least.

nitko12, 2023-08-09:
I've found this

https://github.com/volumit/tricore_gcc940_linux_bins

Could it be used, would it solve the license problem?

maxgerhardt, 2023-08-09:
If https://github.com/volumit/package_940 doesn't infringe on any copyrights from HighTec or another company, I would deem it usable.

I don't however know how well "chip is used in professional, safety-critical automotive environments" and "we're compiling our code with a GCC version that we found on Github and says it had Tricore support" fits together. It would need to be very clear that this isn't the manufacturer's recommended toolchain and there's 0 liability or expectations.

rvxfahim, 2023-10-23:
These automotive guys simply hate open-source because they are afraid of losing jobs and losing monopoly.

totocaca123, 2025-04-21:
Infineon manufactures products (aurix) mainly for automotive oem and suppliers. few automotive customers (BMW....) force everybody to fit to their norms (autosar, iso26262, ASPICE...) and everybody in automotive has to buy expansive software and tools. Infineon fits to its customers wishes. So customers does not want open source and believe that when they provide money, software is better. At the end,  Autosar requires bigger MCU, expansive toolchain... and car manufacturers pay for it. The winner of this situation is software editors (compiler, toolchain, ...) who sells expansive softwares. And automotive want to save money and they outsource jobs in low cost country. Automotive is really a bad industry regarding software development. I don't know if outside Europe, they do the same