platformio/platformio-core #4129: Remote agent fails while installing packages - cryptography-3.4.8 (requires-python:>=3.6)

MarvinLind, 2021-12-04:
What kind of issue is this?
- [x ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.
------------------------------------------------------------------

### Configuration

**Operating system**:
Raspbian GNU/Linux 10
Raspbian GNU/Linux 11

**PlatformIO Version**:
PlatformIO Core, version 5.2.3

### Description of problem
`pio remote agent start` fails after trying to install packages.
Installed platformio with the installation script `get-platformio.py`. Logged in with/through `pio home` and after that remote agent fails while installing dependencies.

#### Steps to Reproduce

1. Install latest raspbian
2. Install platformio with script:
```
python3 -c "$(curl -fsSL https://raw.githubusercontent.com/platformio/platformio/master/scripts/get-platformio.py)"
```
3. Start pio home and login from web browser.
4. Run:
```
pio remote agent start
```
5. Wait for 20 minutes or so for it to fail and try again, and then fail again and so on.

### Actual Results
```
  Downloading cryptography-3.4.8.tar.gz (546 kB)
     |████████████████████████████████| 546 kB 569 kB/s            
  Installing build dependencies ... error
  ERROR: Command errored out with exit status 1:
   command: /home/platformio/.platformio/penv/bin/python /tmp/pip-standalone-pip-uj7fkwgo/__env_pip__.zip/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-eyxhib9k/overlay --no-warn-script-location --no-binary :all: --only-binary :none: -i https://pypi.org/simple --extra-index-url https://www.piwheels.org/simple -- 'setuptools>=40.6.0' wheel 'cffi>=1.12; platform_python_implementation != '"'"'PyPy'"'"'' 'setuptools-rust>=0.11.4'
       cwd: None
  Complete output (41 lines):
  Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple, https://www.piwheels.org/simple
  Collecting setuptools>=40.6.0
    Using cached setuptools-59.4.0.tar.gz (2.3 MB)
    Getting requirements to build wheel: started
    Getting requirements to build wheel: finished with status 'done'
    Installing backend dependencies: started
    Installing backend dependencies: still running...
    Installing backend dependencies: finished with status 'done'
    Preparing metadata (pyproject.toml): started
    Preparing metadata (pyproject.toml): finished with status 'done'
  Collecting wheel
    Using cached wheel-0.37.0.tar.gz (65 kB)
    Preparing metadata (setup.py): started
    Preparing metadata (setup.py): still running...
    Preparing metadata (setup.py): finished with status 'done'
  Collecting cffi>=1.12
    Downloading cffi-1.15.0.tar.gz (484 kB)
    Preparing metadata (setup.py): started
    Preparing metadata (setup.py): finished with status 'done'
  Collecting setuptools-rust>=0.11.4
    Downloading setuptools-rust-1.1.1.tar.gz (284 kB)
    Installing build dependencies: started
    Installing build dependencies: still running...
    Installing build dependencies: still running...
    Installing build dependencies: still running...
    Installing build dependencies: still running...
    Installing build dependencies: still running...
    Installing build dependencies: still running...
    Installing build dependencies: still running...
    Installing build dependencies: still running...
    Installing build dependencies: still running...
    Installing build dependencies: still running...
    Installing build dependencies: still running...
    Installing build dependencies: still running...
    Installing build dependencies: still running...
    Installing build dependencies: still running...
    Installing build dependencies: still running...
    Installing build dependencies: finished with status 'done'
    Getting requirements to build wheel: started
    Getting requirements to build wheel: finished with status 'done'
  ERROR: Some build dependencies for setuptools-rust>=0.11.4 from https://files.pythonhosted.org/packages/3b/17/39804596680094ddabf2f44814c8919edb14da4f5db7b45e38915133f7fb/setuptools-rust-1.1.1.tar.gz#sha256=dc04afb52a70e56f5d6800a7ed9c45ad5937ba95e51a042b9096459edc038ad6 conflict with the backend dependencies: setuptools_scm==6.0.1 is incompatible with setuptools_scm>=6.3.2.
  ----------------------------------------
WARNING: Discarding https://files.pythonhosted.org/packages/cc/98/8a258ab4787e6f835d350639792527d2eb7946ff9fc0caca9c3f4cf5dcfe/cryptography-3.4.8.tar.gz#sha256=94cc5ed4ceaefcbe5bf38c8fba6a21fc1d365bb8fb826ea1688e3370b2e24a1c (from https://pypi.org/simple/cryptography/) (requires-python:>=3.6). Command errored out with exit status 1: /home/platformio/.platformio/penv/bin/python /tmp/pip-standalone-pip-uj7fkwgo/__env_pip__.zip/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-eyxhib9k/overlay --no-warn-script-location --no-binary :all: --only-binary :none: -i https://pypi.org/simple --extra-index-url https://www.piwheels.org/simple -- 'setuptools>=40.6.0' wheel 'cffi>=1.12; platform_python_implementation != '"'"'PyPy'"'"'' 'setuptools-rust>=0.11.4' Check the logs for full command output.
```
### Expected Results
A started remote agent.

ivankravets, 2021-12-04:
Could you try manually installing `cryptography`?
```
source ~/.platformio/penv/bin/activate
pip install "cryptography==3.3.2"
```

Does it work now?

MarvinLind, 2021-12-04:
> Could you try manually installing `cryptography`?
> 
> ```
> source ~/.platformio/penv/bin/activate
> pip install "cryptography==3.3.2"
> ```
> 
> Does it work now?
Thank you for you quick response.

I've now tried manually install `cryptography` with:
```
source ~/.platformio/penv/bin/activate
pip install "cryptography==3.3.2"
```
and
```
source ~/.platformio/penv/bin/activate
pip install "cryptography==3.4.8" # This is the first version it tries to install.
```
Unfortunately I get the same result every time.
Also it does not seem to honor my manual installation:
```
Collecting cryptography<35.0.0,>=3.3
  Using cached cryptography-3.4.8.tar.gz (546 kB)
  Installing build dependencies ...
```

MarvinLind, 2021-12-05:
So eventually it got installed over night when installation reached and tried `cryptography==3.3.2`
```
  Using cached cryptography-3.3.2.tar.gz (539 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting attrs>=19.1.0
  Using cached attrs-21.2.0.tar.gz (184 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting pyasn1
  Downloading pyasn1-0.4.8.tar.gz (146 kB)
     |████████████████████████████████| 146 kB 534 kB/s            
  Preparing metadata (setup.py) ... done
Collecting cffi>=1.12
  Using cached cffi-1.15.0.tar.gz (484 kB)
  Preparing metadata (setup.py) ... done
Collecting pycparser
  Using cached pycparser-2.21.tar.gz (170 kB)
  Preparing metadata (setup.py) ... done
Skipping wheel build for six, due to binaries being disabled for it.
Skipping wheel build for pyasn1, due to binaries being disabled for it.
Skipping wheel build for pyasn1-modules, due to binaries being disabled for it.
Skipping wheel build for cffi, due to binaries being disabled for it.
Skipping wheel build for pycparser, due to binaries being disabled for it.
Building wheels for collected packages: service-identity, attrs, cryptography
  Building wheel for service-identity (pyproject.toml) ... done
  Created wheel for service-identity: filename=service_identity-21.1.0-py2.py3-none-any.whl size=12608 sha256=a98d267fe68534b3a6f919eb19f09f32d31ae93ef16bcd25aa738e2beaf2657f
  Stored in directory: /home/platformio/.cache/pip/wheels/26/9e/49/b30bbf98dbc02ad8d65189b5e34a6493ce49d17ef122f92f48
  Building wheel for attrs (pyproject.toml) ... done
  Created wheel for attrs: filename=attrs-21.2.0-py2.py3-none-any.whl size=53741 sha256=88604b23103fcbe70dc3be23f560b1b6218716f2f23e8b5b0cb3b4fa7b976dfa
  Stored in directory: /home/platformio/.cache/pip/wheels/16/c5/10/f4f3745d449d987c88878f0327089a3a275362a633e05b0d95
  Building wheel for cryptography (pyproject.toml) ... done
  Created wheel for cryptography: filename=cryptography-3.3.2-cp37-cp37m-linux_armv7l.whl size=782126 sha256=3dca3655d6a312e71c69607ccd5975b5c254f891be2b8e508306a378e6642be5
  Stored in directory: /home/platformio/.cache/pip/wheels/b5/85/11/f80434ce4d816945abc9d79690a1064f750862de7bc9888908
Successfully built service-identity attrs cryptography
Installing collected packages: pycparser, six, pyasn1, cffi, pyasn1-modules, cryptography, attrs, service-identity
    Running setup.py install for pycparser ... done
    Running setup.py install for six ... done
    Running setup.py install for pyasn1 ... done
    Running setup.py install for cffi ... done
    Running setup.py install for pyasn1-modules ... done
Successfully installed attrs-21.2.0 cffi-1.15.0 cryptography-3.3.2 pyasn1-0.4.8 pyasn1-modules-0.2.8 pycparser-2.21 service-identity-21.1.0 six-1.16.0
WARNING: Target directory /home/platformio/.platformio/packages/contrib-pysite/pycparser already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/platformio/.platformio/packages/contrib-pysite/cffi already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/platformio/.platformio/packages/contrib-pysite/cryptography already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/platformio/.platformio/packages/contrib-pysite/cryptography-3.3.2.dist-info already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/platformio/.platformio/packages/contrib-pysite/six-1.16.0.egg-info already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/platformio/.platformio/packages/contrib-pysite/six.py already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/platformio/.platformio/packages/contrib-pysite/attr already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/platformio/.platformio/packages/contrib-pysite/cffi-1.15.0.egg-info already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/platformio/.platformio/packages/contrib-pysite/_cffi_backend.cpython-37m-arm-linux-gnueabihf.so already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/platformio/.platformio/packages/contrib-pysite/attrs-21.2.0.dist-info already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/platformio/.platformio/packages/contrib-pysite/pycparser-2.21.egg-info already exists. Specify --upgrade to force replacement.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting idna!=2.3,>=0.6
  Using cached idna-3.3.tar.gz (286 kB)
  Preparing metadata (setup.py) ... done
Skipping wheel build for idna, due to binaries being disabled for it.
Installing collected packages: idna
    Running setup.py install for idna ... done
Successfully installed idna-3.3
WARNING: Target directory /home/platformio/.platformio/packages/contrib-pysite/idna-3.3.egg-info already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/platformio/.platformio/packages/contrib-pysite/idna already exists. Specify --upgrade to force replacement.
```

closed 2021-12-05 by MarvinLind

ivankravets, 2021-12-06:
Thanks, I reverted this commit https://github.com/platformio/platformio-core/commit/39494d18bf66cf24ae9e7b6db6c2b638985c6837

Seems it fixes the original issue.

jcw, 2022-06-28:
There was another `pip` at `/usr/local/bin/` and it was earlier in the PATH. But making sure the penv version comes first did not solve the issue, it still says "This package requires Rust >=1.41.0." (which I'd rather not bring in, it's huge).