platformio/platformio-core #4132: c_cpp_properties.json has invalid autogenerated widows share paths

kadavris, 2021-12-06:
- [ ] **PlatformIO Core**.

----------------------------------------------------------------

### Configuration

**Operating system**: Windows 10

**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 5.2.3

### Description of problem
I have home linux server (fedora 34) that host my projects using samba share.
I've installed some libraries for a new project and Vstudio cannot find include files and build fails due to "source cannot be found" error.
I spotted that in c_cpp_properties.json the paths to samba shares are looking invalid by default:
```
    "includePath": [
                "/server's IP/projectpath/include",
                "/server's IP/projectpath/src",
                "C:/Users/XXXXX/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src",
                "/server's IP/projectpath/.pio/libdeps/esp01/Time",
                "/server's IP/projectpath/.pio/libdeps/esp01/PubSubClient/src",
```
Note the unusual, single slash before server's IP. And in fact Vstudio paint wiggle lines under these to indicate path problem.
When I click on the C/C++ config button on the bottom status line and get to Microsoft C/C++ Extension GUI config,
the "Include path" section uses correct (Microsoft's) form of share addresses: \\IP\path

The thing is that I use permanent drive letter to access this share and who and why translate it to network share form is a mystery to me.
This is all being more confusing as I don't know is it Platformio's bug or MS extension that produce c_cpp_properties.json.
That's because the head of it contain warning about auto-generated file:
```
    // !!! WARNING !!! AUTO-GENERATED FILE!
    // PLEASE DO NOT MODIFY IT AND USE "platformio.ini":
```
So I assume that I need to report it here.

#### Steps to Reproduce

1. Use samba share to host your project files
2. Install some libraries that is local to this project
3. Try to do build

### Actual Results
Build fail with source files of library dependencies was not found

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:esp01]
platform = espressif8266
board = esp01
framework = arduino
lib_deps = 
	knolleary/PubSubClient@^2.8
	paulstoffregen/Time@^1.6.1

```

### Additional info

kadavris, 2021-12-06:
Followup:
Looked closely at the build log and spotted a message there:
```
> Executing task in folder esp8266-as-mqtt-dongle: C:\Users\ap\.platformio\penv\Scripts\platformio.exe run <

Processing esp01 (platform: espressif8266; board: esp01; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------There is a known issue with Python 3.8+ and mapped network drives on Windows.
See a solution at:
https://github.com/platformio/platformio-core/issues/3417
```
I've created suggested registry key and rebooted, but build still fails:
```
Verbose mode can be enabled via `-v, --verbose` option   
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01.html
PLATFORM: Espressif 8266 (3.2.0) > Espressif Generic ESP8266 ESP-01 512k
HARDWARE: ESP8266 80MHz, 80KB RAM, 512KB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.30002.0 (3.0.2)
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - toolchain-xtensa 2.100300.210717 (10.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 45 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <PubSubClient> 2.8.0
|-- <Time> 1.6.1
|-- <ESP8266WiFi> 1.0
Building in release mode
Compiling .pio\build\esp01\src\main.cpp.o
'\\IP\path'
CMD.EXE was started with the above path as the current directory.
UNC paths are not supported.  Defaulting to Windows directory.
Compiling .pio\build\esp01\src\ntp.cpp.o
```
I also tried to manually change all //IP/share to a drive letter that used to access share, but nevertheless build still fails stubbornly using raw \\IP\ form.

ivankravets, 2021-12-06:
Duplicate of https://github.com/platformio/platformio-core/issues/3926

Could you re-test with `pio upgrade --dev`? 5.2.4 is planned for release next week.

closed 2021-12-06 by ivankravets