platformio/platformio-core #4133: add some possibility to test that a given serial output is generated by the board to the computer

jerabaul29, 2021-12-07:
I am looking at writing some unit tests for my project.

I would like to unit test that a given message is well sent over serial when running a function. I looked at the unit test documentation here https://docs.platformio.org/en/latest/plus/unit-testing.html but I cannot find (maybe I am not looking correctly / missing something?) how to do this. I think this may be a bit "hybrid", since "part of the test" is run on the board (generating the message), while part of the test is run on the computer (checking that the correct message is received).

To re-formulate, I would want to run a test that checks that, when a given part of the code is run on the board, a given message is printed to serial. Any ways to do this? :) .

jcw, 2021-12-12:
As I understand it, this somewhat stretches what unit tests are - what is sent to PIO over serial is the success/fail status of tests, not textual information (even though it obviously gets sent across in text form). PIO collects this status, reports it, and concludes with an overall red/green fail-or-ok outcome. The serial link is (at least conceptually) not part of your app, but of the test harness.

If you want to test serial output, I think you'll need to set up a [test double](https://en.wikipedia.org/wiki/Test_double) for the serial port, which replaces real serial output, collects it in memory, and then test that the collected text matches what you expect.

(Unity is documented here: https://github.com/ThrowTheSwitch/Unity/tree/master/docs)

ivankravets, 2022-05-03:
I'm not sure that I understand what you are going to achieve, but the latest version of PlatformIO Core 6.0 allows you to override the exiting test runner and handle multiple things, including the "testing stage". See updated docs https://docs.platformio.org/en/latest/advanced/unit-testing/frameworks/custom/index.html