platformio/platformio-core #4136: Add support for Cypress EZ-USB FX3

nschurando, 2021-12-09:
What kind of issue is this?
- [x] **Development Platform or Board**.
- [x] **Feature Request**.
------------------------------------------------------------------
I am interested in adding support for the Cypress EZ-USB FX3 microcontroller. I would be happy to help, but I'm not sure where to start.

Here are the [product page](https://www.cypress.com/products/ez-usb-fx3-superspeed-usb-30-peripheral-controller) and the [datahseet](https://www.cypress.com/file/140296/download).

It is an ARM926 with 512 kB SRAM with multiple peripherals in particular a 5 Gbps USB 3 controller with built-in PHY. Which makes it frequently used in video over usb products.

There is an [official CYUSB3KIT-003 development board](https://www.cypress.com/documentation/development-kitsboards/cyusb3kit-003-ez-usb-fx3-superspeed-explorer-kit) with an integrated debugger.

[image]

Dantali0n, 2022-08-12:
I'd like to add that the same toolchain is used to program the FX2 that supports 480mbps USB 2.

ivankravets, 2022-11-21:
Please vote 👍  for this issue https://github.com/platformio/platformio-core/issues/2052

closed 2022-11-21 by ivankravets