platformio/platformio-core #4137: PIO debug does not work under for stm32cubemx project under Windows

elmot, 2021-12-10:
- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Windows

**IDE**: CLion

**PlatformIO Version** : 5.2.3

### Description of problem
`pio debug` does not handle breakpoints properly

#### Steps to Reproduce

1. Create project for STM32F3-Discovery, stm32cube framework
2. add a trivial main.c(see below)
3. Connected the board normally with ST-LINK
4. Start ` pio.exe debug --interface=gdb -x .pioinit --iex "set mi-async on" -e disco_f303vc`
5. when the debugger stops at temporary breakpoint, try commands:
5.1 `list main.c` - that shows that main.c debug info is loaded, 4 lines inside
5.2 `b main.c:3` - breakpoint is not set
5.3 `b src/main.c:3` - breakpoint is not set

### Actual Results
Breakpoints do not work

### Expected Results
Two breakpoints are set at line 3 of main.c

**The content of `platformio.ini`:**
```ini
[env:disco_f303vc]
platform = ststm32
board = disco_f303vc
framework = stm32cube
```

**Source file to reproduce issue:**
```c
int main() {
// write your code here
 while(1);
}
```

valeros, 2021-12-14:
Hi @elmot ! Any chance you could try a bit more complex example (e.g. [this one](https://github.com/platformio/platform-ststm32/tree/develop/examples/stm32cube-hal-blink))?

I believe the behavior you're observing is expected as your code simply compiles to a single assembly line even in debug mode with the `-Og` optimization. As as workaround you can try to completely turn off the optimization by adding the following line to your configuration:
```
[env:disco_f303vc]
platform = ststm32
board = disco_f303vc
framework = stm32cube
; Set optimization level and amount of debug information generated by the compiler
debug_build_flags = -O0 -ggdb3 -g3
```

elmot, 2021-12-15:
hi @valeros 
Unfortunately I can't even verify now. 
PIO debug under windows looks completely broken now
Debug log for all my stm32/platformio projects:
```txt
Reading symbols from C:\Users\Ilia.Motornyi\CLionProjects\untitled5\.pio\build\nucleo_l432kc\firmware.elf...done.
PlatformIO Unified Debugger -> https://bit.ly/pio-debug
PlatformIO: debug_tool = stlink
PlatformIO: Initializing remote target...
xPack OpenOCD x86_64 Open On-Chip Debugger 0.11.0+dev (2021-10-16-21:19)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
hla_swd
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : DEPRECATED target event trace-config; use TPIU events {pre,post}-{enable,disable}
Info : tcl server disabled
Info : telnet server disabled
Info : clock speed 500 kHz
Error: open failed

.pioinit:13: Error in sourced command file:
```
```txt
PS C:\Users\Ilia.Motornyi\CLionProjects\untitled5> platformio --version
PlatformIO Core, version 5.2.3
```

.pioinit file I found in a service folder:
```txt=
echo PlatformIO Unified Debugger -> https://bit.ly/pio-debug\n
echo PlatformIO: debug_tool = stlink\n
echo PlatformIO: Initializing remote target...\n

define pio_reset_halt_target
    monitor reset halt
end

define pio_reset_run_target
    monitor reset
end

target extended-remote | "C:/Users/Ilia.Motornyi/.platformio/packages/tool-openocd/bin/openocd.exe" -s "C:/Users/Ilia.Motornyi/.platformio/packages/tool-openocd" -c "gdb_port pipe; tcl_port disabled; telnet_port disabled" -s "C:/Users/Ilia.Motornyi/.platformio/packages/tool-openocd/scripts" -f "interface/stlink.cfg" -c "transport select hla_swd" -f "target/stm32l4x.cfg"
monitor init
load
pio_reset_halt_target
tbreak main

define pio_restart_target
   pio_reset_halt_target
   tbreak main
   continue
end
echo PlatformIO: Initialization completed\n
```

valeros, 2021-12-15:
Seems like an issue with OpenOCD and L4 targets. Maybe you have another board to verify this behavior?

elmot, 2021-12-15:
Sorry I don't at the moment
But nucleo_f103rb gives the same result.

elmot, 2021-12-15:
BTW I have just got fresh windows update. Might be system command interpreter behavior has changed?

valeros, 2021-12-15:
> But nucleo_f103rb project gives the same result, it does not even start openocd

The log is the same as for `nucleo_l432kc`?

elmot, 2021-12-15:
yeah, exactly the same

valeros, 2021-12-15:
OK, let's try removing PLatformIO from the equation. Could you please try to launch OpenOCD in a separate command line via the following command:
```
"C:/Users/Ilia.Motornyi/.platformio/packages/tool-openocd/bin/openocd.exe" -s "C:/Users/Ilia.Motornyi/.platformio/packages/tool-openocd" -s "C:/Users/Ilia.Motornyi/.platformio/packages/tool-openocd/scripts" -f "interface/stlink.cfg" -f "target/stm32f1x.cfg"
```

elmot, 2021-12-16:
@valeros 
I don't have F1 in hands ATM, but for L4 openocd works
Btw, let's chat via Telegram, noone needs long chat transcript here
@ivankravets has my contact

elmot, 2021-12-16:
It looks like the problem is in new GCC, which extensively optimizes the code
Non working debugger was a strange glitch
The fix for the problem is to add 
```txt
debug_build_flags = -g3
```
to the `platformio.ini` file

closed 2021-12-16 by elmot

majorzpley, 2024-01-21:
thanks,i met same issue,that works fine:debug_build_flags = -g3