platformio/platformio-core #4139: Actual git version is not displayed for `platform_packages`

mcspr, 2021-12-15:
What kind of issue is this?

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**:
Fedora Linux 35 (Container Image)

**PlatformIO Version** (`platformio --version`):
PlatformIO Core, version 5.2.5a1

### Description of problem

Given the following platformio.ini
```ini
[env]
platform = espressif8266@3.2.0
framework = arduino
platform_packages =
    platformio/framework-arduinoespressif8266 @ git+https://github.com/esp8266/Arduino.git
; notice that there's no version pin, it fetches the master branch

[env:d1_mini]
board = d1_mini
```

Extra package override is successfully fetched into `~/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/` and reported as version `3.1.0-dev+sha.42aa0e6d`
However, manually changing the commit by using `git checkout 55ef3e73978ad010d44ad7e4c9bd4b4f32455989` does not seem to affect the version displayed in the `PACKAGES` list when doing `pio run`, and instead the `.git/.piopm` metadata file "version" field is used (from the installation time).

```
$ pio run
Processing d1_mini (platform: espressif8266@3.2.0; board: d1_mini; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp12e.html
PLATFORM: Espressif 8266 (3.2.0) > Espressif ESP8266 ESP-12E
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.1.0-dev+sha.42aa0e6d
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - toolchain-xtensa 2.100300.210717 (10.3.0)
^CBuild interrupted.
$ git -C ~/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/ rev-parse HEAD
55ef3e73978ad010d44ad7e4c9bd4b4f32455989
```

Shouldn't the package manager run `git rev-parse HEAD` (or track `.git/HEAD` changes) and use that value instead of version saved in the `.piopm`? Or, at least attempt to refresh it. I have noticed inconsistent version reports at the esp8266/Arduino repo when user has master-branch package configured, it would be nice to know the actual revision of the repo without asking the user to go to the ~/.platformio/packages dir and run `git` command manually :)

(and I assume the same behaviour for `lib_deps`, but I have not tried it)

ivankravets, 2021-12-15:
It seems that `pio platform update` handles these changes. I'm not sure that it works with overridden packages and will test later. Thanks for the report.

ivankravets, 2022-03-31:
We could implement it in theory but this will downgrade the whole PlaformIO performance. There are a lot of places where different components interact with the Package Manager. To reduce I/O operations, the `.piopm` manifest was introduced in PlatformIO Core 5.0. 

Please use [pio pkg update](https://docs.platformio.org/en/latest/core/userguide/pkg/cmd_update.html) command to update your dependencies. In this case, PlatformIO will manage package manfiests.

ivankravets, 2022-03-31:
I forgot to add, calling `git rev-parse HEAD` or another command via subprocess is a very expensive operation.

ivankravets, 2022-04-02:
> .and another problem comes from the way git specs are encoded. For example, the git repo package version from above is currently 3.1.0-dev

I can't reproduce this issue. Did you use PlatformIO for package installation?

See my test for https://github.com/dbSuS/libpax.git:

[image]

mcspr, 2022-04-05:
> ```ini
> platform_packages =
>     platformio/framework-arduinoespressif8266 @ git+https://github.com/esp8266/Arduino.git
> ```

Just the platformio.ini from above, `pio run` installed the platform and the git framework package. Output of latest version is just the `3.1.0-dev`, no additional sha hash like you have.
```ini
platform_packages =
    platformio/framework-arduinoespressif8266 @ git+https://github.com/esp8266/Arduino.git
```
Package was installed some time ago to the `~/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910`
```
> cat ~/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/.git/.piopm
{"type": "tool", "name": "framework-arduinoespressif8266", "version": "3.1.0-dev+sha.42aa0e6d", "spec": {"owner": null, "id": null, "name": "framework-arduinoespressif8266", "requirements": null, "url": "git+https://github.com/esp8266/Arduino.git"}}
```

PaxCounter package has a strange dependency line though
https://github.com/cyberman54/ESP32-Paxcounter/blob/2dd01b6bbcb9e00842a2e16aa679d582da335824/platformio_orig.ini#L80
```ini
lib_deps_basic =
    https://github.com/dbSuS/libpax.git @ ^1.0.0
```

mcspr, 2022-04-07:
I think I have found the problem... Unless the local git repo is using the branch known the remote, it  won't show anything in the `Latest`. Which should've been apparent if I only now looked at the vcsclient code... :/ Attempting to change the branch to the earlier commit and also fixing the .piopm value, it finally works.

Should it fallback to the `HEAD` when it cannot find anything?
(and we could also reduce the amount of stuff fetched / filtered by specifying the ref path ourselves)
```diff
diff --git a/platformio/package/vcsclient.py b/platformio/package/vcsclient.py
index adbcd6f5..57aba711 100644
--- a/platformio/package/vcsclient.py
+++ b/platformio/package/vcsclient.py
@@ -218,12 +218,22 @@ class GitClient(VCSClientBase):
         branch = self.get_current_branch()
         if not branch:
             return self.get_current_revision()
-        result = self.get_cmd_output(["ls-remote"])
+
+        branch_ref = f"refs/heads/{branch}"
+        result = self.get_cmd_output([
+            "ls-remote", self.remote_url, "HEAD", branch_ref
+        ])
+
+        head = None
         for line in result.split("\n"):
-            ref_pos = line.strip().find("refs/heads/" + branch)
-            if ref_pos > 0:
-                return line[:ref_pos].strip()[:7]
-        return None
+            sha, ref = line.split("\t")
+            if ref == "HEAD":
+                head = sha[:7]
+
+            if ref == branch_ref:
+                return sha[:7]
+
+        return head
```

ivankravets, 2022-04-07:
Could you make a PR with the changes?

mcspr, 2022-04-07:
I will, as soon as I'll get to the PC

PR #4217 (2022-04-08): Avoid working with detached / non-existent git branches when checking for updates
b/c we can't use `pull` anyway in that situation
also ask *only* for the specific branch via `refs/heads/{branch}` and fail when it is not available. `pkg update` / `pkg outdated` will not do or display anything if that's the case.

resolves #4139 
(and I am not familiar with the svn & hg cases, so only git implementation was changed)
closed 2022-04-08 by ivankravets