platformio/platformio-core #4149: Impossible to build project No such File or Directory

StefanoTesla, 2022-01-06:
Hello everyone, I'm building my git hub project in a new fresh formatted pc, I don't understand why I get this error:

All the missing file exist...

No antivirus running
```
Building in release mode
Compiling .pio\build\esp32doit-devkit-v1\src\dome.cpp.o
In file included from .pio/libdeps/esp32doit-devkit-v1/ArduinoJson/src/ArduinoJson/Configuration.hpp:156:0,
                 from .pio/libdeps/esp32doit-devkit-v1/ArduinoJson/src/ArduinoJson.hpp:7,
                 from .pio/libdeps/esp32doit-devkit-v1/ArduinoJson/src/ArduinoJson.h:9,
                 from src/dome.cpp:1:
C:/Users/Tesla/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:142:21: fatal error: algorithm: No such file or directory
compilation terminated.
Compiling .pio\build\esp32doit-devkit-v1\lib5ad\AsyncTCP-esphome\AsyncTCP.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\lib632\AsyncTCP\AsyncTCP.cpp.o
In file included from .pio\libdeps\esp32doit-devkit-v1\AsyncTCP-esphome\src\AsyncTCP.cpp:22:0:
C:\Users\Tesla\.platformio\packages\framework-arduinoespressif32\cores\esp32/Arduino.h:142:21: fatal error: algorithm: No such file or directory
compilation terminated.
Compiling .pio\build\esp32doit-devkit-v1\libaea\FS\FS.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\libaea\FS\vfs_api.cpp.o
In file included from .pio\libdeps\esp32doit-devkit-v1\AsyncTCP\src\AsyncTCP.cpp:22:0:
C:\Users\Tesla\.platformio\packages\framework-arduinoespressif32\cores\esp32/Arduino.h:142:21: fatal error: algorithm: No such file or directory
compilation terminated.
In file included from C:\Users\Tesla\.platformio\packages\framework-arduinoespressif32\libraries\FS\src\FS.cpp:21:0:
C:\Users\Tesla\.platformio\packages\framework-arduinoespressif32\libraries\FS\src\FS.h:24:18: fatal error: memory: No such file or directory
compilation terminated.
In file included from C:\Users\Tesla\.platformio\packages\framework-arduinoespressif32\libraries\FS\src\vfs_api.h:18:0,
                 from C:\Users\Tesla\.platformio\packages\framework-arduinoespressif32\libraries\FS\src\vfs_api.cpp:15:
C:\Users\Tesla\.platformio\packages\framework-arduinoespressif32\libraries\FS\src\FS.h:24:18: fatal error: memory: No such file or directory
compilation terminated.
Compiling .pio\build\esp32doit-devkit-v1\lib8a7\WiFi\ETH.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\lib8a7\WiFi\WiFi.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\lib8a7\WiFi\WiFiAP.cpp.o
*** [.pio\build\esp32doit-devkit-v1\src\dome.cpp.o] Error 1
*** [.pio\build\esp32doit-devkit-v1\lib5ad\AsyncTCP-esphome\AsyncTCP.cpp.o] Error 1
*** [.pio\build\esp32doit-devkit-v1\lib632\AsyncTCP\AsyncTCP.cpp.o] Error 1
*** [.pio\build\esp32doit-devkit-v1\libaea\FS\FS.cpp.o] Error 1
*** [.pio\build\esp32doit-devkit-v1\libaea\FS\vfs_api.cpp.o] Error 1
In file included from C:\Users\Tesla\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src\WiFiSTA.h:28:0,
                 from C:\Users\Tesla\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src\WiFi.h:32,
                 from C:\Users\Tesla\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src\ETH.h:24,
                 from C:\Users\Tesla\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src\ETH.cpp:21:
C:\Users\Tesla\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src\WiFiGeneric.h:28:22: fatal error: functional: No such file or directory
compilation terminated.
*** [.pio\build\esp32doit-devkit-v1\lib8a7\WiFi\ETH.cpp.o] Error 1
In file included from C:\Users\Tesla\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src\WiFiSTA.h:28:0,
                 from C:\Users\Tesla\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src\WiFi.h:32,
                 from C:\Users\Tesla\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src\WiFi.cpp:24:
C:\Users\Tesla\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src\WiFiGeneric.h:28:22: fatal error: functional: No such file or directory
compilation terminated.
*** [.pio\build\esp32doit-devkit-v1\lib8a7\WiFi\WiFi.cpp.o] Error 1
In file included from C:\Users\Tesla\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src\WiFiSTA.h:28:0,
                 from C:\Users\Tesla\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src\WiFi.h:32,
                 from C:\Users\Tesla\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src\WiFiAP.cpp:25:
C:\Users\Tesla\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src\WiFiGeneric.h:28:22: fatal error: functional: No such file or directory
compilation terminated.
*** [.pio\build\esp32doit-devkit-v1\lib8a7\WiFi\WiFiAP.cpp.o] Error 1
============================================================================== [FAILED] Took 4.06 seconds ==============================================================================
Il processo del terminale "C:\Users\Tesla\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'esp32doit-devkit-v1'" è stato terminato. Codice di uscita: 1.

Terminale verrà riutilizzato dalle attività, premere un tasto qualsiasi per chiuderlo.`
```

maxgerhardt, 2022-01-06:
Remove `C:\Users\Tesla\.platformio\packages\toolchain-xtensa32` an retry.

StefanoTesla, 2022-01-06:
`Building in release mode
Compiling .pio\build\esp32doit-devkit-v1\src\dome.cpp.o
"xtensa-esp32-elf-g++" non \x8a riconosciuto come comando interno o esterno,
 un programma eseguibile o un file batch.`

maxgerhardt, 2022-01-06:
You have an antivirus installed that interferes with the download, installation and invocation of the toolchain? This is not normal.

StefanoTesla, 2022-01-06:
Any antivirus

StefanoTesla, 2022-01-07:
Finally I found the problem, i delete all content Inside package folder and all .cache folder

And build is ok

closed 2022-01-07 by StefanoTesla

MainWire, 2024-02-09:
Hi,

I am sorry to reopen the issue, but **I am having the same problem as @StefanoTesla**. I can't find a way to bypass that error.
Deleting all content inside package and .cache folder doesn't help. No antivirus, **platformio downloads everything but gets stuck on that error**.
Could anyone give me a hint on what may be going on ?

I'm running the latest version of platformio, 
**my ini config :**

```
[env:ch552]
platform = intel_mcs51
board = ch552
build_flags =
  --model-medium
  --LC:/Users/mainwire/.platformio/packages/toolchain-sdcc/lib/medium
  -IC:
  -IC:/Users/mainwire/Documents/PlatformIO/Projects/ch552_controller/src/Driver/inc
  -Isrc/PLAT/inc
  extra_scripts = link_medium.py
```
**link_medium.py file :**
```
Import("env")
env.Append(LINKFLAGS=["--model-medium"])
```
  
> ```**Building in release mode Compiling .pio\build\esp32doit-devkit-v1\src\dome.cpp.o "xtensa-esp32-elf-g++" non \x8a riconosciuto come comando interno o esterno, un programma eseguibile o un file batch.**```

maxgerhardt, 2024-02-09:
You didn't post your `ch552` board definition and used absolute pahs from your project, so this is a bit hard to reproduce. When I use a known board with
```ini
[env:CH559]
platform = intel_mcs51
board = CH559
build_flags =
  --model-medium
  --LC:/Users/Max/.platformio/packages/toolchain-sdcc/lib/medium
  -IC:
  -Isrc/Driver/inc
  -Isrc/PLAT/inc
extra_scripts = link_medium.py
```
and your above script conent, it compiles fine, but complains 
```
at 1: warning 117: unknown compiler option '--LC:/Users/Max/.platformio/packages/toolchain-sdcc/lib/medium' ignored
```
Since that should be `-L` and not `--L`. When I correct that, it builds even without warnings.

maxgerhardt, 2024-02-09:
Ah, if you **actually** use
```ini
build_flags =
  --model-medium
  --LC:/Users/mainwire/.platformio/packages/toolchain-sdcc/lib/medium
  -IC:
  -IC:/Users/mainwire/Documents/PlatformIO/Projects/ch552_controller/src/Driver/inc
  -Isrc/PLAT/inc
  extra_scripts = link_medium.py
```
then that's wrong because there are two spaces in front of `extra_scripts` which should not be there. Then that `extra_scripts = link_medium.py` is added as a compiler argument, which the compiler thinks is a file to be build. And of course that's not there. Double check your spaces.

MainWire, 2024-02-09:
> You didn't post your `ch552` board definition and used absolute pahs from your project, so this is a bit hard to reproduce. When I use a known board with
> 
> ```ini
> [env:CH559]
> platform = intel_mcs51
> board = CH559
> build_flags =
>   --model-medium
>   --LC:/Users/Max/.platformio/packages/toolchain-sdcc/lib/medium
>   -IC:
>   -Isrc/Driver/inc
>   -Isrc/PLAT/inc
> extra_scripts = link_medium.py
> ```
> 
> and your above script conent, it compiles fine, but complains
> 
> ```
> at 1: warning 117: unknown compiler option '--LC:/Users/Max/.platformio/packages/toolchain-sdcc/lib/medium' ignored
> ```
> 
> Since that should be `-L` and not `--L`. When I correct that, it builds even without warnings. See [build logs](https://pastebin.com/rBJe807q).

Thank you for answering,

I updated my ini file :
```
[env:CH559]
platform = intel_mcs51
board = CH559
build_flags =
  --model-medium
  --LC:%%userprofile%%/.platformio/packages/toolchain-sdcc/lib/medium
  -IC:
  -Isrc/Driver/inc
  -Isrc/PLAT/inc
extra_scripts = C:\Users\mainwire\.platformio\link_medium.py
```
Here is my board definition : https://pastebin.com/J8HgWyvh
The space appeared when I wrote the message, I don't have it in my config file.

I tried to delete .cache and packages again, so far no changes
build logs : https://pastebin.com/XhMAL3CB

MainWire, 2024-02-09:
This one : 
`"CC" non \x8a riconosciuto come comando interno o esterno,
 un programma eseguibile o un file batch.`
 What do you mean by "can't be the same one with xtensa" ?

maxgerhardt, 2024-02-09:
In <https://github.com/platformio/platformio-core/issues/4149#issuecomment-1935637991> you quoted your error message as `xtensa-esp32-elf-g++`.  

The error is weird. Can you open a [core CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli) and post the output of 
```
pio run -t clean
pio run -v -j1
```
?

maxgerhardt, 2024-02-09:
..This is a MCS51 based chip. You absolutely cannot use an XTensa toolchain with this.

MainWire, 2024-02-09:
> In [#4149 (comment)](https://github.com/platformio/platformio-core/issues/4149#issuecomment-1935637991) you quoted your error message as `xtensa-esp32-elf-g++`.
> 
> The error is weird. Can you open a [core CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli) and post the output of
> 
> ```
> pio run -t clean
> pio run -v -j1
> ```
> 
> ?

Same result : https://pastebin.com/8xBQBXGc

maxgerhardt, 2024-02-09:
That is not supposed to happen. `CC` should be replaced by `sdcc` per [builder code](https://github.com/platformio/platform-intel_mcs51/blob/08d581eb5832c0a4a19c7bc349c7989f3a3af8b4/builder/main.py#L62-L65). I can also not reproduce your bug on my machine. You should open a bug report in https://github.com/platformio/platform-intel_mcs51/issues

MainWire, 2024-02-09:
Very well, thank you for taking your time on this !
Do you believe it could be a simple mistake caused by my configs ? Or is it definitely a bug ?

maxgerhardt, 2024-02-09:
If you don't happen to have executed some `export CC="CC"` that would try to override PlatformIO's `CC` variable, this should not happen. Also, the bug should happen in the standard config too then, right? E.g., imported https://github.com/platformio/platform-intel_mcs51/tree/develop/examples/native-blink.

MainWire, 2024-02-09:
Thank you for that link, I will try it one last time

MainWire, 2024-02-27:
Hi, I am reporting back here because there is something weird I can't explain ; 
the compilation is **working completely fine with an empty/full _C_ file** in the project, but is _NOT_ working with an **empty/full _CPP_ file**. Am I completely missing how the CH559 works ? Or is it a bug ?

MainWire, 2024-03-12:
Hi @maxgerhardt, I am sorry to ping you, but I am currently unable to find a solution to my problem. It would seem that the CH552G can only be programmed in C. Is there any way to get it to work with C++ files ? Or should I rewrite my entire code in C (without classes) ?