platformio/platformio-core #4152: Library dependency not recognized in conditional include

maxgerhardt, 2022-01-08:
What kind of issue is this?

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Win 10 x64

**PlatformIO Version** (`platformio --version`): `PlatformIO Core, version 5.2.5a5`

### Description of problem

Per https://community.platformio.org/t/liquidmenu-library-installation/25329.

After including the LiquidMenu library ([registry](https://registry.platformio.org/libraries/fmalpartida/LiquidCrystal/installation), [repo](https://github.com/VaSe7u/LiquidMenu)), compilation always fails with 

```
.pio\libdeps\uno\LiquidMenu\src\LiquidMenu.h:53:10: fatal error: LiquidCrystal.h: No such file or directory
=====================[FAILED] Took 1.58 seconds ======================
```

The library has a mechanism in which it can either include `LiquidCrystal.h` or `LiquidCrystal_I2C.h`  depending on the value of the `LIQUIDMENU_LIBRARY` macro, which, if not overriden by a global, defaults to the value `LiquidCrystal_LIBRARY` (1) and thus activates the first include in the 

```cpp
#if LIQUIDMENU_LIBRARY == LiquidCrystal_LIBRARY
# pragma message ("LiquidMenu: Selected 'LiquidCrystal' (parallel) library. Edit 'LiquidMenu_config.h' file to change it.")
#include <LiquidCrystal.h>
#elif LIQUIDMENU_LIBRARY == LiquidCrystal_I2C_LIBRARY
# pragma message ("LiquidMenu: Selected 'LiquidCrystal_I2C' (I2C) library. Edit 'LiquidMenu_config.h' file to change it.")
#include <LiquidCrystal_I2C.h>
#else
# pragma message ("LiquidMenu: Selected custom library. Edit 'LiquidMenu_config.h' file to change it.")
#endif
```

statement. 

See code at 
* https://github.com/VaSe7u/LiquidMenu/blob/master/src/LiquidMenu.h#L51-L59
 *https://github.com/VaSe7u/LiquidMenu/blob/master/src/LiquidMenu_config.h#L25-L30
* https://github.com/VaSe7u/LiquidMenu/blob/master/src/LiquidMenu_const.h#L13-L17

Setting `lib_ldf_mode = deep+` and adding the LiquidCrystal libary to the dependencies does not change the output. It seems that PlatformIO's library dependency scanner is overwhelmed with the conditional include mechanism done in this library.

#### Steps to Reproduce

1. Create a new Uno + Arduino project
2. Copy `platformio.ini` and code from below
3. Attempt compilation

### Actual Results

Compilation error with `LiquidCrystal.h` not being found, dependency graph not showing a dependency from `LiquidMenu` to `LiquidCrystal`.

### Expected Results

Compilation successful, dependency from `LiquidMenu` to `LiquidCrystal` is in the dependency graph.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:uno]
platform = atmelavr
board = uno
framework = arduino
lib_deps =
   vase7u/LiquidMenu@^1.6.0
   fmalpartida/LiquidCrystal@^1.5.0
   Wire
lib_ldf_mode = deep+
```

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>
#include <LiquidMenu.h>

void setup() {
   Serial.begin(9600);
}

void loop() {}
```

### Additional info
-/-

sheraz-janjua, 2022-02-19:
was this issue resolved? any workaround? I am facing the same issue

c2v4, 2022-03-08:
@sheraz-janjua 
As a workaround, you may remove the conditional from the LiquidMenu's header file and just leave the include that you wish to use (LiquidCrystal/LiquidCrystal_I2C)

ivankravets, 2022-04-02:
This issue will by fixed by https://github.com/platformio/platformio-core/issues/3113

closed 2022-04-02 by ivankravets

RoterMilan72, 2022-10-20:
Is this problem solved in core version 6.1.4?
I have exactly the same problem with the LiquidMenu.lib as described in issue #4152.

ivankravets, 2022-10-20:
Duplicate of https://github.com/platformio/platformio-core/issues/3113