platformio/platformio-core #4154: completion for bash do not work for not english locale

pat1, 2022-01-10:
What kind of issue is this?

### Configuration

**Operating system**:
Linux Fedora

**PlatformIO Version** (`platformio --version`):
PlatformIO Core, version 5.2.4

### Description of problem
after :
```
pio system completion install bash
```
logging in a bash session produce:
```
Error: Traceback (most recent call last):
  File "/home/pat1/.platformio/penv/lib64/python3.8/site-packages/platformio/__main__.py", line 121, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/home/pat1/.platformio/penv/lib64/python3.8/site-packages/click/core.py", line 1137, in __call__
    return self.main(*args, **kwargs)
  File "/home/pat1/.platformio/penv/lib64/python3.8/site-packages/click/core.py", line 1057, in main
    self._main_shell_completion(extra, prog_name, complete_var)
  File "/home/pat1/.platformio/penv/lib64/python3.8/site-packages/click/core.py", line 1132, in _main_shell_completion
    rv = shell_complete(self, ctx_args, prog_name, complete_var, instruction)
  File "/home/pat1/.platformio/penv/lib64/python3.8/site-packages/click/shell_completion.py", line 45, in shell_complete
    echo(comp.source())
  File "/home/pat1/.platformio/penv/lib64/python3.8/site-packages/click/shell_completion.py", line 324, in source
    self._check_version()
  File "/home/pat1/.platformio/penv/lib64/python3.8/site-packages/click/shell_completion.py", line 319, in _check_version
    raise RuntimeError(
RuntimeError: Couldn't detect Bash version, shell completion is not supported.

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
```

#### Steps to Reproduce

1. pio system completion install bash
2. use not english language in your locale settings
3. login in bash terminal session

### Actual Results
Works with english only locale

### Expected Results
Works with all locale settings

### Additional info
I sugget to use  env=dict() in:
```
        output = subprocess.run(["bash", "--version"],env=dict(), stdout=subprocess.PIPE)
```
in click/shell_completion.py file
in class BashComplete(ShellComplete)

I have tested it and it works.

ivankravets, 2022-01-19:
It seems like an issue linked with the Python Click library. Could I ask you to forward this issue to https://github.com/pallets/click/issues?

closed 2022-01-19 by ivankravets

pat1, 2022-01-19:
solved in Click release 8 by:
https://github.com/pallets/click/pull/1942