platformio/platformio-core #4155: Error: Unknown board ID - due to wrong absolute path used instead of relative path in case of subst-ed Windows drive

lmagyar, 2022-01-17:
What kind of issue is this?

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Win7 + VS Code version 1.63.2

**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 5.2.4

### Description of problem

Build fails with **Error: Unknown board ID 'esp8266_1M'** on project that's been **working flawlessly for the past several months**. Only VS Code and PIO IDE has got updated. Reinstalling VS Code, PIO IDE, even with deleting \<user\>\\.platformio and .vscode folders doesn't change anything.

I've spent several days locating the error, and finally it is reproducible.

Short version: During build, the relative path to .pio directory gets substituted with a wrong absolute path. See below **Steps to Reproduce**

#### Steps to Reproduce

Configuration: `> subst Z: C:\DRIVE_Z`

1. Project located at `Z:\_\Project` (physically at `C:\DRIVE_Z\_\Project`) previously has built flawlessly, but due to some change in the past approx. 2 months in PIO now fails with **Error: Unknown board ID 'esp8266_1M'**
2. If I copy the files to `C:\_\Project`, it magically builds.

If the project is located on real drives C: or Y:, log shows: `Compiling .pio\build\............c.o`
 If the project is located on a subst'ed virtual drive Z:, log shows: `Compiling C:\_\Project\.pio\build\............c.o`

So PIO somehow figures out, that this is a virtual drive, and starts using absolute path, but uses only the original drive letter but not the original path, somehow it uses `<original drive letter>:\<path to project on subst drive>\...` instead of `<original drive letter>:\<path to project on original drive>\...`.

Though I don't understand why it tries to replace the perfectly working relative path with an absolute path?

#### Workaround

Add a junction that redirects `C:\_\Project` to `C:\DRIVE_Z\_\Project`.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**

(This is the Tasmota project, that works perfectly when it is NOT on a subst-ed drive.)

```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter, extra scripting
;   Upload options: custom port, speed and extra flags
;   Library options: dependencies, extra library storages
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/en/stable/projectconf.html
;
; *********************************************************************
; *** Selection of Tasmota build variant is done with VSC
; *** alternatively can be done in: platformio_override.ini
; *** See example: platformio_override_sample.ini
; *********************************************************************

[platformio]
description                 = Provide ESP8266 / ESP32 based devices with Web, MQTT and OTA firmware
src_dir                     = tasmota
lib_dir                     = lib/default
boards_dir                  = boards
build_cache_dir             = .cache
extra_configs               = platformio_tasmota32.ini
                              platformio_tasmota_env.ini
                              platformio_tasmota_env32.ini
                              platformio_override.ini

[common]
platform                    = ${core.platform}
platform_packages           = ${core.platform_packages}
framework                   = arduino
board                       = esp8266_1M
board_build.filesystem      = littlefs
custom_unpack_dir           = unpacked_littlefs
build_unflags               = ${core.build_unflags}
build_flags                 = ${core.build_flags}
monitor_speed               = 115200
monitor_port                = COM5
; *** Upload Serial reset method for Wemos and NodeMCU
upload_resetmethod          = nodemcu
upload_port                 = COM5
extra_scripts               = ${scripts_defaults.extra_scripts}
lib_ldf_mode                = chain
lib_compat_mode             = strict
shared_libdeps_dir          = lib
lib_extra_dirs              =
                              lib/lib_basic
                              lib/lib_i2c
                              lib/lib_display
                              lib/lib_ssl
                              lib/lib_audio
                              lib/lib_rf
                              lib/lib_div

[scripts_defaults]
extra_scripts               = pio-tools/strip-floats.py
                              pio-tools/name-firmware.py
                              pio-tools/gzip-firmware.py
                              pio-tools/override_copy.py
                              pio-tools/download_fs.py

[esp_defaults]
; *** remove undesired all warnings
build_unflags               = -Wall
;                              -mtarget-align
                              -Wdeprecated-declarations
build_flags                 = -DCORE_DEBUG_LEVEL=0
                              -Wl,-Map,firmware.map
                              -Wno-deprecated-declarations
;                              -mno-target-align
                              -mtarget-align
                              -free
                              -fipa-pta
                              -Wreturn-type
; *********************************************************************
; *** Use custom settings from file user_config_override.h
                              -DUSE_CONFIG_OVERRIDE
; *********************************************************************

[esp82xx_defaults]
build_flags                 = ${esp_defaults.build_flags}
                              -DNDEBUG
                              -DFP_IN_IROM
                              -DBEARSSL_SSL_BASIC
                              ; NONOSDK22x_190703 = 2.2.2-dev(38a443e)
                              -DPIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_190703
                              -DPIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH_LOW_FLASH
                              ; VTABLES in Flash
                              -DVTABLES_IN_FLASH
                              ; remove the 4-bytes alignment for PSTR()
                              -DPSTR_ALIGN=1
                              ; restrict to minimal mime-types
                              -DMIMETYPE_MINIMAL
                              ; uncomment the following to enable TLS with 4096 RSA certificates
                              ;-DUSE_4K_RSA

[core]
; *** Esp8266 Tasmota modified Arduino core based on core 2.7.4. Added Backport for PWM selection
platform                    = espressif8266 @ 2.6.3
platform_packages           = tasmota/framework-arduinoespressif8266 @ ~2.7.4
build_unflags               = ${esp_defaults.build_unflags}
build_flags                 = ${esp82xx_defaults.build_flags}
; *** Use ONE of the two PWM variants. Tasmota default is Locked PWM
                              ;-DWAVEFORM_LOCKED_PHASE
                              -DWAVEFORM_LOCKED_PWM
```

### Additional info

This very similar to #4056, but I think it has some very different root cause.

lmagyar, 2022-01-17:
AFAIU (I'm relatively new to PIO)
- there is a [bug](https://bugs.python.org/issue37993) in Python 3.8+, that is categorized as "not a bug", so won't be resolved ever
- PIO has a fix in commit 4687665 for issue #3417
- this fix uses `os.path.abspath(projects_dir)` instead of `os.path.realpath(projects_dir)` to solve the problem of network drives
- that seems to be the root cause of this issue, the problem of subst-ed drives

I used PIO last time exactly before this commit 4687665 (build was OK), then, after a long break only now, 2 months later (build fails with this fix).

ivankravets, 2022-01-18:
Could you provide an output from the `net use` command?

lmagyar, 2022-01-18:
```
Status       Local     Remote                    Network

-------------------------------------------------------------------------------
OK           H:        \\192.168.22.22\config    Microsoft Windows Network
OK           I:        \\192.168.22.22\backup    Microsoft Windows Network
OK           J:        \\192.168.22.22\addons    Microsoft Windows Network
OK           K:        \\192.168.22.22\media     Microsoft Windows Network
OK           L:        \\192.168.22.22\share     Microsoft Windows Network
```

They used in non-PIO related context, they point to a virtual net card that is used by ssh client to forward ports through ssh tunnel to a server.

ivankravets, 2022-02-15:
I see. How does `$` appear in Windows path? It can lead to a problem on SCons site (when building project) because SCons will try to substitute `$VARs` with their values.

lmagyar, 2022-02-15:
I simply named manually a folder with a $ in it's name because it will be on top, separated from other folders (I hate mixing stuff on drive C:, so I also subst-ed it into a separate drive). I'm using Win, and it has never been a problem, so it took me days until I figured out this can be the root cause. I'm new this ecosystem and learning the hard way. :)

In summary: don't use ANY special characters in folder/branch/etc. names, \w should be a safe option.

HankLloydRight, 2022-07-04:
> I simply named manually a folder with a $ in it's name because it will be on top

Just FYI, I use ! for this purpose (or even !!) and it works great and doesn't interfere with anything else.

ivankravets, 2023-04-21:
The https://github.com/platformio/platformio-core/issues/3417 has been resolved. Please re-test with `pio upgrade --dev`.

closed 2023-04-21 by ivankravets