platformio/platformio-core #4156: Platformio creates folder instead of using existing one

CommanderRedYT, 2022-01-19:
What kind of issue is this?

- [ ] **PlatformIO Core**.
     Issue with Files created by the PlatformIO Core.

------------------------------------------------------------------

### Configuration

**Operating system**: Arch Linux (5.16.1-arch1-1; x86_64)

**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 5.2.4

### Description of problem
When using different locale (for example german), the Default-Folder is in the wrong place. On german / austrian systems, we don't have the "Documents"-folder, but the "Dokumente"-folder. When Platformio now creates the PlatformIO/Projects/... folder, it creates a ~/Documents folder. I am aware that I can change the default projects folder path, but this is the default path so i would assume that I would not have to change this.

#### Steps to Reproduce

1. Use a german / austrian locale (I think this needs to be done when installing the OS, so reproducing could be little bit hard)
2. Install PIO
3. There should be a folder called "Dokumente" and one called "Documents"

### Actual Results
Two folders, each with the same name but in different languages.

### Expected Results
PlatformIO should check the locale and create the folder based on that. This could be done by for example using the `xdg-user-dir` command.
```bash
[user@archlinux ~]$ xdg-user-dir DOCUMENTS
/home/user/Dokumente # on german systems
/home/user/Documents # on english systems
```

ivankravets, 2022-01-19:
See https://docs.platformio.org/en/latest/core/userguide/cmd_settings.html#projects-dir

0xFEEDC0DE64, 2022-01-19:
Platformio is not meant for anything productive, just rapid prototyping for an evening session and then throw away projects. No wonder they even do this completely wrong

ivankravets, 2022-01-19:
@CommanderRedYT I updated the initial comment above adding a temporary solution. There is no ETA when this issue will be resolved. It is very specific to the OS/Locale. If you have an idea with a quick fix, please submit PR.

ivankravets, 2022-01-19:
@0xFEEDC0DE64 what is wrong with PlatformIO? Did you have any issues?

CommanderRedYT, 2022-01-20:
@ivankravets I created a PR (#4158) that should resolve the issue. I saw that under windows it already used the API for getting the correct path, so I just added my snippet in the except-case

ivankravets, 2022-01-20:
@CommanderRedYT thanks for the PR!

I refactored it a little bit. Please re-test with `pio upgrade --dev`.

Does it work now?

closed 2022-01-20 by ivankravets